raspiblitz/raspiblitz #4027: LND dies after upgrade to 1.9.0

scarolan, 2023-07-22:
I used to leave my Raspiblitz running for months with no issues. Now LND crashes after about a day of running. Systemd shows it as dead until i restart it again

```
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sat 2023-07-22 02:12:29 BST; 19h ago
    Process: 3743191 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 3743470 ExecStart=/usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf (code=exited, status=0/SUCCESS)
   Main PID: 3743470 (code=exited, status=0/SUCCESS)
     Status: "Wallet unlocked"
        CPU: 3h 56min 10.190s

Jul 21 13:03:03 raspberrypi systemd[1]: Starting Lightning Network Daemon...
Jul 21 13:03:28 raspberrypi systemd[1]: Started Lightning Network Daemon.
Jul 22 02:12:29 raspberrypi systemd[1]: lnd.service: Succeeded.
Jul 22 02:12:29 raspberrypi systemd[1]: lnd.service: Consumed 3h 56min 10.190s CPU time.
```

scarolan, 2023-07-22:
If anyone knows which log file to look in to see why this is dying I can post the error message(s).

thecharacter, 2023-07-23:
Hi team, Thanks for your work on this project. I've encountered a known bug (ref: #3885) also in v1.8 blitz. LND works after resetting but I would use that time to close any channels you have open. Bitcoin still runs so I'm primarily running my blitz to keep the blockchain synced. Hoping this helps in troubleshooting. Appreciate all your efforts.

scarolan, 2023-07-24:
Update:  Running stable for over 24 hours now.  Fingers crossed it stays that way.

openoms, 2023-07-25:
> If anyone knows which log file to look in to see why this is dying I can post the error message(s).

there are two places to look. The shortcut:
```
lndlog
```
brings up the logs, also in SYSTEM menu.

The journal can be viewed (like for every service) with:
```
sudo journalctl -fu lnd -n 100
```

scarolan, 2023-07-25:
Thanks, neither of those revealed any errors and the node has been stable for two days now.  If it makes it through the weekend I'll close out this issue.

scarolan, 2023-07-28:
Ok finally captured the log.  This is what it looks like when it fails.

```
2023-07-27 04:36:42.808 [INF] CRTR: Pruning channel graph using block 0000000000000000000019512fa87238eda9c79728e072cd955ade472fc46f90 (height=800428)
2023-07-27 04:36:42.919 [INF] NTFN: New block: height=800428, sha=0000000000000000000019512fa87238eda9c79728e072cd955ade472fc46f90
2023-07-27 04:36:43.053 [INF] UTXN: Attempting to graduate height=800428: num_kids=0, num_babies=0
2023-07-27 04:36:55.378 [INF] CRTR: Block 0000000000000000000019512fa87238eda9c79728e072cd955ade472fc46f90 (height=800428) closed 0 channels
2023-07-27 04:37:14.085 [INF] CRTR: Processed channels=0 updates=68 nodes=4 in last 59.998401687s
2023-07-27 04:38:14.084 [INF] CRTR: Processed channels=0 updates=66 nodes=0 in last 59.999160037s
2023-07-27 04:39:14.085 [INF] CRTR: Processed channels=0 updates=39 nodes=0 in last 1m0.00081756s
2023-07-27 04:39:33.945 [INF] HLCK: Health check: chain backend, call: 1 failed with: health check: chain backend timed out after: 30s, backing off for: 2m0s
2023-07-27 04:40:02.291 [INF] CRTR: Pruning channel graph using block 00000000000000000000d441e8dd73201bc36668c74a3506a6d0a2e253af74e1 (height=800429)
2023-07-27 04:42:03.957 [INF] HLCK: Health check: chain backend, call: 2 failed with: health check: chain backend timed out after: 30s, backing off for: 2m0s
2023-07-27 04:43:10.659 [INF] CRTR: Block 00000000000000000000d441e8dd73201bc36668c74a3506a6d0a2e253af74e1 (height=800429) closed 1 channels
2023-07-27 04:43:10.840 [INF] CRTR: Processed channels=0 updates=79 nodes=0 in last 3m56.729205575s
2023-07-27 04:43:14.792 [INF] CRTR: Processed channels=0 updates=43 nodes=4 in last 3.95247107s
2023-07-27 04:44:14.085 [INF] CRTR: Processed channels=0 updates=176 nodes=0 in last 59.292360369s
2023-07-27 04:44:34.445 [CRT] SRVR: Health check: chain backend failed after 3 calls
2023-07-27 04:44:34.445 [INF] SRVR: Sending request for shutdown
2023-07-27 04:44:34.469 [INF] LTND: Received shutdown request.
2023-07-27 04:44:34.473 [INF] LTND: Shutting down...
2023-07-27 04:44:34.504 [INF] LTND: Systemd was notified about stopping
2023-07-27 04:44:34.505 [INF] LTND: Gracefully shutting down.
2023-07-27 04:44:34.506 [INF] NANN: Channel Status Manager shutting down
2023-07-27 04:44:34.507 [INF] HSWC: HTLC Switch shutting down
2023-07-27 04:44:34.525 [ERR] CRTR: Resuming payment 1ec768386de90e03f7a73eccfd952422bad278e887a2ec98acf389680203e05e failed: htlcswitch shutting down.
2023-07-27 04:44:34.525 [ERR] CRTR: Resuming payment 8ba077fc064aa61c8e00f893f75a9bfe94543c6c489d864ce0d87c764861a298 failed: htlcswitch shutting down.
2023-07-27 04:44:34.525 [ERR] CRTR: Resuming payment fb8eb9ddd64d9798c8f6243afc22a71babb645c631f0290b79c7b828e59c5964 failed: htlcswitch shutting down.
```

openoms, 2023-07-28:
Likely a performance issue. How is the load in htop?

Try these settings in the lnd.conf (discussed in https://github.com/raspiblitz/raspiblitz/pull/4010):
```
[healthcheck] 
healthcheck.chainbackend.attempts=3
healthcheck.chainbackend.timeout=2m0s 
healthcheck.chainbackend.interval=1m30s
```

scarolan, 2023-07-28:
Ok will give it a try and report back.

scarolan, 2023-08-02:
Adding the health check kept it up and running for a few more days but still eventually crashes:

```
2023-08-02 01:59:01.503 [INF] UTXN: Attempting to graduate height=801276: num_kids=0, num_babies=0
2023-08-02 01:59:05.789 [INF] CRTR: Block 00000000000000000001a5455150df515b3d5900a6320ca46a821d5e046c5530 (height=801276) closed 0 channels
2023-08-02 02:06:53.296 [INF] HLCK: Health check: chain backend, call: 1 failed with: health check: chain backend timed out after: 2m0s, backing off for: 2m0s
2023-08-02 02:07:55.748 [INF] CRTR: Pruning channel graph using block 000000000000000000022f44c3fb4183f6c1fb5751d4ab9da1c8d962c53abaee (height=801277)
2023-08-02 02:10:33.621 [INF] CRTR: Block 000000000000000000022f44c3fb4183f6c1fb5751d4ab9da1c8d962c53abaee (height=801277) closed 0 channels
2023-08-02 02:12:53.130 [INF] HLCK: Health check: chain backend, call: 1 failed with: health check: chain backend timed out after: 2m0s, backing off for: 2m0s
2023-08-02 02:16:53.183 [INF] HLCK: Health check: chain backend, call: 2 failed with: health check: chain backend timed out after: 2m0s, backing off for: 2m0s
2023-08-02 02:20:53.537 [CRT] SRVR: Health check: chain backend failed after 3 calls
2023-08-02 02:20:53.639 [INF] SRVR: Sending request for shutdown
2023-08-02 02:20:53.653 [INF] LTND: Received shutdown request.
2023-08-02 02:20:53.657 [INF] LTND: Shutting down...
2023-08-02 02:20:53.748 [INF] LTND: Systemd was notified about stopping
2023-08-02 02:20:53.748 [INF] LTND: Gracefully shutting down.
2023-08-02 02:20:53.853 [INF] NANN: Channel Status Manager shutting down
2023-08-02 02:20:53.864 [INF] HSWC: HTLC Switch shutting down
```

scarolan, 2023-08-02:
htop load doesn't seem unreasonable

[image]

rootzoll, 2023-08-03:
@scarolan can you try if the new new v1.10.0rc2 solves this problem: https://github.com/raspiblitz/raspiblitz/tree/v1.10#downloading-the-software

it contains the new healthcheck parameters and a new lnd with the memleak fix.

closed 2024-01-28 by scarolan