raspiblitz/raspiblitz #4029: [amd64] raspi repo signature error

openoms, 2023-07-25:
```
W: GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'http://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
Solved by removing:
```
sudo rm /etc/apt/sources.list.d/raspi.list 
```
so shouldn't be added

rootzoll, 2023-08-11:
@openoms can this issue be closed for v1.10.0?

openoms, 2023-08-11:
yes, the solution is merged and working.

closed 2023-08-11 by openoms