raspiblitz/raspiblitz #4046: reboot does not work

cstenglein, 2023-07-30:
When triggering a reboot over the WebUI and then SSH into the blitz, the following happens:

First, the screen "Preparing reboot" appears.

Afterwards, I get the following message in the SSH menu:

```
***********************************************************
RaspiBlitz going to reboot
***********************************************************
SSH again into system with:
ssh admin@<localIp>
Use your password A
***********************************************************
```

But it does not really shut down, I'm just back to the command line.

When I logout and SSH into it again, the same happens.

rootzoll, 2023-08-01:
I was not able to reproduce on my testmachine. When I chose "reboot" it triggered a reboot. But it can be that in special scenarios (other service is under heavy load, etc) the reboot script reacts differently. Will push this as an edge case to a later release to take a deeper look.

rootzoll, 2023-08-01:
But trigger on WebUI and then SSH into is a really strong edge case. When I jujst triggered from WebUI reboot worked.

BituBi, 2025-08-28:
I got similar issue: 

***********************************************************
***********************************************************
RaspiBlitz going to reboot
***********************************************************
SSH again into system with:
ssh admin@192.168.80.111
Use your password A
***********************************************************
admin@192.168.80.111:~ ₿

How to verify if it is rebooting or shutdown? on PI4 red led still on and green and HDD blinking