raspiblitz/raspiblitz #4050: How to warn users of long running, compute heavy tasks on Raspberry Pis

fusion44, 2023-07-31:
We need a system in place to warn the user of when he's about to start a process that uses up all the computing power of the Blitz. This usually renders the API / WebUI unusable for as long as the task is running. For example, I've activated the Bitcoin RPC Explorer App as a test and it rendered the API / WebUI completely unresponsive. Bitcoin Core builds the txindex for the first time which pretty much maxes out all cores and takes hours.

I don't think I can do much on API side because the processing power just is not there to answer requests from the frontend.

As a first step I'll add an info endpoint to the API so the frontend can ask the API if the txindex is already build and doesn't need to warn the user.

rootzoll, 2023-07-31:
Also see #4049

cstenglein, 2023-08-02:
I think we could prioritize the blitzapi service 

https://bristot.me/setting-real-time-priorities-to-services-using-systemd/