raspiblitz/raspiblitz #4065: Update V1.9.0 to V1.10.0rc2 doesn't work

mr-q17, 2023-08-06:
Update from a well working V1.9.0 system (RPI4 4GB/1TB HDD USB3) to V1.10.0rc2 hangs in Status "Preparing Blockchain Sync Please Wait..."  for over 48h now.
For me it seems that there is nothing usefully in process. Service bitcoind is not found.
 
Update process followed as described in menu on a new SD-card.
Refreshing the card comes to the same result.

Any idea with log I can use to locate the problem?

The normal debug-log is here:
https://l.termbin.com/mcis9

Switch back the old SD-card to V1.9.0 works directly and is now syncing from 99,88% because of 3-4 days of.
Debug form this running setup is here:
https://l.termbin.com/kna8z

Many thanks to all of the Raspiblitz-Team for your great work...

rootzoll, 2023-08-08:
Thanks for the report. This could be related to the changed bitcoind.service creation. Will run some test on this again.

mr-q17, 2023-08-08:
@rootzoll Thanks for your feedback.

After switching back to V1.9.0 as I described above it seems working for short time, but hangs after the blockchainsync was reached 100%.

After restart it hangs here:
[image]

Debug-file is here: [https://l.termbin.com/d10f](url)
`Printing debug logs. Be patient, this should take maximum 2 minutes ..
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory
Redacting ..
redacting file: /var/cache/raspiblitz/debug.log
https://l.termbin.com/d10f`

Reflashing the SD-card with the same V1.9.0 comes to the same result after the recovery process.

Any solution to get back a running system?

...don´t touch a running system...

rootzoll, 2023-08-08:
Log file says:

`Aug 08 13:29:37 raspberrypi bitcoind[2912]: Error during initializaton - check debug.log for details?

But then there seems to be no debuglog .. can you check:
`sudo tail -n 50 /mnt/hdd/bitcoin/debug.log`

mr-q17, 2023-08-09:
Status update, I tryed some things without success...

> But then there seems to be no debuglog .. can you check:
`sudo tail -n 50 /mnt/hdd/bitcoin/debug.log` 

Right, there is no `/mnt/hdd/bitcoin/debug.log` but why? Permissions?

When I create a `/mnt/hdd/bitcoin/debug.log` manually as bitcoin user then it seems that this file will be deleted on restart.
So, the question is: Can't Raspiblitz create that file or can't Raspiblitz start bitcoind.service?

`sudo systemctl status bitcoind`

```
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-08-08 17:45:45 BST; 17s ago
    Process: 85051 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUC>
    Process: 85084 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 85097 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoi>
        CPU: 323ms

Aug 08 17:45:45 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Aug 08 17:45:45 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Aug 08 17:45:45 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Aug 08 17:45:45 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Aug 08 17:45:45 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
```

After that I tried V1.10.0rc2 again...
1st SD-card flash:
Startup-process via recovery option
-> same result like above: "Blockchain Progress: 5 restarts..."

2nd SD-card flash:
Startup-process with fresh install but keep blockchain and upload the lnd-rescue-*.tar.gz option
-> ran into error after few hours.
LCD shows "System ran into an error: etc-testrun-fail"
SSH shows this: https://termbin.com/c9ju

Restart from this state shows this:
[image]

3rd SD-card flash:
Same procedure as the 2nd try but comes to the same result.

mr-q17, 2023-08-10:
Status Update:

**Node is back and alive!**
Runs now again with V1.9.0 inclusive all LN-Channels and funds.

Procedure to bring it back to live was the same as I tryed with the V1.10.0rc2 described above:
> 2nd SD-card flash:
Startup-process with fresh install but keep blockchain and upload the lnd-rescue-*.tar.gz option

Setup runs without any problems in about 2h back to a running system.

After that I patched the system via Update/Patch-Menu from the V1.9.0 branch. I recommend to install the patches directly after setup, otherwise when you install other apps / connect mobile apps before it runs into a crash ([see Log-file](https://termbin.com/gbxy9)) and you must start again from the scratch.
My experience was that the WebUI works much more better after the patches was installed. Before that, only see the twisting circle... 

Then I conneced Zeus and FullyNoded and install RTL without any problems.
Installing BTCpayServer was failed but for this I open a new issue.

Now I let it run until V1.10.0 is released.

rootzoll, 2023-08-26:
This should be fixed in v1.10.0 - closing for now. Please reopen if still is a problem from v1.10.0rc4 onwards.

closed 2023-08-26 by rootzoll