raspiblitz/raspiblitz #4069: 1.10.0rc2 cannt connect to electrs

rootzoll, 2023-08-08:
_Electrum wouldn't connect to Electrs. The properties in the Electrum shortcut were right written, however in the config file was written: "oneserver": false. Setting that to "true" didn't help, so I did set it back to "false". Then, I blanked the line in the Electrum shortcut properties in Windows and filled the same line again. That helped! Now, fingers crossed that LND keeps running!_

original report: https://github.com/raspiblitz/raspiblitz/issues/3800#issuecomment-1666838674

rootzoll, 2023-08-08:
I am not sure how to do it that on Windows. But I added info how to start the elctrum App with "oneserver" on MacOS with `open -a /Applications/Electrum.app --args --oneserver --server ${localIP}:${portSSL}:s`. Not sure if there is something similar for Win10. Chat GPT sys this should work - @jessieraspi can you try this:
`"C:\Program Files (x86)\Electrum\electrum.exe" --oneserver --server 192.168.178.61:50002:s`

JessieRaspi, 2023-08-08:
In windows I work with the shortcut for Electrum. This is what I learned some time ago. 

To always connect to your own node when clicking the Electrum-shortcut go to C:\Program Files (x86)\Electrum and look for the right version of your electrum-exe. Right click this version and choose: create shortcut. When asked for creation on the Desktop click Yes.
Then go to your Desktop and right click the Electrum shortcut and choose Properties. Look for Target and replace the string with the following:

 "C:\Program Files (x86)\Electrum\electrum-x.x.x.exe" --oneserver --server 192.168.X.X:50002:s
  
 Save the settings.  Every time you open Electrum with that shortcut you will have your own custom node settings. 

Each time after updating Electrum you have to check if the string in the shortcut is still there. If not, then fill it in again with the updated version. For testing if it really connects to my node, I made a test wallet in Electrum.

rootzoll, 2023-08-08:
@cstenglein For the WebUI it might be a good mission to create tutorials how to connect Electrum on different platforms, etc. This could be part of our UX project.