raspiblitz/raspiblitz #4107: v1.10.0rc2 and v1.10rc2 having issues with autostart bitcoind service

cwgriswold, 2023-08-14:
autostarting bitcoind daemon after server reboot seems to be broken on debian 11.7.

```
             RaspiBlitz v1.10.0rc3 Chakalaka
               Bitcoin Fullnode + Lightning Network + Tor
        ,/     -------------------------------------------
      ,'/      Refreshed: Mon, 14 Aug 2023 11:00:48 +0200
    ,' /       CPU load 3.10, 0.85, 0.29, temp 0°C 0°F
  ,'  /_____   Free Mem 5177M / 7944M  HDDuse 626G (66%)
,'_____    ,'  SSH admin@172.31.1.202 d957.5KiB u2.5MiB
      /  ,'    Web Admin --> http://172.31.1.202
     / ,'      bitcoin v25.0.0 mainnet 4 peers
    /,'        Blocks 803091/803091 Sync OK 100.00%
   /'
               LND 0.16.4-beta Wallet 866,584 sat
               2222/2222 Channels 1 sat 0 peers
               Fee Report (D-W-M): 0-0-0 sat

xxxxxx.onion:9735
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

Gathering information to update info ... please wait.
```

```
admin@172.31.1.202:~₿ sudo systemctl status bitcoind
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-08-14 11:00:09 CEST; 2min 49s ago
   Main PID: 860 (bitcoind)
      Tasks: 18 (limit: 9490)
     Memory: 4.8G
        CPU: 4min 1.898s
     CGroup: /system.slice/bitcoind.service
             └─860 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Aug 14 10:59:50 raspiblitz systemd[1]: Starting Bitcoin daemon...
Aug 14 10:59:50 raspiblitz blitz.systemd.sh[684]: Could not connect to Redis at 127.0.0.1:6379: Connection refused
Aug 14 10:59:50 raspiblitz blitz.systemd.sh[719]: Could not connect to Redis at 127.0.0.1:6379: Connection refused
Aug 14 11:00:09 raspiblitz systemd[1]: Started Bitcoin daemon.
```

sudo systemctl restart bitcoind fixes the problem but only until next server reboot. haven't had issues with rc1.

```
admin@172.31.1.202:~₿ sudo systemctl status redis-server
● redis-server.service - Advanced key-value store
     Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-08-14 10:59:50 CEST; 6min ago
       Docs: http://redis.io/documentation,
             man:redis-server(1)
   Main PID: 629 (redis-server)
     Status: "Ready to accept connections"
      Tasks: 5 (limit: 9490)
     Memory: 7.5M
        CPU: 6.473s
     CGroup: /system.slice/redis-server.service
             └─629 /usr/bin/redis-server 127.0.0.1:6379

Aug 14 10:59:50 raspiblitz systemd[1]: Starting Advanced key-value store...
Aug 14 10:59:50 raspiblitz systemd[1]: Started Advanced key-value store.
```

maybe it's worth a try to let bitcoind start after redis-server. 

my bitcoin service:

```
# RaspiBlitz: systemd unit for bitcoind
# based  on https://github.com/bitcoin/bitcoin/blob/master/contrib/init/bitcoind.service
[Unit]
Description=Bitcoin daemon

#After=bootstrap.service
#Wants=bootstrap.service
After=network-online.target
Wants=network-online.target

# for use with sendmail alert
#OnFailure=systemd-sendmail@%n

[Service]
Environment='MALLOC_ARENA_MAX=1'
ExecStartPre=-/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED
ExecStart=/usr/local/bin/bitcoind -daemonwait \
                                  -conf=/mnt/hdd/bitcoin/bitcoin.conf \
                                  -datadir=/mnt/hdd/bitcoin \
                                  -debuglogfile=/mnt/hdd/bitcoin/debug.log

# Make sure the config directory is readable by the service user
PermissionsStartOnly=true
ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin

# Process management
####################
Type=forking
Restart=on-failure
TimeoutStartSec=infinity
TimeoutStopSec=600

# Directory creation and permissions
####################################
# Run as bitcoin:bitcoin
User=bitcoin
Group=bitcoin

StandardOutput=null
StandardError=journal

# Hardening measures
####################
# Provide a private /tmp and /var/tmp.
PrivateTmp=true
# Mount /usr, /boot/ and /etc read-only for the process.
ProtectSystem=full
# Deny access to /home, /root and /run/user
ProtectHome=true
# Disallow the process and all of its children to gain
# new privileges through execve().
NoNewPrivileges=true
# Use a new /dev namespace only populated with API pseudo devices
# such as /dev/null, /dev/zero and /dev/random.
PrivateDevices=true
# Deny the creation of writable and executable memory mappings.
MemoryDenyWriteExecute=true

[Install]
WantedBy=multi-user.target

```
any ideas?

rootzoll, 2023-08-25:
The 1.10.0rc3 already adresses the missing debug.log but has some other problem .. this should be retested when v1.10.0rc4 is out.

cwgriswold, 2023-08-28:
just testing 1.10rc4. issue persists. 

```
jbacvghhyd.onion:9735
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

Screen is updating in a loop .... press 'x' now to get back to menu.^CSIGINT or SIGTERM received, exiting...
admin@172.31.1.202:~₿ sudo systemctl status bitcoind
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-08-28 09:26:21 CEST; 3min 42s ago
   Main PID: 872 (bitcoind)
      Tasks: 18 (limit: 9490)
     Memory: 4.7G
        CPU: 4min 48.194s
     CGroup: /system.slice/bitcoind.service
             └─872 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitco>

Aug 28 09:26:07 raspiblitz systemd[1]: Starting Bitcoin daemon...
Aug 28 09:26:08 raspiblitz blitz.systemd.sh[737]: Could not connect to Redis at 127.0.0.1:6379: Connection refused
Aug 28 09:26:08 raspiblitz blitz.systemd.sh[747]: Could not connect to Redis at 127.0.0.1:6379: Connection refused
Aug 28 09:26:21 raspiblitz systemd[1]: Started Bitcoin daemon.
lines 1-14/14 (END)
```

still have to manually restart bitcoind service after reboot.

rootzoll, 2023-08-28:
Are you testing with min or fat sd card image?

cwgriswold, 2023-08-28:
neither. being on debian 11.

rootzoll, 2023-08-29:
I committed in `dev` branch a changed service fpr bitcoind to wait for redis. Can you test if this cleans things up for you?

cwgriswold, 2023-08-30:
mmh. still the same on rc5.

agentmic487, 2023-09-01:
> mmh. still the same on rc5. didn't find any changes in the after or wanted section in bitcoind.service

Hi mate, are you still having this issue or got it solved? If not you can get in touch with me so we can both get it solved together because a lot of people have come with same complain and it work for them i believe we can solve this together...Don't wait to say hi

cwgriswold, 2023-09-11:
updated to 1.10.0 but still having issues after reboot. the bitcoin service doesn't come up. also need to restart lnd after manually restarting bitcoin. did you change anything in the systemd file so the service comes up after a reboot?