raspiblitz/raspiblitz #4108: add uptime to 00infoBlitz.sh

cwgriswold, 2023-08-14:
is possible to add uptime on the info screen. after each patch i have manually have to read it.

datetime=$(date -R)

**uptime=$(uptime -p)**

stty sane
sleep 1
clear

printf "
${color_yellow}
${color_yellow}
${color_yellow}
${color_yellow}               ${color_amber}%s ${color_green} ${ln_alias} ${upsInfo}
${color_yellow}               ${color_gray}${network^} Fullnode${LNinfo} ${torInfo}
${color_yellow}               ${color_yellow}%s
**${color_yellow}        ,/     ${color_gray}%s**
${color_yellow}      ,'/      ${color_gray}%s
${color_yellow}    ,' /       ${color_gray}%s, temp %sÂ°C %sÂ°F
${color_yellow}  ,'  /_____   ${color_gray}Free Mem ${color_ram}${ram} ${color_gray} HDDuse ${color_hdd}%s${color_gray}
${color_yellow},'_____    ,'  ${color_gray}SSH admin@${internet_localip}${color_gray} d${internet_rx} u${internet_tx}
${color_yellow}      /  ,'    ${color_gray}${webuiinfo} 
${color_yellow}     / ,'      ${color_gray}${network} ${color_green}${networkVersion}${color_gray}${chain}net ${networkConnectionsInfo}
${color_yellow}    /,'        ${color_gray}${blockInfo} %s
${color_yellow}   /'          ${color_gray}
${color_yellow}               ${color_gray}${LNline}
${color_yellow}               ${color_gray}${ln_channelInfo} ${ln_peersInfo}
${color_yellow}               ${color_gray}${ln_feeReport}
$lastLine
" \
"RaspiBlitz v${codeVersion}" \
"-------------------------------------------" \
***"${uptime}" \ ***
"Refreshed: ${datetime}" \

would look like:
```
 RaspiBlitz v1.9.0  moosmutzel
               Bitcoin Fullnode + Lightning Network + Tor
               -------------------------------------------
        ,/     up 5 weeks, 3 days, 3 hours, 45 minutes
      ,'/      Refreshed: Mon, 14 Aug 2023 11:48:41 +0200
    ,' /       CPU load 3.68, 3.57, 3.22, temp 60°C 140°F
  ,'  /_____   Free Mem 1608M / 3844M  HDDuse 626G (72%)
,'_____    ,'  SSH admin@192.168.178.10 d64.7GiB u109.7GiB
      /  ,'    Web Admin --> http://192.168.178.10
     / ,'      bitcoin v25.0.0 mainnet 19 peers
    /,'        Blocks 803099/803099 Sync OK 100.00%
   /'
               LND 0.16.4-beta Wallet 1 sat
               3/3 Channels 1 sat 4 peers
               Fee Report (D-W-M): 0-0-0 sat
```

thx.

cwgriswold, 2023-08-17:
sry i don't use lcd, just ssh. so i don't know if the extra line fits the lcd. 

uptime can be useful if you have unrecognizable reboots or if it's time to run an apt upgrade.

rootzoll, 2024-04-23:
The uptime would need to be a very short string .. so just something like: "6d" or "1m" then it could be added to the "Refreshed: ..." line when we shorten a bit the datestamp.

`uptime -p` is still much to long with an output like `up 6 days, 21 hours, 5 minutes`
so maybe the following can be much shorter:
`uptime -p | cut -d ' ' -f 2- | cut -d ',' -f 1 | awk '{print $1 substr($2, 1, 1)}'`

And also this would not be written into the infoscreen ... it would be part of the `background.scan` service, written to cache and the LCD script would get it it from cache then like the other values.

rootzoll, 2024-05-29:
OK uptime will be part of v1.11.1rc1 ... looking like this:

[image]

PR #4600 (2024-05-29): feat: Add system uptime text to infoBlitz script #4108
fixes #4108