raspiblitz/raspiblitz #4110: pin CLBOSS fork for msat compatibilty

openoms, 2023-08-16:
CLBOSS is not compatible for the latest CLN releases after "msat cleanup".

Will need to test a pendimg PR and pin it in the install script: https://github.com/ZmnSCPxj/clboss/pull/162

chrisguida, 2023-08-19:
[This line](https://github.com/raspiblitz/raspiblitz/blame/ebfb13b55dd805455542b60716c4edf3da2b024f/home.admin/config.scripts/cl-plugin.clboss.sh#L63) is also missing `sudo -u bitcoin autoreconf -i` before the `sudo -u bitcoin .configure` step

openoms, 2023-08-20:
We got some merges on the weekend so can build from master: https://github.com/ZmnSCPxj/clboss/commits/master

openoms, 2023-08-20:
@chrisguida opened #4121 to fix clboss with the latest CLN
realized that it does not work yet on signet, but hoping for https://github.com/ZmnSCPxj/clboss/pull/149 to be merged

chrisguida, 2023-08-20:
Cool, what does blitz need signet for?

ksedgwic, 2023-08-20:
VLS currently uses CLBOSS in testnet to create load for signers under test.  I've wondered what the signet lightning network looked like and if it was better.

openoms, 2023-08-20:
The LN on the default signet is indeed much smaller, but most nodes are functional.
There are many more zombie ones on testnet.

The deciding problem with testnet3 for me is the occasional block storms which make bitcoind unusable for a day or so. Last when I caught it: https://twitter.com/openoms/status/1537780122673086464

Here is a list of tools available to move testing to signet: https://dev.galoy.io/docs/signet

It makes sense to make the `min_node_to_process` minimal on signet especially that custom (even smaller) aignet network could also be used. See the Plebnet Playground: https://plebnet.wiki/wiki/Plebnet_Playground or Mutinynet for example: https://blog.mutinywallet.com/mutiny-wallet-signet-release/

chrisguida, 2023-08-21:
mutinynet.com is also a fantastic signet for testing, as it has 30-second blocktimes.

rootzoll, 2023-08-25:
Merged #4121 - will be part of v1.10.0rc4. Final testing.

closed 2023-09-14 by rootzoll