raspiblitz/raspiblitz #4112: v1.10.0rc3 - Not given the option how to get the blockchain on a new build.

marssystems, 2023-08-17:
During Setup, I wasn't given the option how to get the blockchain.

Self-certify.
Copy from another RaspiBlitz.
Download from a PC.

The setup went straight into the self-certify option.

Another thing - there is no ctrl-c option in SSH to get to the command prompt/
Ctrl-c just closes the SSH session.
[image]

It would be nice to go to the command prompt to have a look at some logs to see how the build is progressing - as in previous releases.

rootzoll, 2023-08-25:
Again like with your other report - this looks like it would behave if you login with `ssh pi@...` and not with `ssh admin@..` ... this is basically what the LCD should display (thats what the pi user is for) not the the admin user.

marssystems, 2023-08-25:
I  always login using the admin user - and this is the screen I get.

rootzoll, 2023-08-25:
This is very strange - because thats 100% the screen with "ssh admin@192.168.0.202 --> Password A" that will be shown when the logged in user is "pi" not "admin". Your the only user reporting this so far - can you please recheck your ssh setup.

marssystems, 2023-08-25:
ok -  I will doble check.

marssystems, 2023-08-25:
Just flashed a new sd card and now I get the correct menu screen. I didn't change my Terminus SSH login settings as they were already correctly set.
So I can't explain it - just glad it fixed itself.

closed 2023-08-26 by rootzoll