raspiblitz/raspiblitz #4118: Lost-HDD ?

tweetytoo, 2023-08-19:
(1) I am not very techsavvy. 

(2) Some time ago, I noticed my raspiblitz was not active on the internet. IIRC, it said "logs full"

I tried to reboot, but then got a "lost-HDD" message, which is what I now see on boot.

I've tried to reflash and replace the SSD, but that's not the issue (I think).

(3) So, do I have any software fixes for this, or is the HDD somehow bricked?

tweetytoo, 2023-08-19:
So do I need to get a new SSD? Or is there a way to check/fix it when it's "not detected"?

rootzoll, 2023-08-25:
Try again with a fresh flashed sd card - if also second try fails your HDD/SSD seems not working in your setup.

tweetytoo, 2023-08-28:
Thanks for the reply @rootzoll , but I've indeed swapped for newly flashed cards 3x. No luck :(

Scotty-99, 2023-08-30:
Hi, I've the same issue. I wanted to install version 1.9 and since then I had this problem

merle94, 2023-10-03:
Same problem here. For me it appeared during updating from 1.9 to 1.10
HDD just worked fine. SD flashed and got this lost HDD error.

rootzoll, 2023-12-18:
OK so the problem is that `hddCandidate` is empty .. can be seen here that the second brackets are empty
`>Running Format: (ext4) ()`

Improved the HDD/SSD detection .. try if the following RaspberryPi v1.11.0RC1 image (Download under Artifacts) fixes this:
https://github.com/raspiblitz/raspiblitz/actions/runs/7201966440

tweetytoo, 2023-12-19:
Thanks for the suggestion but I gave up long ago and threw the whole rig away.

closed 2023-12-19 by tweetytoo

jhub-beep, 2024-01-08:
I do have the same issue with lost hdd message and 

> >Running Format: (ext4) ()

..while the SSD is connected to an USB 3.0 port of my Raspi 4. 
I am struggling to find the download of the suggested RaspberryPi v1.11.0RC1 under artifacts. I would willingly try the image.
Please advice @rootzoll. Thank you

jhub-beep, 2024-01-08:
After login at github the download link appeared... 👍 let you know the result after flashing the 1.11.0RC1

jhub-beep, 2024-01-08:
Still does not work for me with my Crucial BX500 2.5 SSD 1 TB while it is connected at an USB 3.0 port. When I use other Unix based OS it does work with this disc on those ports. I go ahead with the minimal image on an USB 2.0 port and might try later with next release.

merle94, 2024-02-19:
Giving again some feedback on this issue :
1. Node lost power 
2. After reboot got into "emergency mode"
3. Get into "emergency mode" after every reboot from now on 
4. Re-flashed SD card
5. Now getting the "lost HDD"-issue