raspiblitz/raspiblitz #4122: v1.19.0 Lnproxy fails to install.

marssystems, 2023-08-20:
*** INSTALL LNPROXY ***
# Check Framework: Go

The requested version of Go is already installed.
go version go1.19.5 linux/arm64

Adding user `lnproxy' ...
Adding new group `lnproxy' (1004) ...
Adding new user `lnproxy' (1004) with group `lnproxy' ...
Creating home directory `/home/lnproxy' ...
Copying files from `/etc/skel' ...
Macaroon saved to lnproxy.macaroon
Cloning into '/home/lnproxy/lnproxy'...
remote: Enumerating objects: 21, done.
remote: Counting objects: 100% (21/21), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 21 (delta 5), reused 0 (delta 0), pack-reused 0
Receiving objects: 100% (21/21), 5.60 KiB | 1.87 MiB/s, done.
Resolving deltas: 100% (5/5), done.
fatal: Could not parse object '7c8a14106b42cfd89471c1dc02d7baab1122dfa2'.

rootzoll, 2023-08-25:
OK updated to latest commit. Fix will be part of v1.10.0rc4 for final testing.

marssystems, 2023-08-25:
I just tried "the fix" and it seems to be working now but I get a couple of errors during the install -

First is with go - go: go.mod file not found in current directory or any parent directory.
Second at the end - # More info at:
https://github.com/lnproxy
# *** LNPROXY IS NOT INSTALLED ***

So I'm not sure if the install worked or not. Here is the install log -

Balance of Satoshis setting unchanged.
lnproxy setting changed ..
*** INSTALL LNPROXY ***
# Check Framework: Go

The requested version of Go is already installed.
go version go1.19.5 linux/arm64

Adding user `lnproxy' ...
Adding new group `lnproxy' (1004) ...
Adding new user `lnproxy' (1004) with group `lnproxy' ...
Creating home directory `/home/lnproxy' ...
Copying files from `/etc/skel' ...
Macaroon saved to lnproxy.macaroon
Cloning into '/home/lnproxy/lnproxy'...
remote: Enumerating objects: 21, done.
remote: Counting objects: 100% (21/21), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 21 (delta 5), reused 0 (delta 0), pack-reused 0
Receiving objects: 100% (21/21), 5.60 KiB | 1.87 MiB/s, done.
Resolving deltas: 100% (5/5), done.
HEAD is now at c1031bb Update README.md

go: go.mod file not found in current directory or any parent directory.
        'go get' is no longer supported outside a module.
        To build and install a command, use 'go install' with a version,
        like 'go install example.com/cmd@latest'
        For more information, see https://golang.org/doc/go-get-install-deprecation
        or run 'go help get' or 'go help install'.
go: cannot find main module, but found .git/config in /home/lnproxy/lnproxy
        to create a module there, run:
        go mod init

[Unit]
Description=lnproxy
After=lnd.service

[Service]
User=lnproxy
Group=lnproxy
Type=simple
ExecStart=/home/lnproxy/lnproxy/lnproxy -lnd-cert /home/lnproxy/.lnd/tls.cert /home/lnproxy/lnproxy.macaroon
Restart=on-failure
RestartSec=30
TimeoutSec=120

# Hardening measures
PrivateTmp=true
ProtectSystem=full
NoNewPrivileges=true
PrivateDevices=true

[Install]
WantedBy=multi-user.target
Created symlink /etc/systemd/system/multi-user.target.wants/lnproxy.service → /etc/systemd/system/lnproxy.service.
# OK - the lnproxy.service is enabled, system is on ready so starting service
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
Rule added
Rule added (v6)

# Hidden Service for lnproxy
HiddenServiceDir /mnt/hdd/tor/lnproxy
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:4749
HiddenServicePort 443 127.0.0.1:4750

Reloading Tor to activate the Hidden Service...

The Tor Hidden Service address for lnproxy is:
57ob45k3qvco7b3tl2bvhbvas7mvy5adq3jjzuz556o7s5tlieqr3nad.onion
use with the port: 80
or the port: 443

# Use your hidden service as a relay on the lnproxy Tor website:
dx7pn6ehykq6cadce4bjbxn5tf64z7e3fufpxgxce7n4f5eja476cpyd.onion
# Your address to be used as the relay:
http://57ob45k3qvco7b3tl2bvhbvas7mvy5adq3jjzuz556o7s5tlieqr3nad.onion/spec
# To use the API from another computer on your LAN:
curl -k https://192.168.0.201:4748/api/{invoice}?routing_msat={budget}\n
# The Tor Hidden Service address to share for using the API:
57ob45k3qvco7b3tl2bvhbvas7mvy5adq3jjzuz556o7s5tlieqr3nad.onion/api
# More info at:
https://github.com/lnproxy
# *** LNPROXY IS NOT INSTALLED ***
PyBLOCK setting unchanged.

rootzoll, 2023-08-26:
@openoms the build procedure of lnproxy seems to have changed after I needed to update to the latest git commit because the old one failed.

marssystems, 2023-10-29:
Hi!
Any update on this?

rootzoll, 2024-02-16:
Install process ran thru on my test for v1.11.0rc3 - please test as soon as RC3 is out.

rootzoll, 2024-03-19:
na its still broken ... looks like the "go get" is not allowed anymore "go install" refuses to accpet this setup. Have to mark `lnproxy` as "Deactivated for repair" in v1.11.0 release.

Might put out a bounty for repair (go kills required).

geco91, 2024-06-04:
recommend for close

marssystems, 2024-06-04:
Has the problem been fixed ?