raspiblitz/raspiblitz #4125: No servers could be used! Exiting.

consultingspeed, 2023-08-22:
Hello
I have the following data in my bitcoin.conf

```
# bitcoind configuration

# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=1
# disablewallet=1
listen=1

# Connection settings
rpcuser=raspibolt
rpcpassword=raspibolt
rpcport=8332

rpcallowip=0.0.0.0/0
rpcallowip=130.211.100.10
# rpcallowip=127.0.0.1
rpcbind=0.0.0.0

zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=600
maxorphantx=10
maxmempool=300
maxconnections=40
maxuploadtarget=5000

datadir=/mnt/hdd/bitcoin
# onlynet=onion
addnode=fno4aakpl6sg6y47.onion
addnode=toguvy5upyuctudx.onion
addnode=ndndword5lpb7eex.onion
addnode=6m2iqgnqjxh7ulyk.onion
addnode=5tuxetn7tar3q5kp.onion
```

also when i run 
`./bfgminer -o http://localhost:9332 -u raspibolt -p raspibolt  --no-hotplug --request-diff=1.5 --queue=8192`

I get this error
[2023-08-22 10:23:46] Started bfgminer 5.5.0-34-g866fd36f
 [2023-08-22 10:23:46] No devices detected!
 [2023-08-22 10:23:46] Waiting for devices
 [2023-08-22 10:23:46] Added local bitcoin RPC server on port 8332 as pool 1
 [2023-08-22 10:23:46] Probing for an alive pool
 [2023-08-22 10:23:46] Pool 1 slow/down or URL or credentials invalid
 [2023-08-22 10:23:46] Pool 0 slow/down or URL or credentials invalid
 [2023-08-22 10:23:47] No servers were found that could be used to get work from.
 [2023-08-22 10:23:47] Please check the details from the list below of the servers you have input
 [2023-08-22 10:23:47] Most likely you have input the wrong URL, forgotten to add a port, or have not set up workers
 [2023-08-22 10:23:47] Pool: 0  URL: http://localhost:9332  User: raspibolt  Password: raspibolt
 [2023-08-22 10:23:47] Pool: 1  URL: http://localhost:8332/#getcbaddr#allblocks  User: raspibolt  Password: raspibolt
No servers could be used! Exiting.

Please let me know what to do? How do I add this? stratum+tcp://stratum.braiins.com:3333
userID:  cods1.workerName
password: anythin

Please help

Regards
Danies

rootzoll, 2023-08-25:
It looks like you try to use your bitcoind as an https address for a pool .. I dont think thats how bfgminer works. But I dont have own experience on this to help ... maybe ask on the Telegram Chats if there is someone to excahnge on this topic.

closed 2023-08-25 by rootzoll

consultingspeed, 2023-08-25:
I am not able to setup the https://my.minerstat.com/workers#1

I can give you access and take a look to configure the server?

Regards
Dan

marssystems, 2023-08-26:
You need to add this - stratum+tcp://stratum.braiins.com:3333 - to your bfgminer config.
Also include your worker -cods1.workerName and your password -  anythin

./bfgminer -o  stratum+tcp://stratum.braiins.com:3333 -u cods1.worker1 -p  anythin --no-hotplug --request-diff=1.5 --queue=8192

You don't need to do anything with the bitcoin node running on your RaspiBlitz.

consultingspeed, 2023-08-26:
Hello Adam,
I removed the server. The bfgminer was workedbut but I dont know if I can
make money being a node for bitcoin and starting on the server, I was
wrong?

You think was near to make some money with Bitcoin node and bfgminer on it?

Regards
Dan

sâm., 26 aug. 2023, 03:13 Mariusz Adamczyk ***@***.***> a
scris:

consultingspeed, 2023-08-26:
Sorry for contacting you again. I have medical problems in family..
I want to deploy a new server on Google Cloud, could you please help me
what server is need to deploy a mining?
I can deply any server you request me, could we make a Share View and a
Google meet link?

Regards,
Dan

marssystems, 2023-08-26:
Are you running a bitcoin lightning node on your RaspiBlitz? - This is how you will make some money.

To make money with bfgminer, you have to connect your ASICs to a mining pool.
If you want to run your own pool on the RaspiBlitz, you need to install some software - try this-

https://github.com/benjamin-wilson/public-pool/tree/master

It works very well.
You can see how it looks and try it for yourself here -

https://web.public-pool.io/#/

Set up your bfgminer to use this pool for solo mining. What sort of ASICs do you have?

consultingspeed, 2023-08-26:
Dear Mariusz,

I dont have RaspiBlitz, I can only use Google Cloud server linux

marssystems, 2023-08-26:
OK - I understand.

If you only have Google cloud - its probably not worth setting up any kind of mining pool - as there are lots already and you wouldn't get enough customers to make any money.
You are better off using your bfgminer and ASICs to mine at an already established pool and make money that way.
This all depends on how much hashing power you have at your disposal.

consultingspeed, 2023-08-26:
Hello Marius

I can deploy any server is need. I have 6500 credit per year from Google
Cloud and I can make any server you need to make most high hashing power is
need. Please tell me and I will deploy it

Regards
Dan

sâm., 26 aug. 2023, 10:07 Mariusz Adamczyk ***@***.***> a
scris:

consultingspeed, 2023-08-26:
Dear Mariusz

You are better off using your bfgminer and ASICs to mine at an already
established pool and make money that way.

I will install a Ubuntu linux with Tesla CPU?

sâm., 26 aug. 2023, 10:07 Mariusz Adamczyk ***@***.***> a
scris:

marssystems, 2023-08-26:
Tesla CPU?   What is that for?

If you want to do bitcoin mining you need to run an Antminer S19 or something like that/

consultingspeed, 2023-08-26:
I see , Ok.

What to do to start?

Should I buy this products https://shop.fulmo.org/ and start bining bitcoin?

marssystems, 2023-08-26:
Sure!  Setting up a bitcoin lightning routing node is a good option but remember that you will have to invest at least 0.5 BTC to fund your channels for good connectivity and reasonable earnings. Nothing involving bitcoin is quick easy and cheap - requires a lot of commitment and high investment for good rewards.

consultingspeed, 2023-08-26:
I dont have a such payment now 0.5BTC ... I want to start earnings about at
the beginning about 50-60 EUR/day, on February 2024 I have with my kid a
brain surgery, I got a lot of donations.
Should I but some products at https://shop.fulmo.org/ ?

consultingspeed, 2023-08-26:
If I buy hashing power from nicehash.com , could you please advice me how
to setup to invest and increase profits using that website?

marssystems, 2023-08-26:
Like I said - yes - but don't expect to make 50 - 60 euro per day without some serious investment.

consultingspeed, 2023-08-26:
Do you have a whatsup nr,?
My phone is 0735569544 on whatsup

I have more questions

sâm., 26 aug. 2023, 13:49 Mariusz Adamczyk ***@***.***> a
scris:

consultingspeed, 2023-08-26:
No problem, for me is ok 10-25 EUR per day, but All I invest

sâm., 26 aug. 2023, 13:49 Mariusz Adamczyk ***@***.***> a
scris:

consultingspeed, 2023-08-28:
Hello Marius

I connected to public pool from that website, I attached screenshot. How
much earnings I make în 2 weeks?

Regards
Dan

sâm., 26 aug. 2023, 13:49 Mariusz Adamczyk ***@***.***> a
scris:

consultingspeed, 2023-09-10:
Dear Marius

I will use

stratum+tcp://public-pool.io:21496
username: <your BTC address>.<worker name>, password: x
When I try to buy hash power, it ask me to select encryption, I dont know
what to choose.

If I buy HashPower on Nicehash.com I could make money? I dont know what
settings to choose and to make it work.

Do you have telegrama id? My username is sunnyviews

On March I have my boy meeting with doctor for surggery

Regards
Dan

consultingspeed, 2023-10-04:
Dear Mariusz

I have installed LND and sync Bitcoin node, may I get your help to
configure my node to make earnings?

Regards,
Dan

marssystems, 2023-10-04:
You need to add funds to your node first - have you done that?

consultingspeed, 2023-10-04:
Dear Mariusz,
I didnt added funds on the wallet from the server, because I dont have
money sir
I can give you access to my server, do you want to setup?

consultingspeed, 2023-10-04:
I dont know what to do next, can you please configure it?

marssystems, 2023-10-04:
You can't configure the server until you add funds to it.

Read this guide for full instructions - https://docs.lightning.engineering/

consultingspeed, 2023-10-04:
thank you for reply. How much funds should i topup?
How much earning I will expect?

marssystems, 2023-10-04:
You can fund your RaspiBlitz with 0.1 BTC to start with.

How much you will earn depends on what other nodes you connect to and how much traffic will run through your node.

consultingspeed, 2023-10-05:
Its for me very expensive, I almose cover the server costs 300 USD per
month , can we work totheter and to borrow me and then give you back
from the fees?
I give you the server access

Remote server Connetion
34.88.247.107

root
TpTy2222?

Access Start9
TpTyTpTyTpTy2222?

marssystems, 2023-10-05:
You can buy sats from the pool to fund your node (300 per month). Read the docs.

consultingspeed, 2023-10-05:
Where to buy whcih pool, how? Only 300 sats per months?
I dont understand, its enouch 300 sats per month?

consultingspeed, 2023-10-05:
> Install this on your RaspiBlitz -
>
> https://github.com/lightninglabs/lightning-terminal#lightning-terminal-lit
>
> then you can borrow sats from the pool

Dear Mariusz
Could you please more to install it by you Please! I gave you access

marssystems, 2023-10-05:
Just SSH into your node and select it from the main menu.

consultingspeed, 2023-10-05:
I am not able to ssh în node, I tryed and I am not able.
I dont know how. Please

marssystems, 2023-10-05:
Open your terminal program (Putty or Terminus) and enter your node IP address. Then login using username - admin and your PasswordA. Look for LIT in the main menu.

https://github.com/raspiblitz/raspiblitz/raw/v1.10/pictures/services.png

consultingspeed, 2023-10-06:
Dear Marius,
I tryed and also installed putty and I cannot connect to the Oracle
virtualmachine instaleld on VM:

***@***.***
:9735

ssh-rsa
AAAAB3NzaC1yc2EAAAADAQABAAABAQChl4cY8CQfRtYm2slUOn+rSG7rRHoCMmKu8dyWq48k67iRegir6rqlvPuz03GdQuRge3EQ7WG6P5JkPVGW7/rb0S+oMWggi/fetwYP/I5Hrxq4awgWzEp/QPLRKm+Z0NECZKuW1nFhialVy6hF+yhGB7XLIp2AedTU8KArDU61ggTnQ4nFwtEKdILjZeiuCVLsegufHF9kJ+nbnZ8rXlWEt9t/KGHllTbARffMmCFUQLQ5AW37oBFE629q5shd+9zECfV6FJghlEkYbvDzaArUE8RzQqqBNge6aXty5PFhH1ti3wjtmpnbZzhTuIZkIb/wSLR+dH/5mpzl91JVuX1R
rsa-key-202310061

Internal IP from Google Server : 10.166.0.3
public ip 34.88.247.107

Could you please try?

Regards,
Dan

marssystems, 2023-10-06:
I tried using Putty on your public IP but I can't log in.

Maybe you should ask Oracle how to SSH into the VM.

consultingspeed, 2023-10-06:
I restarted now, just a second

consultingspeed, 2023-10-06:
Dear Marius,

Please check now. I restarted it successfully

Regards,
Dan Oproiu

marssystems, 2023-10-06:
NO - still not working.

consultingspeed, 2023-10-06:
Dear Mariusz

I attach RDP connection template at this email. Just click Connect.
A*fter* connection you may use:

root
TpTy2222?

Start9 OS pass:
TpTyTpTyTpTy2222?

I attached private Key,one of them can be used

marssystems, 2023-10-06:
I don't see the RDP template anywhere.

consultingspeed, 2023-10-06:
Please see attached Default.rdp

marssystems, 2023-10-06:
I don't see anything attached.

consultingspeed, 2023-10-06:
https://drive.google.com/file/d/1O7WoCHgAcAmrugegzu4zOIkDkirwzXdP/view?usp=drivesdk

vin., 6 oct. 2023, 14:39 Mariusz Adamczyk ***@***.***> a
scris:

marssystems, 2023-10-06:
Ok - managed to log in.

What now?

I don't see any way to log into your RaspiBlitz.
How did you set it up in the first place?

consultingspeed, 2023-10-06:
Its installed on Oracle VM Virtualbox

vin., 6 oct. 2023, 15:03 Mariusz Adamczyk ***@***.***> a
scris:

marssystems, 2023-10-06:
How did you log in to configure the RaspiBlitz node?

consultingspeed, 2023-10-06:
Bitcoin credentials can be seen on Oracle Vm VirtualMachine > About , there
you may see the credentials, SSH access, etc
On Orache Vm Vicrtual machine  Settings > Network and you may change IP to
bridge network

All other application get the credentials automatically from StartOS were
are installed Bitcoin Node, LND etc, at everyone you may see credentials in
about at every app

marssystems, 2023-10-06:
Yes I see all that - but where is RaspiBlitz?

What is your password for RTL?

Did you set up the 3 passwords A, B and C when you installed RaspiBlitz?

consultingspeed, 2023-10-06:
password for RTL you find it on click on RTL app and there is a menu about
credentials under the logo

marssystems, 2023-10-06:
OK - Found it.

What about the 3 passwords for RaspiBlitz?

consultingspeed, 2023-10-06:
One of them are seen at Start9 > About,
other ones are Bitcoin core > Credentials,
LND > credentials, etc
you just click on every App, and you find them bellow the app logo

[ A ] Master user password

One of them are seen at Start9 > About,
[ B ] Bitcoin RPC password

other ones are Bitcoin core > Credentials,
[ C ] LND wallet password

LND > credentials, etc
[ D ] BTC-RPC-Explorer password (optional)
[ E ] Ride The Lightning password

All o

They were added automatically at install

consultingspeed, 2023-10-06:
> How did you install RaspiBlitz?
>
> Tell me step by step.

first I created a virtualized machine, cloned the disk and created an image
of installed Debian with RDP access
after install I installed Orache VirtualMachine and created a disk to
install Start9 iso image using Oracle Virtual Macine installed on the
debian server

Something has happened wrong?

consultingspeed, 2023-10-06:
> So when did you install the RaspiBlitz-1.10.0.img file ?
>
> You don't need to install Debian - RaspiBlitz has its own OS - Rasbian

I installed Start9 Operating System last 2-3 weeks ago with all the
requested applications which contains Bitcoin Core and LND
I installed Debian first to install Oracle VM to open the .img from Starto
OS

marssystems, 2023-10-06:
So when did you install RaspiBlitz?

consultingspeed, 2023-10-06:
I have installed in on 2 Octomber 2023

consultingspeed, 2023-10-06:
this server, but the image for longer

marssystems, 2023-10-06:
So why is RaspiBlitz not on the list os Services?

consultingspeed, 2023-10-07:
Dear Mariusz,

How are you? The script has been installed? What to do?

I got an information that "I generally try to stay away from Raspis if you
plan on something more “serious” with your node like committing some funds
to run Lightning" I got this info from Their telegram group
https://t.me/start9_labs

vin., 6 oct. 2023, 18:02 Mariusz Adamczyk ***@***.***> a
scris:

marssystems, 2023-10-07:
If you have installed the RaspiBlitz software on your VM - SSH into your node and follow the setup instructions on the RaspiBlitz Github website.

Most of the 15000 lightning nodes run on a Raspberry Pi 4B with no problems. I have been running 2 nodes for 5 years.

Check out - https://1ml.com/

[28 more replies]