raspiblitz/raspiblitz #4126: v1.10.0rc3 build_sdcard.sh: `apt update` fails on Debian12 bullseye - missing signature

steepdawn974, 2023-08-23:
Environment: Debian 12 bullseye (fresh install)
Script: `https://github.com/raspiblitz/raspiblitz/blob/accea759f6ee862ce2b02d65367c8b8267b18d2d/build_sdcard.sh`

Error: `GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E`

Full:
```
pi@tosh:~$ sudo apt update
[sudo] password for pi: 
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://security.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Err:4 http://archive.raspberrypi.org/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists... Done
W: GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'http://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

This is will cause the build_sdcard.sh to fail, in particular here:

`if [ "${baseimage}" = "raspios_arm64" ] || [ "${baseimage}" = "debian" ]; then`
https://github.com/raspiblitz/raspiblitz/blob/accea759f6ee862ce2b02d65367c8b8267b18d2d/build_sdcard.sh#L400

steepdawn974, 2023-08-23:
Adding the missing key manually from the Ubuntu keyring fixes it. 
 
>> https://forums.raspberrypi.com/viewtopic.php?t=338301#p2026829

However `apt-key` is deprecated and will stop working soon..

rootzoll, 2023-08-25:
@openoms is this something to update?

openoms, 2023-09-23:
This is fixed now in the updates since rc3.:
See:
https://github.com/raspiblitz/raspiblitz/blob/eb1969d26745230211a8c65b50d11f9f7b2f885b/build_sdcard.sh#L408-L409
Closing.

closed 2023-09-23 by openoms