raspiblitz/raspiblitz #4127: CLN v23.08 install fails

steepdawn974, 2023-08-24:
Trying to run: `config.scripts/cl.update.sh reckless` to upgrade to the lastest CLN version (23.08)

Results in this failure during compile:

```
# Building from source Core Lightning 
- Configuring EXPERIMENTAL_FEATURES enabled

Unknown option '--enable-experimental-features'
Usage: ./configure [--reconfigure] [setting=value] [options]
If --reconfigure is specified, config.vars will set defaults.

[...]

 Building Core lightning from source

File config.vars not found: you must run ./configure before running make.
make: *** No rule to make target 'config.vars', needed by 'show-flags'.  Stop.

HEAD is now at 98ad46795 CHANGELOG.md: update for 23.08 release final.
- Install to /usr/local/bin/
File config.vars not found: you must run ./configure before running make.
mkdir -p /bin
mkdir -p /libexec/c-lightning
mkdir -p /libexec/c-lightning/plugins
mkdir -p /share/man/man1
mkdir -p /share/man/man5
mkdir -p /share/man/man7
mkdir -p /share/man/man8
mkdir -p /share/doc/c-lightning
make: *** No rule to make target 'config.vars', needed by 'ccan/config.h'.  Stop.

```

rootzoll, 2023-08-25:
@openoms I get the same in branch `4127-updatecln` when I try to install cln with the new new version v23.08 .. here is a bit longer output:

```
  • Installing pyln-testing (23.05 /home/bitcoin/lightning/contrib/pyln-testing)
- Configuring EXPERIMENTAL_FEATURES enabled

Unknown option '--enable-experimental-features'
Usage: ./configure [--reconfigure] [setting=value] [options]
If --reconfigure is specified, config.vars will set defaults.
Default settings:
CSANFLAGS =
  CC (default cc)
  CWARNFLAGS (default -Wall -Wundef -Wmissing-prototypes -Wmissing-declarations -Wstrict-prototypes -Wold-style-definition -Werror -Wno-maybe-uninitialized -Wshadow=local)
  COPTFLAGS (default -Og)
  CDEBUGFLAGS (default -std=gnu11 -g -fstack-protector-strong)
  CONFIGURATOR_CC (default cc)
    To override compile line for configurator itself
  PYTEST (default python3 -m pytest)
  VALGRIND (default 0)
Options include:
  --prefix= (default /usr/local)
    Prefix for make install
  --enable/disable-developer (default disable)
    Developer mode, good for testing
  --enable/disable-debug (default disable)
    Extra debug checks in the build, good for testing
  --enable/disable-compat (default enable)
    Compatibility mode, good to disable to see if your software breaks
  --enable/disable-valgrind (default (autodetect))
    Run tests with Valgrind
  --enable/disable-static (default disable)
    Static link sqlite3 and zlib libraries
  --enable/disable-coverage (default disable)
    Compile with Clang coverage instrumentation
  --enable/disable-address-sanitizer (default disable)
    Compile with address-sanitizer
  --enable/disable-ub-sanitizer (default disable)
    Compile with undefined behaviour sanitizer
  --enable/disable-fuzzing (default disable)
    Compile with fuzzing
  --enable/disable-rust (default disable)
    Compile with Rust support

- Building Core lightning from source

File config.vars not found: you must run ./configure before running make.
make: *** No rule to make target 'config.vars', needed by 'show-flags'.  Stop.

HEAD is now at 98ad46795 CHANGELOG.md: update for 23.08 release final.
- Install to /usr/local/bin/
File config.vars not found: you must run ./configure before running make.
mkdir -p /bin
mkdir -p /libexec/c-lightning
mkdir -p /libexec/c-lightning/plugins
mkdir -p /share/man/man1
mkdir -p /share/man/man5
mkdir -p /share/man/man7
mkdir -p /share/man/man8
mkdir -p /share/doc/c-lightning
make: *** No rule to make target 'config.vars', needed by 'ccan/config.h'.  Stop.
```

wchancao, 2023-08-26:
Maybe this is the reason, from the v23.08 release highlights : `Experimental builds are no longer required: all experimental features are now explicit runtime options.`

I can try updating CLN without the --enable-experimental-features in the cl.install.sh script

rootzoll, 2023-08-26:
Yes without `--enable-experimental-features` it builds thru. Thx.

PR #4134 (2023-08-27): Update & fix configure of CLN v23.08
fixes #4127
removed `--enable-experimental-features` flag

@openoms please test if now anything important is missing
closed 2023-09-14 by rootzoll