raspiblitz/raspiblitz #4136: After update v1.9 0.16.2 blockchain sync stopped every few days  !!

Sporty999, 2023-08-27:
After update v1.9 LND0.16.2 blockchain sync stopped every few days  !!
freeze not ssh login , after restart sync is ok ! LND is ok and in a few days the same !

Thank you !

Generating debug logs. Be patient, this should take maximum 2 minutes ..
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.9.0
chainnetwork: bitcoin / main
 19:26:48 up  1:04,  3 users,  load average: 2.19, 2.20, 2.55

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-08-27 18:25:39 BST; 1h 1min ago
    Process: 6481 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6492 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6495 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6527 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 23min 9.691s
     CGroup: /system.slice/bitcoind.service
             └─6527 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Aug 27 18:23:35 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 27 18:25:39 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Tue 2023-08-01 18:01:39 BST, ends at Sun 2023-08-27 19:26:48 BST. --
Aug 27 18:23:35 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 27 18:25:39 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-08-27T17:44:57Z loadblk thread exit
2023-08-27T17:45:05Z txindex is enabled at height 805040
2023-08-27T17:45:05Z txindex thread exit
2023-08-27T17:45:11Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-08-27T17:46:00Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-08-27T17:46:04Z UpdateTip: new best=00000000000000000001ac1801568805c4e0954a213aaafbd98d0ea317d64a73 height=805041 version=0x20a26000 log2_work=94.383411 tx=885219005 date='2023-08-27T11:59:10Z' progress=0.999931 cache=151.1MiB(1096373txo)
2023-08-27T17:46:04Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-08-27T17:46:11Z UpdateTip: new best=00000000000000000004426ae3679d1d027f485b48a644130f6a22385dbde3dc height=805042 version=0x2965c000 log2_work=94.383424 tx=885220996 date='2023-08-27T12:03:58Z' progress=0.999932 cache=150.9MiB(1094931txo)
2023-08-27T17:46:19Z UpdateTip: new best=00000000000000000000d8dbe0c1d4bf22125a2227d168c60d4573a9b1a75c2d height=805043 version=0x20000000 log2_work=94.383437 tx=885227929 date='2023-08-27T12:06:54Z' progress=0.999933 cache=150.8MiB(1097487txo)
2023-08-27T17:46:41Z UpdateTip: new best=000000000000000000006ce538023e1e9e01351488ad9af66569d760d007a43a height=805044 version=0x271d2000 log2_work=94.383451 tx=885231428 date='2023-08-27T12:20:21Z' progress=0.999935 cache=152.5MiB(1109847txo)
2023-08-27T17:47:01Z UpdateTip: new best=000000000000000000024864379d30c71c29b6ba947e935c2f80175929fedc39 height=805045 version=0x236fe000 log2_work=94.383464 tx=885235982 date='2023-08-27T12:29:03Z' progress=0.999937 cache=153.1MiB(1115210txo)
2023-08-27T17:47:09Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-08-27T17:47:18Z UpdateTip: new best=00000000000000000001a2df44a0dfe0259433efe6a1274a8608b7eec09b6b44 height=805046 version=0x2a652000 log2_work=94.383477 tx=885239728 date='2023-08-27T12:32:54Z' progress=0.999938 cache=154.7MiB(1125910txo)
2023-08-27T17:47:23Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-08-27T17:47:39Z UpdateTip: new best=000000000000000000016611f67f3f8ac636389ce96a3beccf5dc48b865787da height=805047 version=0x26f58000 log2_work=94.383491 tx=885244651 date='2023-08-27T12:39:32Z' progress=0.999939 cache=155.6MiB(1132726txo)
2023-08-27T17:48:04Z UpdateTip: new best=000000000000000000027f247250c6e740f775ea32ee69cf0e19204fefb9206c height=805048 version=0x20000000 log2_work=94.383504 tx=885250952 date='2023-08-27T12:40:33Z' progress=0.999939 cache=156.5MiB(1141707txo)
2023-08-27T17:48:26Z UpdateTip: new best=00000000000000000002be6a1a2f56717785aff6815d8cdb2da20d6b8503c2fd height=805049 version=0x306de000 log2_work=94.383517 tx=885254987 date='2023-08-27T12:48:12Z' progress=0.999940 cache=157.1MiB(1147043txo)
2023-08-27T17:48:49Z UpdateTip: new best=000000000000000000050e59e0130fa890def94a72ae9ba5dec8d795f72c7c23 height=805050 version=0x20000000 log2_work=94.383531 tx=885261707 date='2023-08-27T12:49:27Z' progress=0.999941 cache=157.5MiB(1152550txo)
2023-08-27T17:49:12Z UpdateTip: new best=000000000000000000028ea630e2384d4a19a332991b62b1df9938abc17bf910 height=805051 version=0x2e8e4000 log2_work=94.383544 tx=885265642 date='2023-08-27T13:04:41Z' progress=0.999944 cache=158.1MiB(1157262txo)
2023-08-27T17:49:25Z UpdateTip: new best=0000000000000000000240e046f38b10e2f5c28d5238a3d9f45cfccabc9d0828 height=805052 version=0x2f52e000 log2_work=94.383557 tx=885268256 date='2023-08-27T14:24:41Z' progress=0.999959 cache=159.3MiB(1167573txo)
2023-08-27T17:49:40Z UpdateTip: new best=0000000000000000000508140fcae5e692c1106f8008b900e26737ba7112a4f7 height=805053 version=0x20000000 log2_work=94.383571 tx=885271253 date='2023-08-27T14:32:31Z' progress=0.999961 cache=160.0MiB(1172613txo)
2023-08-27T17:50:01Z UpdateTip: new best=0000000000000000000485b5efeb3058f5ac6203bff3a7efbacfec6c593d7f76 height=805054 version=0x20008000 log2_work=94.383584 tx=885275127 date='2023-08-27T14:34:34Z' progress=0.999961 cache=161.1MiB(1182072txo)
[... 402 lines cut ...]
btc_default_synced="1"
btc_default_blocks_data_kb="561515948"
btc_default_blocks_behind="0"
btc_default_sync_percentage="100.00"
btc_default_error_short=""
btc_default_sync_initialblockdownload="0"
btc_default_error_full=""
btc_default_ready="1"
btc_default_online="1"
btc_default_mempool_transactions="7643"
btc_default_sync_initial_done="1"
btc_default_blocks_verified="805077"
btc_default_peers="11"
btc_default_activated="1"
btc_default_sync_progress="0.9999985095907528"
btc_default_version="v24.0.1"
btc_default_running="1"
btc_default_blocks_headers="805077"

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

rootzoll, 2023-08-29:
It looks like you have touchscreen installed. its an experimental feature and recommended to be turned off if your system acts instable. Also please try the new RaspiBlitz v1.10.0rc4 - it has a more stable LND and lots of improvements to optimize on stability: https://github.com/raspiblitz/raspiblitz/tree/v1.10#downloading-the-software

Sporty999, 2023-09-06:
thx mate !

closed 2024-08-10 by openoms