raspiblitz/raspiblitz #4146: Remove from SSH menus BitcoinMinds/HomerDashboard

geco91, 2023-08-31:
As shown in the survey those two projects did not gained much traction in the raspiblitz user base and are not direct Bitcoin/Lightning tools - so to reduce complexity it recommended to at least remove them from the SSH menus and just keep them as "deprecated" ssh config scripts.

ChuckNorrison, 2023-09-04:
I would recommend extension based behaviour of the blitz menu. The menu could only show entries if the underlying script exists. So some could add a script as extension and than show the menu entry to install/configure.

Than raspiblitz could be delivered in a minimal state and get extended with more script files from a extensions menu.

Otherwise its hard to decide which feature should be dumped afterwards. I think no feature should be dumped after it was added to release version.

rootzoll, 2023-09-09:
An extensions menu sounds quite complicated - would need some kind of central extension repository server, etc .. I would recommed to keep this simple for now: 

The scripts itself can stay in the config scripts folder - maybe marked as deprecated/not-supported in the header/info section. This way people that really want to use them can do so. 

But config-scripts that can be triggered from the SSH menus signal to new user that this script is an active/maintained part of raspiblitz. So it makes sense to also remove such scripts from there.

rootzoll, 2023-11-22:
OK PR waiting to me merged.

closed 2024-02-25 by rootzoll