raspiblitz/raspiblitz #4149: 1.10.0rc5 v. : var/log fills 128GB SD Card in 2 days

salenjak, 2023-09-03:
As title says 1.10.0rc5 version have the issue that var/log fills SD Card in 2 days.
2 biggest files are syslog.1 and daemon.log.1, each 40GB at the moment.

Printing debug logs. Be patient, this should take maximum 2 minutes ..
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.10.0rc5
chainnetwork: bitcoin / main
 08:13:34 up 1 day,  7:49,  2 users,  load average: 5.98, 6.61, 6.89

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-09-02 00:26:20 BST; 1 day 7h ago
    Process: 7053 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 7081 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 7138 (bitcoind)
      Tasks: 19 (limit: 9352)
        CPU: 15h 40min 22.152s
     CGroup: /system.slice/bitcoind.service
             └─7138 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Warning: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Sun 2023-09-03 07:53:22 BST, ends at Sun 2023-09-03 08:13:34 BST. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-09-03T06:19:24Z Saw new header hash=000000000000000000007f2040f1cfbe4f20ff5926b2311a029db22358c58d48 height=805997
2023-09-03T06:19:24Z [net] Saw new cmpctblock header hash=000000000000000000007f2040f1cfbe4f20ff5926b2311a029db22358c58d48 peer=1702
2023-09-03T06:19:25Z UpdateTip: new best=000000000000000000007f2040f1cfbe4f20ff5926b2311a029db22358c58d48 height=805997 version=0x20000004 log2_work=94.396107 tx=888260570 date='2023-09-03T06:18:52Z' progress=1.000000 cache=89.9MiB(613241txo)
2023-09-03T06:19:25Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-09-03T06:19:25Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-09-03T06:20:00Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-09-03T06:23:44Z Saw new header hash=00000000000000000004009f07587f76bf521541baa0b5402a4a55bce443bd50 height=805998
2023-09-03T06:23:44Z [net] Saw new cmpctblock header hash=00000000000000000004009f07587f76bf521541baa0b5402a4a55bce443bd50 peer=1702
2023-09-03T06:23:45Z UpdateTip: new best=00000000000000000004009f07587f76bf521541baa0b5402a4a55bce443bd50 height=805998 version=0x20010000 log2_work=94.396121 tx=888265199 date='2023-09-03T06:23:26Z' progress=1.000000 cache=90.9MiB(619765txo)
2023-09-03T06:23:45Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-09-03T06:28:43Z New outbound peer connected: version: 70016, blocks=805998, peer=2051 (outbound-full-relay)
2023-09-03T06:28:47Z New outbound peer connected: version: 70016, blocks=805998, peer=2052 (block-relay-only)
2023-09-03T06:32:31Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-09-03T06:34:52Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-09-03T06:35:44Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-09-03T06:35:50Z New outbound peer connected: version: 70016, blocks=805998, peer=2059 (outbound-full-relay)
2023-09-03T06:36:22Z Saw new header hash=00000000000000000001346a1f949530641faace55501b32ac789cb587adf3d4 height=805999
2023-09-03T06:36:22Z [net] Saw new cmpctblock header hash=00000000000000000001346a1f949530641faace55501b32ac789cb587adf3d4 peer=1702
2023-09-03T06:36:23Z UpdateTip: new best=00000000000000000001346a1f949530641faace55501b32ac789cb587adf3d4 height=805999 version=0x22d34000 log2_work=94.396134 tx=888267270 date='2023-09-03T06:36:21Z' progress=1.000000 cache=92.0MiB(626936txo)
2023-09-03T06:36:23Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-09-03T06:39:35Z New outbound peer connected: version: 70016, blocks=805999, peer=2067 (block-relay-only)
2023-09-03T06:41:00Z Saw new header hash=000000000000000000047c45fecacb5b55d47a589ce0725d6363bb1ed3ec5798 height=806000
2023-09-03T06:41:00Z [net] Saw new cmpctblock header hash=000000000000000000047c45fecacb5b55d47a589ce0725d6363bb1ed3ec5798 peer=1702
2023-09-03T06:41:01Z UpdateTip: new best=000000000000000000047c45fecacb5b55d47a589ce0725d6363bb1ed3ec5798 height=806000 version=0x28c56000 log2_work=94.396147 tx=888273477 date='2023-09-03T06:40:47Z' progress=1.000000 cache=92.2MiB(630133txo)
2023-09-03T06:42:19Z Saw new header hash=0000000000000000000104ab3e736e6b65f21795da5467b51776cc1bce4bbe1b height=806001
2023-09-03T06:42:21Z UpdateTip: new best=0000000000000000000104ab3e736e6b65f21795da5467b51776cc1bce4bbe1b height=806001 version=0x27e9a004 log2_work=94.396160 tx=888279381 date='2023-09-03T06:41:42Z' progress=1.000000 cache=92.4MiB(633010txo)
2023-09-03T06:44:19Z Socks5() connect to ***.onion:8333 failed: host unreachable
[... 561 lines cut ...]
btc_default_synced="1"
btc_default_running="1"
btc_default_error_short=""
btc_default_error_full=""
btc_default_activated="1"
btc_default_blocks_headers="806005"
btc_default_mempool_transactions="166569"
btc_default_peers="21"
btc_default_sync_initial_done="1"
btc_default_sync_progress="0.9999969325251716"
btc_default_blocks_verified="806005"
btc_default_version="v25.0.0"
btc_default_port="8333"
btc_default_sync_initialblockdownload="0"
btc_default_ready="1"
btc_default_sync_percentage="100.00"
btc_default_blocks_data_kb="564359628"
btc_default_online="1"

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

salenjak, 2023-09-14:
I made a test several days ago by switching from CLN as default, to LND node. 
-------------------------------------
Test started 10.09.2023 with:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G   74G   39G  66% /
--------------------------------------
Today, 14.09.2023:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G   74G   39G  66% /
--------------------------------------

So, with LND enabled as the default lightning node my logs are not growing.

I will now switch to CLN again and report back in hour or two. I bet that logs will start growing again.

salenjak, 2023-09-14:
Only 1 hour after switching from LND to CLN as default implementation, the logs are 3GB larger. 

 Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G   77G   36G  68% /

salenjak, 2023-09-14:
Just to confirm that after 5 hours since I switched from CLN to LND as default, logs are not growing

openoms, 2023-09-19:
The root problem seems to be here that CLN is in DEBUG mode which is very noisy.
Also there are other services failing like LNDg.

The CLN log-level is set in the cl.conf to:
`log-level=info`
by default

but set to `debug` in recovery mode.
The problem could be that it was stuck there or changed manually.

We could add an extra check that the CLN logs are not left in debug mode.

Noting here that even with logrotate working the disk space might not be freed until restart:
https://serverfault.com/questions/501963/how-do-i-recover-free-space-on-deleted-files-without-restarting-the-referencing

Can try running this to check if that could be the issue:
```
# check free space
df -h
# Find all opened file descriptors, grep deleted, StdError to /dev/null
sudo find /proc/*/fd -ls 2> /dev/null | grep '(deleted)'
# Find and truncate all deleted files
sudo find /proc/*/fd -ls 2> /dev/null | awk '/deleted/ {print $11}' | xargs -n 1 sudo truncate -s 0
# check free space
df -h
```

openoms, 2023-09-20:
To try to reproduce switched the logging to debug for both LND and CLN on mainnet and signet and switched to CLN as default.
So far there is no high disk space usage:
```
₿ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G   17G   12G  60% /
devtmpfs        1.9G     0  1.9G   0% /dev
tmpfs           1.9G   36K  1.9G   1% /dev/shm
tmpfs           769M  1.2M  768M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            32M   28K   32M   1% /var/cache/raspiblitz
/dev/mmcblk0p1  255M   31M  225M  13% /boot
/dev/sda1       916G  658G  212G  76% /mnt/hdd
tmpfs           385M   16K  385M   1% /run/user/1000
tmpfs           385M   12K  385M   1% /run/user/1001

₿ sudo du -had1 /var/log
4.2M	/var/log/messages.1
1.2M	/var/log/auth.log.2.gz
4.0K	/var/log/sysstat
269M	/var/log/journal
312K	/var/log/messages.2.gz
56K	/var/log/ufw.log.2.gz
4.0K	/var/log/btmp
32K	/var/log/wtmp
312K	/var/log/kern.log.2.gz
40K	/var/log/user.log.1
8.0K	/var/log/cups
0	/var/log/ufw.log
85M	/var/log/daemon.log.1
12K	/var/log/postgresql
4.0K	/var/log/private
48K	/var/log/user.log.2.gz
4.0K	/var/log/mysql
4.0K	/var/log/tor
3.1M	/var/log/syslog.2.gz
24K	/var/log/nginx
8.0K	/var/log/redis
360K	/var/log/daemon.log.3.gz
12K	/var/log/runit
76K	/var/log/messages.3.gz
12K	/var/log/lastlog
16K	/var/log/lightdm
132K	/var/log/apt
2.5M	/var/log/daemon.log.2.gz
4.0K	/var/log/debug.2.gz
16K	/var/log/dpkg.log
4.0K	/var/log/fail2ban.log
89M	/var/log/syslog.1
9.4M	/var/log/i2pd
4.2M	/var/log/kern.log.1
4.0K	/var/log/debug.1
6.4M	/var/log/ufw.log.1
76K	/var/log/kern.log.3.gz
256K	/var/log/user.log.3.gz
20K	/var/log/auth.log.3.gz
4.0K	/var/log/dpkg.log.1
8.0K	/var/log/fail2ban.log.1
51M	/var/log/auth.log.1
812K	/var/log/syslog.3.gz
4.0K	/var/log/debug.3.gz
525M	/var/log

₿ sudo ls -lah /mnt/hdd/lnd/logs/bitcoin/lnd.log
ls: cannot access '/mnt/hdd/lnd/logs/bitcoin/lnd.log': No such file or directory
₿ sudo ls -lah /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
-rw-r--r-- 1 bitcoin bitcoin 4.2M Sep 20 11:12 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log

₿ sudo ls -lah /mnt/hdd/lnd/logs/bitcoin/signet/lnd.log
-rw-r--r-- 1 bitcoin bitcoin 7.5M Sep 19 18:00 /mnt/hdd/lnd/logs/bitcoin/signet/lnd.log
₿ admin@192.168.4.252:~ ₿ sudo ls -lah /home/bitcoin/.lightning/signet/cl.log
-rw-r--r-- 1 bitcoin bitcoin 22K Sep 19 18:00 /home/bitcoin/.lightning/signet/cl.log
```

0xtrr, 2023-09-20:
I am experiencing the same problem on Raspiblitz v1.9.0 where /var/log/daemon.log fills up pretty quick with the following stuff

```
*** BLITZAPI STATUS ***
installed=1
repo='https://github.com/fusion44/blitz_api.git'
branch='blitz-v1.9'
commit='9255a513cdf64fdc6a060a4fa67a8d9121e6474f'
● blitzapi.service - BlitzBackendAPI
     Loaded: loaded (/etc/systemd/system/blitzapi.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-09-20 18:30:58 BST; 16min ago
   Main PID: 749 (python)
      Tasks: 4 (limit: 9352)
        CPU: 3min 56.347s
     CGroup: /system.slice/blitzapi.service
             └─749 /home/blitzapi/blitz_api/venv/bin/python -m uvicorn app.main:app --port 11111 --host=*.*.*.* --root-path /api

Sep 20 18:47:49 raspberrypi python[749]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
Sep 20 18:47:49 raspberrypi python[749]:     return ctx.invoke(self.callback, **ctx.params)

*** LAST BLITZAPI LOGS ***
sudo journalctl -u blitzapi -b --no-pager -n20
-- Journal begins at Wed 2023-09-20 17:45:22 BST, ends at Wed 2023-09-20 18:47:54 BST. --
Sep 20 18:47:49 raspberrypi python[749]:     exec(code, run_globals)
Sep 20 18:47:49 raspberrypi python[749]:          │     └ {'__name__': '__main__', '__doc__': None, '__package__': 'uvicorn', '__loader__': <_frozen_importlib_external.SourceFileLoade...
Sep 20 18:47:49 raspberrypi python[749]:          └ <code object <module> at 0x7f8051a450, file "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__main__.py", ...
Sep 20 18:47:49 raspberrypi python[749]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__main__.py", line 4, in <module>
Sep 20 18:47:49 raspberrypi python[749]:     uvicorn.main()
Sep 20 18:47:49 raspberrypi python[749]:     │       └ <Command main>
Sep 20 18:47:49 raspberrypi python[749]:     └ <module 'uvicorn' from '/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__init__.py'>
Sep 20 18:47:49 raspberrypi python[749]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
Sep 20 18:47:49 raspberrypi python[749]:     return self.main(*args, **kwargs)
Sep 20 18:47:49 raspberrypi python[749]:            │    │     │       └ {}
Sep 20 18:47:49 raspberrypi python[749]:            │    │     └ ()
Sep 20 18:47:49 raspberrypi python[749]:            │    └ <function BaseCommand.main at 0x7f7fcc73a0>
Sep 20 18:47:49 raspberrypi python[749]:            └ <Command main>
Sep 20 18:47:49 raspberrypi python[749]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
Sep 20 18:47:49 raspberrypi python[749]:     rv = self.invoke(ctx)
Sep 20 18:47:49 raspberrypi python[749]:          │    │      └ <click.core.Context object at 0x7f805c0040>
Sep 20 18:47:49 raspberrypi python[749]:          │    └ <function Command.invoke at 0x7f7fcc58b0>
Sep 20 18:47:49 raspberrypi python[749]:          └ <Command main>
Sep 20 18:47:49 raspberrypi python[749]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
Sep 20 18:47:49 raspberrypi python[749]:     return ctx.invoke(self.callback, **ctx.params)
```

openoms, 2023-09-20:
@fusion44 is this something expected to.be solved in the `dev` / `v1.10` branch?

rootzoll, 2023-09-21:
But in general even if we have a spammy log process .. logrotate should handle this and should use the `copytruncate` option to avoid the open file reference problem:

> The copytruncate option in logrotate is used to avoid the issue of open file references when rotating log files. When copytruncate is used, logrotate will copy the current log file to a new file and then truncate the original file to zero bytes. This allows the process writing to the log file to continue writing to the same file, even though the file has been rotated. This option is useful when the process writing to the log file does not support reopening the log file, or when restarting the process is not desirable. 

`copytruncate` is not in the global settings like `/etc/logrotate.conf` yet.
@openoms is there any reason not to add this option?

openoms, 2023-09-21:
>copytruncate is not in the global settings like /etc/logrotate.conf yet.
@openoms is there any reason not to add this option?

no, it sounds good, will start testing

rootzoll, 2023-09-21:
added `copytruncate` to `/etc/logrotate.conf `& `/etc/logrotate.d/rsyslog`

openoms, 2023-09-21:
my logs did not grow significantly or rotated as needed.
The `copytruncate` entry does not cause any issues so looking good here.

0xtrr, 2023-09-21:
Anything I can do manually to mitigate this problem until the next update? I'm not able to run the node atm because of this issue. I tried adding the copytruncate entry to the two log files but nothing changed in terms of the log file size increase.

openoms, 2023-09-21:
@0xtrr to switch to the logrotate settings from the `dev` branch:
```
# switch to root
sudo su -
```
and paste this block to the terminal:
https://github.com/raspiblitz/raspiblitz/blob/a9f927238837354c55619d9af418cfe218758ba5/build_sdcard.sh#L505-L546

0xtrr, 2023-09-22:
> @0xtrr to switch to the logrotate settings from the `dev` branch:
> 
> ```
> # switch to root
> sudo su -
> ```
> 
> and paste this block to the terminal:
> 
> https://github.com/raspiblitz/raspiblitz/blob/a9f927238837354c55619d9af418cfe218758ba5/build_sdcard.sh#L505-L546

Tried adding this but it's still growing past 100MB before being rotated

[image]

openoms, 2023-09-22:
@0xtrr you should check the conf files edited if they are correct and also see if logrotate shows any errors:

```
sudo journalctl -fu logrotate
```

salenjak, 2023-09-23:
> Anything I can do manually to mitigate this problem until the next update? I'm not able to run the node atm because of this issue. I tried adding the copytruncate entry to the two log files but nothing changed in terms of the log file size increase.

I have disabled Blitz API + WebUI in settings to stop logs from growing.

closed 2023-09-28 by rootzoll