raspiblitz/raspiblitz #4154: Permission issues on 1.10.0rc5

talvasconcelos, 2023-09-07:
First of all, I'm running on a minipc (amd64 image). Managed to install Debian, ran `build_sdcard.sh` and after some issues with the blockchain and SSD, and a new IBD, the node is up and running! (I'll try and document this later)

I found a couple of issues, now the node is running (BTC and LND), which seem related to permissions:
- nginx: ```nginx -t
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
2023/09/07 09:49:42 [emerg] 145892#145892: open() "/run/nginx.pid" failed (13: Permission denied)
nginx: configuration file /etc/nginx/nginx.conf test failed```
- Although i could install thunderhub, can't access the webpage: ```sudo service thunderhub status
● thunderhub.service - ThunderHub daemon
     Loaded: loaded (/etc/systemd/system/thunderhub.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2023-09-07 09:50:49 EDT; 4s ago
    Process: 156363 ExecStart=/usr/bin/npm run start (code=exited, status=127)
   Main PID: 156363 (code=exited, status=127)```
- RTL, fails on install: ```# Get the RTL Source Code
Cloning into '/home/rtl/RTL'...
remote: Enumerating objects: 37247, done.
remote: Counting objects: 100% (1607/1607), done.
remote: Compressing objects: 100% (628/628), done.
remote: Total 37247 (delta 1109), reused 1443 (delta 976), pack-reused 35640
Receiving objects: 100% (37247/37247), 274.56 MiB | 23.12 MiB/s, done.
Resolving deltas: 100% (30716/30716), done.
/home/admin/config.scripts/bonus.rtl.sh: line 146: cd: /home/rtl/RTL: Permission denied
fatal: not a git repository (or any of the parent directories): .git
sudo: unable to execute /home/admin/config.scripts/blitz.git-verify.sh: Permission denied```
- managed to install LIT, but can't login as it says wrong password
- webUI also says wrong password, or doesn't open at all

openoms, 2023-09-07:
Seen similar happen with one of the latest premade amd64 image also, but I was mounting a ZFS pool and did not have time to investigate yet.

If you used the latest Debian 12.1 image as a base it might be the common thing.

talvasconcelos, 2023-09-07:
Yes latest Debian.

Process was : 
1. install debian on SSD
2. ssh into PC with pi / raspiblitz
3. run `sudo build_sdcard.sh .....`
4. reboot
5. ssh with admin / raspiblitz
6. ...

openoms, 2023-09-08:
Can reproduce on :
```
uname -a                                                                                  
Linux raspiblitz-amd64 6.1.0-11-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.38-4 (2023-08-08) x86_64 GNU/Linux 
```

```
 ₿ config.scripts/bonus.rtl.sh on
# Running: 'bonus.rtl.sh on' 
# LNTYPE(lnd)
# CHAIN(mainnet)
# netprefix()
# portprefix()
# typeprefix()
# RTLHTTP(3000)
# configEntry(rtlWebinterface)
# systemdService(RTL)
# install of codebase is needed first
# Running: 'bonus.rtl.sh install'
# LNTYPE(lnd)
# CHAIN(mainnet)
# netprefix()
# portprefix()
# typeprefix()
# RTLHTTP(3000)
# configEntry(rtlWebinterface)
# systemdService(RTL)
# Installing RTL codebase
v20.5.1
nodeJS 20 is already installed
# Get the RTL Source Code
Cloning into '/home/rtl/RTL'...
remote: Enumerating objects: 37247, done.
remote: Counting objects: 100% (1607/1607), done.
remote: Compressing objects: 100% (631/631), done.
remote: Total 37247 (delta 1109), reused 1441 (delta 973), pack-reused 35640
Receiving objects: 100% (37247/37247), 274.57 MiB | 7.34 MiB/s, done.
Resolving deltas: 100% (30646/30646), done.
/home/admin/config.scripts/bonus.rtl.sh: line 146: cd: /home/rtl/RTL: Permission denied
fatal: not a git repository (or any of the parent directories): .git
sudo: unable to execute /home/admin/config.scripts/blitz.git-verify.sh: Permission denied
```

openoms, 2023-09-08:
The problem is because in the difference in:
`/etc/adduser.conf`
Debian 11:
```
# If DIR_MODE is set, directories will be created with the specified
# mode. Otherwise the default mode 0755 will be used.
DIR_MODE=0755
```

Debian 12:
```
# The permissions mode for home directories of non-system users.
# Default: DIR_MODE=0700
#DIR_MODE=0700

# The permissions mode for home directories of system users.
# Default: SYS_DIR_MODE=0755 
#SYS_DIR_MODE=0755

```

talvasconcelos, 2023-09-08:
so `DIR_MODE` should be uncommented and set to `0755` ?

openoms, 2023-09-08:
> so `DIR_MODE` should be uncommented and set to `0755` ?

yes, that should work or we need to change the `adduser` commands to (eg for a user: `test`):
```
sudo adduser --system --group --home /home/test test
```

openoms, 2023-09-10:
@talvasconcelos this should be solved if you build from the `dev` branch again

talvasconcelos, 2023-09-10:
Will try and report back. This time should be easy to plug the data SSD
back in and resume everything!

I tried making the hack without starting over but it doesn't work... admin
user was built without it in the first place!

Thank you for your help!

talvasconcelos, 2023-09-12:
BTW, i found a new quirk, on Debian 12, or maybe it's Python11...

```error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.```

It will only work with: `--break-system-packages` , this breaks installing a few bonus apps

talvasconcelos, 2023-09-12:
From previous install, i wanted to install with the permissions on home folders. Before udating, i did the prepare update/release, backed up the SCB.

Then:
1. installed a fresh debian 
2. reboot and ssh'd with pi / raspiblitz
3. ran `sudo bash build_sdcard.sh -f true -b dev -d headless -t false -w off` from `dev`
4. reboot with data SSD pluged in, and the blitz keeps rebooting after a few seconds
5. if i unplug the data SSD, i can ssh into the pc, no reboots

talvasconcelos, 2023-09-12:
`sudo tail raspiblitz.log` outputs this, when i connect the SSD, before reboot! If it helps... (something wrong with the lcd/headless thing?!)

```isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate: sdb
HDD/SSD connected: sdb
No HDMI switch found.
No FS EXPAND needed. needsExpansion(0) fsexpanded(0)
No WIFI RESTORE because no /var/cache/raspiblitz/hdd-inspect/wpa_supplicant.conf
SSH SERVER CERTS RESTORE activating old SSH host keys
# *** /home/admin/config.scripts/blitz.ssh.sh restore
# restore sshd host keys from: /var/cache/raspiblitz/hdd-inspect/sshd/ssh/sshd
# OK - sshd keys restore done
# INFO - ssh-root keys backup not available
No SSHRESET switch found.
check that display class in raspiblitz.conf from HDD is different from as it is now in raspiblitz.info ...
infoFileDisplayClass(headless)
confFileDisplayClass(lcd)
DISPLAY RESTORE - need to update displayClass from (headless) to (lcd)'
# blitz.display.sh set-display lcd
baseimage(debian)
# old(headless)
# new(lcd)
err='baseimage not supported'
No UASP FIX needed

Broadcast message from root@sparkblitz (Tue 2023-09-12 14:10:58 EDT):

The system will reboot now!

Broadcast message from root@sparkblitz (Tue 2023-09-12 14:10:58 EDT):

The system will reboot now!

Connection to 192.168.1.96 closed by remote host.
Connection to 192.168.1.96 closed.
```

talvasconcelos, 2023-09-12:
Ok, defenittely a mismatch on `raspiblitz.conf` in the SSD that was triggering the reboots!

talvasconcelos, 2023-09-12:
Re-running build_sdcard srews the install, seems that the checks to see if certain things are in place, skips some steps... i don't have now `bitcoind.service` for example! command aliases are not working... maybe there should be a way to reset everything and redo again!

tomorrow i'll try with fresh debian install

rootzoll, 2023-09-20:
merged #4153 for v1.10.0 release which shjould address this issue

openoms, 2023-09-23:
Confirmed fixed on v1.10rc6 after recovery:
```
 ₿ sudo ls -la /home
total 64
drwxr-xr-x 16 root           root           4096 Sep 22 21:30 .
drwxr-xr-x 18 root           root           4096 Sep 22 20:54 ..
drwxr-xr-x 12 admin          admin          4096 Sep 23 06:58 admin
drwxr-xr-x 10 bitcoin        bitcoin        4096 Sep 22 21:19 bitcoin
drwxr-xr-x  6 blitzapi       blitzapi       4096 Sep 22 00:28 blitzapi
drwxr-xr-x  4 bos            bos            4096 Sep 23 07:08 bos
drwxr-xr-x 11 btcpay         btcpay         4096 Sep 23 06:58 btcpay
drwxr-xr-x  6 btcrpcexplorer btcrpcexplorer 4096 Sep 22 01:09 btcrpcexplorer
drwxr-xr-x  5 jam            jam            4096 Sep 22 00:47 jam
drwxr-xr-x  9 joinmarket     joinmarket     4096 Sep 23 07:07 joinmarket
drwxr-xr-x  5 lnbits         lnbits         4096 Sep 22 00:32 lnbits
drwxr-xr-x  5 mempool        mempool        4096 Sep 22 01:10 mempool
drwxr-xr-x 14 pi             pi             4096 Sep 21 22:58 pi
drwxr-xr-x  5 rtl            rtl            4096 Sep 22 21:20 rtl
drwxr-xr-x  5 specter        specter        4096 Sep 22 21:24 specter
drwxr-xr-x  7 thunderhub     thunderhub     4096 Sep 22 01:04 thunderhub
```

closed 2023-09-23 by openoms