raspiblitz/raspiblitz #4159: Freezing during blockchain sync (both Pi 3 and Pi 4)

benthetechguy, 2023-09-10:
Same exact problem as #3024 but with a 4 GB RAM RPi 4. I have tested two different drives on both an RPi 4 and an RPi 3, on both the minimal and fat images. Every single time, after several hours of syncing, the Pi freezes and I can't connect via SSH until I reboot and start the cycle again.

openoms, 2023-09-10:
can you post the screenshot of htop to check if ZRAM is active.
Also you can activate it if not:
`config.scripts/blitz.zram.sh on`

benthetechguy, 2023-09-10:
ZRAM was not enabled, and now it is. Isn't the default behavior supposed 
to be ZRAM enabled so that #3024 doesn't happen again?

I will update you on whether it freezes again in the next few hours.

benthetechguy, 2023-09-11:
Things are going well so far.

closed 2023-09-13 by rootzoll

benthetechguy, 2023-09-13:
Is ZRAM enabled by default? If it's not, it should be. If it is, why wasn't it running on either of my Pis?

rootzoll, 2023-09-13:
With v1.10.0 we corrected a bug that didnt started zram by default in v1.9.0

benthetechguy, 2023-09-14:
Ah, okay. I was using 1.9 since it's the latest release for download.