raspiblitz/raspiblitz #4169: [amd64] fatpack option fails with: cchardet: finished with status 'error' on blitzapi install

openoms, 2023-09-13:
From the failed build: https://github.com/raspiblitz/raspiblitz/actions/runs/6139119894/job/16656697284?pr=4160

```
 qemu.debian: * Adding Raspiblitz API ...
    qemu.debian: # GITHUB_USER(fusion44)
    qemu.debian: GITHUB_REPO(blitz_api)
    qemu.debian: GITHUB_BRANCH(blitz-v1.10)
    qemu.debian: GITHUB_COMMITORTAG()
    qemu.debian: # INFO: No GITHUB_COMMITORTAG provided .. will use latest code on branch
    qemu.debian: # INSTALL Web API ...
    qemu.debian: Adding system user `blitzapi' (UID 112) ...
    qemu.debian: Adding new group `blitzapi' (GID 117) ...
    qemu.debian: Adding new user `blitzapi' (UID 112) with group `blitzapi' ...
    qemu.debian: Creating home directory `/home/blitzapi' ...
    qemu.debian: # allowing user as part of the bitcoin group to RW RPC hook
==> qemu.debian: chmod: cannot access '/home/bitcoin/.lightning/bitcoin': No such file or directory
==> qemu.debian: chmod: cannot access '/home/bitcoin/.lightning/bitcoin/lightning-rpc': No such file or directory
==> qemu.debian: cat: /home/bitcoin/.lightning/config: No such file or directory
==> qemu.debian: tee: /home/bitcoin/.lightning/config: No such file or directory
    qemu.debian: rpc-file-mode=0660
    qemu.debian: # clone github: fusion44/blitz_api
==> qemu.debian: Cloning into 'blitz_api'...
    qemu.debian: # checkout branch: blitz-v1.10
==> qemu.debian: Switched to a new branch 'blitz-v1.10'
    qemu.debian: branch 'blitz-v1.10' set up to track 'origin/blitz-v1.10'.
    qemu.debian: # using lastest code in branch
    qemu.debian: # running install
    qemu.debian: Ignoring cffi: markers 'python_version >= "3.9" and python_version < "4.0" and implementation_name == "pypy"' don't match your environment
    qemu.debian: Ignoring colorama: markers 'python_version >= "3.9" and python_version < "4.0" and (sys_platform == "win32" or platform_system == "Windows")' don't match your environment
    qemu.debian: Ignoring exceptiongroup: markers 'python_version >= "3.9" and python_version < "3.11"' don't match your environment
    qemu.debian: Ignoring pycparser: markers 'python_version >= "3.9" and python_version < "4.0" and implementation_name == "pypy"' don't match your environment
    qemu.debian: Ignoring win32-setctime: markers 'python_version >= "3.9" and python_version < "4.0" and sys_platform == "win32"' don't match your environment
    qemu.debian: Collecting aiohttp==3.8.3
    qemu.debian:   Downloading aiohttp-3.8.3-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.0 MB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.0/1.0 MB 2.0 MB/s eta 0:00:00
    qemu.debian: Collecting aiojobs==1.1.0
    qemu.debian:   Downloading aiojobs-1.1.0-py3-none-any.whl (12 kB)
    qemu.debian: Collecting aioredis==2.0.1
    qemu.debian:   Downloading aioredis-2.0.1-py3-none-any.whl (71 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 71.2/71.2 kB 1.4 MB/s eta 0:00:00
    qemu.debian: Collecting aiosignal==1.3.1
    qemu.debian:   Downloading aiosignal-1.3.1-py3-none-any.whl (7.6 kB)
    qemu.debian: Collecting anyio==3.7.0
    qemu.debian:   Downloading anyio-3.7.0-py3-none-any.whl (80 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 80.9/80.9 kB 1.9 MB/s eta 0:00:00
    qemu.debian: Collecting async-timeout==4.0.2
    qemu.debian:   Downloading async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
    qemu.debian: Collecting attrs==23.1.0
    qemu.debian:   Downloading attrs-23.1.0-py3-none-any.whl (61 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.2/61.2 kB 1.3 MB/s eta 0:00:00
    qemu.debian: Collecting cchardet==2.1.7
    qemu.debian:   Downloading cchardet-2.1.7.tar.gz (653 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 653.6/653.6 kB 3.1 MB/s eta 0:00:00
    qemu.debian:   Preparing metadata (setup.py): started
    qemu.debian:   Preparing metadata (setup.py): finished with status 'done'
    qemu.debian: Collecting certifi==2023.5.7
    qemu.debian:   Downloading certifi-2023.5.7-py3-none-any.whl (156 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 157.0/157.0 kB 1.9 MB/s eta 0:00:00
    qemu.debian: Collecting charset-normalizer==2.1.1
    qemu.debian:   Downloading charset_normalizer-2.1.1-py3-none-any.whl (39 kB)
    qemu.debian: Collecting click==8.1.3
    qemu.debian:   Downloading click-8.1.3-py3-none-any.whl (96 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.6/96.6 kB 2.0 MB/s eta 0:00:00
    qemu.debian: Collecting deepdiff==5.8.1
    qemu.debian:   Downloading deepdiff-5.8.1-py3-none-any.whl (69 kB)
    qemu.debian:      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 69.5/69.5 kB 1.8 MB/s eta 0:00:00
    qemu.debian: Collecting fastapi-plugins==0.11.0
    qemu.debian:   Downloading fastapi_plugins-0.11.0-py3-none-any.whl (23 kB)
    qemu.debian: Collecting fastapi==0.94.0
    qemu.debian:   Downloading fastapi-0.94.0-py3-none-any.whl (56 kB)
[... 119 lines cut ...]
==> qemu.debian: note: This is an issue with the package mentioned above, not pip.
==> qemu.debian: hint: See above for output from the failure.
    qemu.debian: error='pip install failed'
Error: /10 23:46:40 packer-plugin-qemu_v1.0.9_x5.0_linux_amd64 plugin: 2023/09/10 23:46:40 [ERROR] Remote command exited with '1': echo 'raspiblitz' | HOME_DIR='/home/pi' PACKER_BUILDER_TYPE='qemu' PACKER_BUILD_NAME='debian' PACKER_HTTP_ADDR='10.0.2.2:8722' PACKER_HTTP_IP='10.0.2.2' PACKER_HTTP_PORT='8722' branch='fix-amd64-fatpack' desktop='none' github_user='raspiblitz' pack='fatpack'  sudo -S -E sh -eux '/tmp/script_2193.sh'
2023/09/10 23:46:40 packer-plugin-qemu_v1.0.9_x5.0_linux_amd64 plugin: 2023/09/10 23:46:40 [INFO] RPC endpoint: Communicator ended with: 1
2023/09/10 23:46:40 [INFO] 342625 bytes written for 'stdout'
2023/09/10 23:46:40 [INFO] 414545 bytes written for 'stderr'
2023/09/10 23:46:40 [INFO] RPC client: Communicator ended with: 1
2023/09/10 23:46:40 [INFO] RPC endpoint: Communicator ended with: 1
2023/09/10 23:46:40 packer-provisioner-shell plugin: [INFO] 342625 bytes written for 'stdout'
2023/09/10 23:46:40 packer-provisioner-shell plugin: [INFO] 414545 bytes written for 'stderr'
2023/09/10 23:46:40 packer-provisioner-shell plugin: [INFO] RPC client: Communicator ended with: 1
2023/09/10 23:46:40 [INFO] (telemetry) ending shell
==> qemu.debian: Provisioning step had errors: Running the cleanup provisioner, if present...
2023/09/10 23:46:40 packer-plugin-qemu_v1.0.9_x5.0_linux_amd64 plugin: 2023/09/10 23:46:40 failed to unlock port lockfile: close tcp 127.0.0.1:5900: use of closed network connection
2023/09/10 23:46:40 packer-plugin-qemu_v1.0.9_x5.0_linux_amd64 plugin: 2023/09/10 23:46:40 failed to unlock port lockfile: close tcp 127.0.0.1:2533: use of closed network connection
==> qemu.debian: Deleting output directory...
2023/09/10 23:46:40 [INFO] (telemetry) ending qemu.debian
==> Wait completed after 3 hours 59 minutes
Build 'qemu.debian' errored after 3 hours 59 minutes: Script exited with non-zero exit status: 1. Allowed exit codes are: [0]
```

Reproduced on bare metal Debian 12 by @talvasconcelos running the `build_sdcard.sh`:
[image]
[image]

fusion44, 2023-09-13:
Seems like a python 3.11 problem. It is odd though... I'm currently running 3.10 but tested the API with 3.11 with no issue found.

https://stackoverflow.com/questions/74979674/gensim-install-in-python-3-11-fails-because-of-missing-longintrepr-h-file

openoms, 2023-09-18:
See the comment: https://github.com/raspiblitz/raspiblitz/pull/4160#issuecomment-1722448676

rootzoll, 2023-11-14:
@openoms can be changed to "final testing"?

openoms, 2023-11-15:
Fixes in #4242 and #4160. Due to test the fatpack build locally.

openoms, 2024-01-23:
will need to check if the fixes are still relevant / needed ad cchardet was removed from the blitz API

rootzoll, 2024-03-19:
@openoms were you able to test this. Can we close the issue?

openoms, 2024-03-19:
yes, there is no cchardet dependency any more

closed 2024-03-19 by openoms