raspiblitz/raspiblitz #4170: Python 3.11 error on Debian 12: This environment is externally managed

openoms, 2023-09-13:
Mentioned in:
https://github.com/raspiblitz/raspiblitz/issues/4154#issuecomment-1716008904
and
https://github.com/raspiblitz/raspiblitz/pull/4153

```

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.```
```

Possible workarounds are discussed extensively in:
https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3
and in: https://stackoverflow.com/questions/75602063/pip-install-r-requirements-txt-is-failing-this-environment-is-externally-mana

Either use in the pip config:
```
[global]
break-system-packages = true
```
or
```
sudo rm /usr/lib/python3.*/EXTERNALLY-MANAGED
```
to the same effect.

rootzoll, 2023-09-20:
merged #4153 for v1.10.0 release which shjould address this issue

openoms, 2023-09-23:
@rootzoll #4153 did not fix this, but since it is only relevant for Python 3.11 in Debian 12 it should not delay the v1.10 release. Tagged for v0.10.1.

meshack-ai, 2023-09-30:
Thank you  problem solved

samuelsumbane, 2023-11-03:
Thank very much!

rootzoll, 2023-12-05:
@openoms what is the status here on the v1.11.0 release?

saimonbrito, 2024-01-21:
o brigado o meu tambem resolvi com sudo rm /usr/lib/python3.*/EXTERNALLY-MANAGED

rootzoll, 2024-02-10:
@openoms not sure if #4400 broke the python lnd scripts .. having problems with lnd after the. merge. Have to investigate further next week.

rootzoll, 2024-02-11:
On a sd card build with the #4400 merged the setup with a fresh lnd failed:

When ssh in and trying `./config.scripts/lnd.initwallet.py` I get:
```
Traceback (most recent call last):
  File "/home/admin/./config.scripts/lnd.initwallet.py", line 6, in <module>
    import grpc
ModuleNotFoundError: No module named 'grpc'
```
@openoms When checking ´pip list | grep grpc´ there is no `grpcio 1.42.0` modules listed.

openoms, 2024-02-11:
@rootzoll I think it would be safer to reintroduce the fix at this stage, there are too many python dependencies installed in various places.

openoms, 2024-02-11:
just checked the logs of the builld: https://pipelinesghubeus12.actions.githubusercontent.com/IYvj0uVSXPPHLO7ttdl6qtHJCXGXn6RaqGzAAHVqqz5E8aky5C/_apis/pipelines/1/runs/905/signedlogcontent/2?urlExpires=2024-02-11T18%3A14%3A01.7455631Z&urlSigningMethod=HMACV1&urlSignature=kSPaVnzdl60iRyxjuaYCs9opekNg6nDF3rEavwBjkII%3D

It has the same error:
```
2024-02-09T15:36:20.4345862Z [0;32m    arm.raspiblitz-arm64-rpi: *** Python DEFAULT libs & dependencies ***[0m
2024-02-09T15:36:20.4692062Z [0;32m    arm.raspiblitz-arm64-rpi: update-alternatives: using /usr/bin/python3.11 to provide /usr/bin/python (python) in auto mode[0m
2024-02-09T15:36:20.5177940Z [0;32m    arm.raspiblitz-arm64-rpi: python calls python3.11[0m
2024-02-09T15:36:20.5518559Z [0;32m    arm.raspiblitz-arm64-rpi: update-alternatives: using /usr/bin/pip3 to provide /usr/bin/pip (pip) in auto mode[0m
2024-02-09T15:36:20.5525947Z [1;31m==> arm.raspiblitz-arm64-rpi: update-alternatives: warning: not replacing /usr/bin/pip with a link[0m
2024-02-09T15:36:20.6964487Z [1;31m==> arm.raspiblitz-arm64-rpi: sudo: unable to resolve host 3a42a9873732: Name or service not known[0m
2024-02-09T15:36:22.5829092Z [1;31m==> arm.raspiblitz-arm64-rpi: error: externally-managed-environment[0m
2024-02-09T15:36:22.5831925Z [1;31m==> arm.raspiblitz-arm64-rpi:[0m
2024-02-09T15:36:22.5836675Z [1;31m==> arm.raspiblitz-arm64-rpi: × This environment is externally managed[0m
2024-02-09T15:36:22.5842749Z [1;31m==> arm.raspiblitz-arm64-rpi: ╰─> To install Python packages system-wide, try apt install[0m
2024-02-09T15:36:22.5848041Z [1;31m==> arm.raspiblitz-arm64-rpi:     python3-xyz, where xyz is the package you are trying to[0m
2024-02-09T15:36:22.5855447Z [1;31m==> arm.raspiblitz-arm64-rpi:     install.[0m
2024-02-09T15:36:22.5861418Z [1;31m==> arm.raspiblitz-arm64-rpi:[0m
2024-02-09T15:36:22.5867587Z [1;31m==> arm.raspiblitz-arm64-rpi:     If you wish to install a non-Debian-packaged Python package,[0m
2024-02-09T15:36:22.5870913Z [1;31m==> arm.raspiblitz-arm64-rpi:     create a virtual environment using python3 -m venv path/to/venv.[0m
2024-02-09T15:36:22.5878172Z [1;31m==> arm.raspiblitz-arm64-rpi:     Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make[0m
2024-02-09T15:36:22.5883077Z [1;31m==> arm.raspiblitz-arm64-rpi:     sure you have python3-full installed.[0m
2024-02-09T15:36:22.5887716Z [1;31m==> arm.raspiblitz-arm64-rpi:[0m
2024-02-09T15:36:22.5892821Z [1;31m==> arm.raspiblitz-arm64-rpi:     For more information visit http://rptl.io/venv[0m
2024-02-09T15:36:22.5898072Z [1;31m==> arm.raspiblitz-arm64-rpi:[0m
2024-02-09T15:36:22.5910430Z [1;31m==> arm.raspiblitz-arm64-rpi: note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.[0m
2024-02-09T15:36:22.5912767Z [1;31m==> arm.raspiblitz-arm64-rpi: hint: See PEP 668 for the detailed specification.[0m
```

openoms, 2024-02-12:
comment on the topic from cdecker re the CLN dependencies install getting the same error:
[image]
https://t.me/lightningd/31417

openoms, 2024-08-10:
This issue prevents the new CL wallet generation on raspiblitz v0.11.1 as the `mnemonic` dependency is not installed.

```
##################
ERROR in _provision.setup.sh --> cl-wallet-new-noseeds on Sat 10 Aug 10:18:08 BST 2024\nDETAIL --> /home/admin/config.scripts/cl.hsmtool.sh new-force mainnet --> seedwords=''\nDEBUG --> /home/admin/raspiblitz.provision-setup.log
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(cl-wallet-new-noseeds)
```

```
/home/admin/config.scripts/cl.hsmtool.sh new-force mainnet
# Running 'cl.hsmtool.sh new-force mainnet'
cl.log_old

[notice] A new release of pip is available: 24.0 -> 24.2
[notice] To update, run: python3 -m pip install --upgrade pip
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

[notice] A new release of pip is available: 24.0 -> 24.2
[notice] To update, run: python3 -m pip install --upgrade pip
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.mnemonic.py", line 4, in <module>
    from mnemonic import Mnemonic
ModuleNotFoundError: No module named 'mnemonic'
seedwords=''
seedwords6x4=''
# No seedwords - exiting
```

rootzoll, 2024-08-15:
Merged #4692 to address issue - set to final testing

rootzoll, 2024-08-15:
OK merged the PR and tested. Looking good now for CL install. Movin this issue to v1.11.3 to check again for other places.

woiwoiwoiwoi, 2024-09-04:
I use email notifications to be informed of restarts, etc. I'm on v1.11.1 now and am getting this error when running the `blitz.notify.sh on` script. Is there something I can do on my Raspiblitz to fix it or can I help with this issue in some way, or is it now sorted and I should wait for v.1.11.3?

openoms, 2024-09-04:
> I use email notifications to be informed of restarts, etc. I'm on v1.11.1 now and am getting this error when running the `blitz.notify.sh on` script. Is there something I can do on my Raspiblitz to fix it or can I help with this issue in some way, or is it now sorted and I should wait for v.1.11.3?

this is is solved in 0.11.2.

You can just run:
```
sudo rm /usr/lib/python3.*/EXTERNALLY-MANAGED
```
as discussed above

openoms, 2024-09-23:
Will need to run
```
pip config set global.break-system-packages true
```
in all the remaining places where pip install is being run without a virtual environment.

PR #4400 (2024-02-09): build: remove externally-managed flag for python
Test if the fix #4170 is still needed.
closed 2024-11-07 by rootzoll