raspiblitz/raspiblitz #4182: Deactivate WebUI/Api on fatpack recovery when off

rootzoll, 2023-09-20:
see https://github.com/raspiblitz/raspiblitz/issues/4171#issuecomment-1728302628

rootzoll, 2023-12-19:
Test once we have a first fatpack to test with.

rootzoll, 2024-03-19:
OK will be part of v1.11.0rc5 closing.

closed 2024-03-19 by rootzoll

woiwoiwoiwoi, 2024-03-23:
So I seem to be the edge case you discovered in #4171 and can't get the WebUI to work. It was deactivated in 1.8.0, but after the update to 1.10.0 fatpack (could not get minimal to work) I could access it, even though it was still deactivated in the menu. Then I activated it in the menu, but now it's broken. After putting in my password, I get:

`An error occurred: Unknown error. The response was: 502 Bad Gateway.`

I noticed that during install I get what you mentioned in #4171:

```
# WARNING! The given API repo is not available:
# user(fusion44) repo(blitz_api) branch(v1.10)
# WORKING WITH FALLBACK REPO - USE JUST FOR DEVELOPMENT - DONT USE IN PRODUCTION

# WARNING! The given WebUI repo is not available:
# user(raspiblitz) repo(raspiblitz-web) branch(v1.10)
# WORKING WITH FALLBACK REPO - USE JUST FOR DEVELOPMENT - DONT USE IN PRODUCTION
```

Is there a straight-forward way to fix this? Otherwise I'll just wait for 1.11.0 and keep using ssh.

rootzoll, 2024-03-25:
Best to wait for v1.11.0 .. if you still have an error then please comment again here.