raspiblitz/raspiblitz #4183: state(error) message(lnd-wallet-seed)

chuegel, 2023-09-20:
Hi,

was trying to setup a new raspiblitz `v1.9.0` but was unable to recover a old LND wallet from `SEED+SCB Words Seed & channel.backup`
The menu exits to the shell with the error 
```
---snip
# OK - password A changed for user pi, root, admin & bitcoin
error=''
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(lnd-wallet-seed)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
```
The seed phrase is correct. What does this error message mean?

rootzoll, 2023-09-21:
so the error happens at this part of the code:
https://github.com/raspiblitz/raspiblitz/blob/a3c2ef9b6e49f0a1da8a0d16c2a88f9d7e340feb/home.admin/_provision.setup.sh#L271

more info you should find in the following file - can you post those last lines?:
`cat /home/admin/raspiblitz.provision-setup.log`

chuegel, 2023-09-21:
Not much, here are the last lines:

```
---snip
**************************************************
OK --> THE VERIFIED LND BINARY CHECKSUM IS CORRECT
**************************************************

- install LND binary
- OK install of LND done
Creating new LND config ...
*** Init LND Service & Start ***
Starting LND Service ...
Starting LND Service ... executed
WALLET --> SEED
```

chuegel, 2023-09-21:
I have to mention that the seed words are encrypted. While choosing so in the menu, it never asks for a passphrase.
I guess thats why it crashes

rootzoll, 2023-09-25:
That sounds like the reason. At the moment the recovery of password encrypted seedwords is not supported by raspiblitz. You could maybe try with https://zaphq.io

chuegel, 2023-09-25:
Thanks for confirming

closed 2023-09-25 by chuegel