raspiblitz/raspiblitz #4188: [v1.9.0] rpc error: code = Unknown desc = wallet already unlocked, WalletUnlocker service is no longer available

adevoss, 2023-09-23:
Not looked at my node (was running fine after #3970) for a while. Today I logged in to the Web UI. Got the spinning circle.
So I rebooted.

I saw in the terminal:
```
# calling: lncli unlock
# unknown error
```
Manually:
```
lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unknown desc = wallet already unlocked, WalletUnlocker service is no longer available
```

In the web UI (top right):
Bitcoin done
Lightning offline
Unable to connect to LND daemon, waiting...

lnd service is running

What is your suggestion to solve this?
Is this something to fix manually?
Is it save to reinstall using v1.10?

fusion44, 2023-09-24:
Please check the blitz api logs:
`sudo journalctl -u blitzapi -b --no-pager -n50`

Is there any error present?

adevoss, 2023-09-24:
Yes.

All are

```
python[734]: E0924 13:16:33.238824476    2587 ssl_transport_security.cc:1501] Handshake failed with fatal error SSL_ERROR_SSL: error:1000007d:SSL routines:OPENSSL_internal:CERTIFICATE_VERIFY_FAILED.
```

Thinking of installing v1.10 because I want to start using LNbits.
That's a good opportunity to upgrade. Should I use Debian 11 or 12 for amd64?

But that is more work maybe than if someone knows how to solve this issue in v1.9.
Then I can postpone the upgrade to the official release of v1.10.

adevoss, 2023-09-24:
'Solved' this by installing v1.10.0 on Debian 11.
Debian 12 failed on #4169 when executing `build_sdcard.sh`

closed 2023-09-25 by adevoss