raspiblitz/raspiblitz #4190: [v1.10] LNBits 502 Bad Gateway after trying to update extension

adevoss, 2023-09-25:
Updating an LNBits extension failed. Now even LNBits won't show.
What's a way to fix this?

adevoss, 2023-09-25:
Removing some files of the offending LNBits extension via SSH and a reboot 'solved' the problem.
Then reinstalled the old version of the extension.

Will ask the people at LNBits for more info.

rootzoll, 2023-09-25:
@dni can you take a look what needs to be improved here?

dni, 2023-09-25:
yes we deployed a version for the upcoming `0.11.0` lnbits release the boltcards `0.0.4` wasnt souppoed to be installable in `0.10.10`, that is fixed. extensions will be incompatible 

@adevoss can you verify this. it should look like that. now

[image]

adevoss, 2023-09-25:
@dni Your screenshot is from another extension.
This is what the boltcard extension looks now:

[image]

dni, 2023-09-25:
so, i guess the issue is resolved? it was a issue on lnbits extensions repository!

adevoss, 2023-09-25:
> so, i guess the issue is resolved?
Yes

closed 2023-09-25 by rootzoll