raspiblitz/raspiblitz #4198: Sweeping of onchain funds not possible

GerdOpp, 2023-09-27:
I want to empty my raspiblitz onchain wallet to zero. In the wallet there is only 1 utxo (10.000sats) I have tried RTL and thunderhub but get the same error message on both:

RTL (sweep all funds)"Insufficient Input To Create Sweep Tx: Input_sum=0.0001 Btc, Output_sum=0.0001128 Btc":
thunderhub (with send all): "UnexpectedSendCoinsError. insufficient input to create sweep tx: input_sum=0.0001 BTC, output_sum=0.0001128 BTC"

How can I empty my wallet? Is sweeping possible at all?

rootzoll, 2023-10-04:
If you use LND ... have you tried out loggin in thru SSH and in the main menu go LND > CASHOUT?

GerdOpp, 2023-10-05:
I get the same error:
"FAIL: [lncli] rpc error: code = Unknown desc = insufficient input to create sweep tx: input_sum=0.0001 BTC, output_sum=0.00012143 BTC"

Is the missing amount the tx fee? Shouldn't (in a sweep tx) the tx fee be calculated such that it is included in the total available funds?

If tx fee ist 2143sats (as I assume in this case), the output would be 10000 - 2143 = 7857.

Or is the calculated tx fee higher than the available funds (i.e. > 10000)

GerdOpp, 2023-10-18:
Even with SSH and in the main menu go LND > CASHOUT, the error persists:

******************************
Sweep all possible Funds
******************************
# redis-cli set ln_lnd_mainnet_wallet_onchain_pending:out 0 EX 900
OK
# redis-cli set ln_lnd_mainnet_wallet_onchain_balance:out 0 EX 900
OK
sudo -u bitcoin /usr/local/bin/lncli -n=mainnet --rpcserver localhost:10009 sendcoins --sweepall --addr=bc1qk***************** --conf_target=36

FAIL: [lncli] rpc error: code = Unknown desc = insufficient input to create sweep tx: input_sum=0.0001 BTC, output_sum=0.00011431 BTC

#################################
FAIL --> Was not able to send transaction (see error above)
#################################

closed 2023-12-01 by rootzoll