raspiblitz/raspiblitz #4201: certificate error when connecting electrum

JessieRaspi, 2023-09-30:
Since this week, my electrum can't connect to Raspiblitz any more. I updated to 1.10.0 the minimal installation with a fresh install, but that didn't help. Previously I had version 1.10.0. rc2 with worked smoothly until this week). 
I work with the Electrum desktop app on windows 10. 

In the Electrum log I found that the certificate of the local host has expired on the 22nd of September: 

INFO | network | connecting to 192.168.178.234:50002:s as new interface
20230929T181015.142126Z |     INFO | interface.[192.168.178.234:50002] | certificate has expired: Certificate (localhost) has expired (at 2023-09-22 07:47:17 UTC).
20230929T181015.156174Z |     INFO | interface.[192.168.178.234:50002] | succeeded in getting cert
20230929T181015.165157Z |  WARNING | interface.[192.168.178.234:50002] | Cannot connect to main server due to SSL error (maybe cert changed compared to "C:\Users\xxxxxx\AppData\Roaming\Electrum\certs\192.168.178.234"). Exc: ConnectError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1129)'))
20230929T181015.166156Z |     INFO | network | couldn't launch iface 192.168.178.234:50002:s -- CancelledError()

How and where can I change the certificate?

JessieRaspi, 2023-10-02:
After thinking for a while, I changed the connection in my desktop Electrum from : 50002:s into 50001:t. That worked. I'm on my cabled home network.

JessieRaspi, 2023-10-02:
I just downloaded Sparrow wallet and managed to connect it to Raspiblitz through 50002 and SSL enabled. 
In de certs of Sparrow wallet on my windows pc, I found that my certificate there was dated at 22 -08-2022. 
I deleted the Electrum cert and copied the sparrow cert into the Electrum cert directory, but unfortunately Electrum deleted that copied Sparrow cert and instead put the recent one that it finds and could not connect....

I also tried to open RTL in Firefox with https. Got the usual certificate error (LND autogenerated cert for local host). When I looked through the certificate, I found that the validation is from 28 July 2022 till 23 September 2023...So, outdated as described in the Electrum log. But no problem. After accepting, RTL starts.

Really don't know what to try else to solve this issue. I have seen two certificate files on Raspiblitz, but don't know how to read them nor how to change those (the date).

freerko, 2023-11-21:
This is a client issue. Electrum uses the TOFU "trust on first use" principle. When a certificate from the server (here the Raspiblitz) expires, the client won't connect to that server even when you replace the certificate on the server with a new one, you need to issue a new certificate on the RaspiBlitz AND delete the old one on your client:

Replacing the server on the RaspiBlitz should be possible like this via SSH:

> cd /home/electrumx/.electrumx/
> openssl genrsa -out keyfile.key 2048
> openssl req -new -key keyfile.key -out server.csr
> openssl x509 -req -days 1825 -in server.csr -signkey server.key -out certfile.crt
> service restart electrumx

On the client, delete the corresponding old cert from C:\Users\xxxxxx\AppData\Roaming\Electrum\certs\192.168.178.234 or something like that.

rootzoll, 2023-11-22:
electrs uses the nginx SSL proxy on port 5002 - if the SSL cert is outdated this needs to be handled by nginx

https://github.com/raspiblitz/raspiblitz/blob/624f8ca1a78d5935f94ea90af2136be088442c8a/home.admin/config.scripts/bonus.electrs.sh#L357

TODO: check nginx SSL certs - undo using the lnd one (system might not have lnd running)

JessieRaspi, 2024-01-12:
After having installed Electrum on a Linux PC and connected to Electrs in my Raspiblitz, the same cert error popped up.
After SSH'd in, I recreated the LND Macaroons and TLS.cert from the Connect Menu. 
That helped! I finally can connect through 50002:s instead of 50001:t

rootzoll, 2024-09-27:
This is now also discussed here: https://github.com/raspiblitz/raspiblitz/issues/4431#issuecomment-2377624050 .. closing this issue as duplicate.

closed 2024-09-27 by rootzoll