raspiblitz/raspiblitz #4204: [v1.10] Make LNBits configuration file persistant

adevoss, 2023-10-01:
After starting to use LNBits I changed the configuration file `/home/lnbits/lnbits/.env` (allowed users added)
When upgrading / reinstalling Raspiblitz that file will be lost.

rootzoll, 2023-12-16:
With a soon comning LNbits update more configuration options will change (super user, etc) - lets add also persisting LNbits manual configs to that momemt. Should best fit with v1.11.1

rootzoll, 2024-10-01:
move it to app-data
and delete on deinstall when yes to remove data option

rootzoll, 2024-10-02:
Commit to `4568-lnbits` and merge this as part of #4768

rootzoll, 2024-10-02:
OK is now part of PR #4768

- [ ] Test persistant .env thru recover
- [ ] Inform Local Meetup that .env can get lost "one more time"

closed 2024-11-07 by rootzoll