raspiblitz/raspiblitz #4211: Raspiblitz doesnt start after updating to version 1.10.0

Sigevi, 2023-10-08:
Raspiblitz stucks after flashing it serveral times with the new update. 
It always works if i flash the version 1.9.0 again. But not once with v1.10.0.
Now it work just to the point when the Display says: "Starting Raspiblitz, Please wait...

That stays for days...

But in the Version 1.9.0. there was an other problem. Lightning shuts down everytime after a few days when it was running. 
A reboot helps just for a few hours or maybe days.
[image]

gecko20, 2023-10-10:
Same here after an update from version 1.8.0

salenjak, 2023-10-11:
> But in the Version 1.9.0. there was an other problem. Lightning shuts down everytime after a few days when it was running. A reboot helps just for a few hours or maybe days. !

I believe this is related to SD card being full. https://github.com/raspiblitz/raspiblitz/issues/4149

rootzoll, 2023-10-18:
It looks like the raspberrypi does not get an local network connection thru the LAN cable. Do you see any status lights next to the LAN cable plug?

Sigevi, 2023-10-21:
[image]
[image]

Yes the status lights are working. I now changed again to the v1.9.0 beacuse there was no chance to start it with the version v.1.10.0. With this version you can see on the picture, that it is shutting down the LND 0.17.0-beta. At least it worked for 1 Week now. 
Before first trying to download the electrum rust server it worked for months.

Sigevi, 2023-10-28:
> It looks like the raspberrypi does not get an local network connection thru the LAN cable. Do you see any status lights next to the LAN cable plug?

Yeah you were right. For some reason the LAN cable did not work properly. So i changed the cable and now i was able to pla the version 1.10.0. 
But now i am just happy that it works again and i wait to reinstall the Electrum server and BTCpay. Maybe in a few weeks again... Thanks for your support!

closed 2023-10-28 by Sigevi