raspiblitz/raspiblitz #4213: [v1.10.0] LNBits: Can not create invoice (500)

adevoss, 2023-10-09:
In LNBits when creating an invoice, after clicking 'create', there is a '500' in red at the bottom for a short period of time.

`sudo journalctl -f -xeu lnbits`
```
File "/home/lnbits/lnbits/lnbits/wallets/void.py", line 23, in create_invoice
     raise Unsupported("")
 lnbits.wallets.base.Unsupported
 2023-10-09 21:00:50.11 | ERROR | Exception:
 2023-10-09 21:00:50.11 | INFO | xxx.xxx.xxx.xxx:0 - "POST /api/v1/payments HTTP/1.0" 500
 2023-10-09 21:00:50.11 | ERROR | Exception in ASGI application
```

xxx.xxx.xxx.xxx is the wrong private ip address (not the node itself)

I can't find where the wrong ip address comes from or how to fix that.

https://l.termbin.com/6epd

rootzoll, 2023-12-18:
I was testing with LNbits 0.11.3 for RaspiBlitz v1.11.0rc1 and all seems good.

rootzoll, 2023-12-18:
Please reopen if you stilll have this problem once RaspiBlitz v1.11.0 final is out.

closed 2023-12-18 by rootzoll