raspiblitz/raspiblitz #4215: New base image for RPi - RaspiOS on Debian Bookworm

openoms, 2023-10-11:
Announcement: https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

On top of the issues already covered for the amd64 build using Bookworm (Debian 12) there are a few things to check:

From: https://wiki.debian.org/NewInBookworm

- [ ] change `/bin/sh` to `/bin/bash`
- [ ] install rsyslog with system-log-daemon packages
- [ ] use `command -v` instead of `which`

From https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/:
- [ ] `dhcpd` is changed to `NetworkManager`
- [ ] X11 is changed to Wayland diaplay manager - check if the LCD is affected

rootzoll, 2023-12-01:
@openoms what is the status of the checkboxes above?

openoms, 2023-12-02:
@rootzoll updated the statuses above, there is only one left about `dhcpd` affecting the `internet.dns.sh` script.

On Debian12 there is neither:
```
/etc/dhcpcd.conf
/etc/dhcpd.conf
```

Also the command `host` was not installed for the `test`.
Needed:
```
sudo apt install bind9-host
```
Then the test is successful:
```
host w3c.org | grep -c "w3c.org has address"
1
```

but if this would fail setting the DNS in `/etc/dhcpcd.conf` wouldn't work.

rootzoll, 2024-01-19:
The different DNS handling is now worked in?

openoms, 2024-01-19:
> The different DNS handling is now worked in?

Nope, not implemented yet, just tested manually.

rootzoll, 2024-02-25:
Ready for ReleaseCandidate

closed 2024-02-25 by rootzoll