raspiblitz/raspiblitz #4216: no module name gunicorn after update to 1.10..0

orangedmind, 2023-10-11:
After updating raspiblitz to version 1.10.0 Core Lightning no longer starts.
When I check the cllog I only find the message:

2023-10-11T15:31:36.408Z INFO    plugin-clnrest.py: Killing plugin: disabled itself: No module named 'gunicorn'

I already tried installing the packagage via "pip install gunicorn" but after restart of the node the issue remains.
Any ideas?

openoms, 2023-10-12:
the clnrest plugin is not used currently and it should not crash lightningd even if it is not running.

@orangedmind please double check if CLN recovered since.
You can check the installed version wih:
```
₿ lightningd --version
v23.08.1
```

openoms, 2023-10-12:
The dependencies are not installed by pip, but using poetry in the directory:
`/home/bitcoin/lightning`
https://github.com/raspiblitz/raspiblitz/blob/60b5c47240459bceca20ea34f8d921d5ba9d0677/home.admin/config.scripts/cl.install.sh#L55

Can try updating the deps with:
```
cd /home/bitcoin/lightning
sudo -u bitcoin git reset --hard v23.08.1
sudo -u bitcoin poetry install
```

orangedmind, 2023-10-12:
I followed your instructions for updating the preps. 
After a restart of the raspiblitz sames behavior, CL does not start.

openoms, 2023-10-12:
@orangedmind you need to share more info.
enter the CLN menu which will display the last lines of `cllog` and `sudo journalctl -fu lightningd`

orangedmind, 2023-10-12:
Please find below the requested logs.
[image]

[image]

openoms, 2023-10-12:
You need to delete the erroneous line 20 from the CLN config as shown in the errorr message.

orangedmind, 2023-10-12:
Thank you. CL is running again.

closed 2023-10-12 by orangedmind