raspiblitz/raspiblitz #4221: Brand new node with version 1.10.0 stuck on "Preparing Blockchain Sync"

VeniVidiViciLA, 2023-10-19:
Hi Guys,

I tried to setup a brand new node using blitzversion 1.10.0 but it seems like I'm stuck on 

I'm not quite sure what could be the problem, could it be the ssd ? It's a 1TB samsung SSD.

Thank you

Here are my debug logs
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / test
 19:17:47 up 17:13,  2 users,  load average: 1.49, 2.16, 2.41

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT             LOAD   ACTIVE SUB    DESCRIPTION
● bitcoind.service loaded failed failed Bitcoin daemon on mainnet

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
1 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> setup
state--> waitsync
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()
# waitsync loop ... btc_default_ready()

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2023-10-19 02:15:18 BST; 17h ago
    Process: 101990 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 102021 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
        CPU: 237ms

Warning: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Thu 2023-10-19 10:57:56 BST, ends at Thu 2023-10-19 19:17:47 BST. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
[... 191 lines cut ...]
serviceLoaded=1
serviceActive=1

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0

*** SYSTEM CACHE STATUS ***
system_ram_available_mb="3444"
system_ups_status="OFF"
system_count_start_blockchain="6"
system_init_time="1695342162"
system_vm_vagrant="0"
system_count_start_lightning="1"
system_temp_fahrenheit="117"
system_ups_battery=""
system_up="62073"
system_cpu_load=" 2.08, 2.22, 2.41"
system_ram_mb="3844"
system_count_undervoltage="0"
system_temp_celsius="47"
system_board="rp4"
system_ram_gb="4"
ln_default_sync_initial_done=""
btc_default_sync_initial_done="0"
*** POSSIBLE ERROR REPORTS ***

rootzoll, 2023-10-23:
You seem stuck in the wait loop for the Bitcoind service:
```
# waitsync loop ... btc_default_ready()
```

It failed to start up - see here:

```
** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2023-10-19 02:15:18 BST; 17h ago
    Process: 101990 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 102021 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
```

But error logs are empty. So basically you can try on console to start manually by command and see what the result is. Todo so check the `sudo cat /etc/systemd/system/bitcoind.service` exec line and try to start it with a `sudo -u bitcoin` in front. Post the result.

TODO DEV: Try to add more debug during waiting loop for next release.

ShinyIsles, 2023-10-31:
I have a similar situation. I'm new to much of this. Have set up the following:

Raspberry Pi 4 8GB
Raspberry Pi 4 Power Supply USB-C 5.1V 3A
16GB SD Card
2TB SSD and SATA cable (USB 3.0 to SSD 2.5”)
Geekworm X862 V2.0 M.2 NGFF SATA SSD Storage Expansion Board
Western Digital WD 2TB SSD, WD Blue 3D NAND, M.2 2280 SATA
CryptoCloaks Palas in White (gonna spray it orange!)
ColdCard

Everything ent well until about 42% of the blockchain synced according to the Blitz UI (I have not SSHed in at this point).

At some point, the progress indicator just stated "Starting Sync" and has stayed that way for days. The lights are flashing on the LAN port, and the heatsink is warm. I tried SSHing in just now and was able. Here is the Debug:

* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / test
 07:55:23 up 2 days,  9:43,  2 users,  load average: 2.35, 2.38, 2.35

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sat 2023-10-28 23:18:23 BST; 2 days ago
    Process: 28800 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 28832 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
        CPU: 38.932s

Warning: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Mon 2023-10-30 16:05:41 GMT, ends at Tue 2023-10-31 07:55:23 GMT. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log

*** LND (MAINNET) SYSTEMD STATUS ***
- OFF by config -

*** CORE LIGHTNING (MAINNET) SYSTEMD STATUS ***
- not activated -

*** BLOCKCHAIN (TESTNET) SYSTEMD STATUS ***
- OFF by config -
*** LND (TESTNET) SYSTEMD STATUS ***
- OFF by config -

*** CORE LIGHTNING (TESTNET) SYSTEMD STATUS ***
- not activated -

*** BLOCKCHAIN (SIGNET) SYSTEMD STATUS ***
- OFF by config -
*** LND (SIGNET) SYSTEMD STATUS ***
- OFF by config -
[... 184 lines cut ...]
system_temp_fahrenheit="131"
system_board="rp4"
system_cpu_load=" 2.40, 2.39, 2.36"
system_temp_celsius="55"
system_ram_gb="8"
ln_default_activated="1"
ln_default_locked="0"
ln_default_error_short=""
ln_default_sync_initial_done="0"
ln_default_ready="0"
ln_default_online="0"
ln_default_version="v23.08.1"
ln_default_running="0"
ln_default_error_full=""
btc_default_running="0"
btc_default_error_full=""
btc_default_sync_initial_done="0"
btc_default_activated="1"
btc_default_version="v25.0.0"
btc_default_online="0"
btc_default_error_short=""
btc_default_ready="0"
*** POSSIBLE ERROR REPORTS ***

Any help for a newbie would be much appreciated.

felivix, 2023-12-05:
Since days I get the same status report:
Lightning Progress : prepare sync

[image]
[image]

rootzoll, 2023-12-19:
@openoms how can someone investigate why CLN is not getting into the ready state for debug?

felivix, 2024-03-25:
> @openoms how can someone investigate why CLN is not getting into the ready state for debug?

Issue continues "Lightning Progress : prepare sync" since 3 months!

openoms, 2024-03-25:
More logs are needed.
Get CLN error logs:

```
sudo journalctl -fu lightningd -n50
```

generic CLN logs:
```
cllog
```

felivix, 2024-05-21:
@openoms

I upgraded to RaspiBlitz v1.11.0 but the problem persists. 

bitcoin v26.0.0 mainnet 39 peers
Blocks 844403/844403 Sync OK 100.00%
CLN v24.02.1
Not Started | Not Ready Yet | No Data