raspiblitz/raspiblitz #4225: Raspiblitz "hangs" after a while - and works again after SD flash

velgard1, 2023-10-23:
Short history of my issue:
- 1.8 working fine
- Read error / Index issue
- Update to 1.9
- Starting Reindex
- Errors / freezing at apx. 25%
- SSD seems to be ok
- Again SD flash with 1.9
- Errors / freezing at apx. 50%
- Update to 1.10
- Issues continued
- Managed to fully sync again, but freezes after a while

For me it seems to be a config issue with a swapfile or something like this that gets too big over time. With a new setup it will work for a while since it starts from scratch but the issue will happen again due to the unchanged config.

Anyone an idea what I could check or is the log needed?

cstenglein, 2023-10-24:
Please attach a debug log.
Some people had freeze issues with 1.9 but haven't heard that with 1.10

velgard1, 2023-10-24:
Here the debug log from the Web UI - hope it helps, thanks.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 18:58:51 up 20:43,  1 user,  load average: 2.95, 2.84, 2.92

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-10-23 22:21:49 BST; 20h ago
    Process: 6451 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 6516 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 6652 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 4h 29min 52.162s
     CGroup: /system.slice/bitcoind.service
             └─6652 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Oct 23 22:16:44 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Oct 23 22:21:49 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Mon 2023-10-23 21:42:46 BST, ends at Tue 2023-10-24 18:58:51 BST. --
Oct 23 22:16:44 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Oct 23 22:21:49 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-10-24T16:42:08Z Saw new header hash=00000000000000000004875a1e4dddfcdf7a56214e3de512d7ae808f0190095b height=813647
2023-10-24T16:42:08Z [net] Saw new cmpctblock header hash=00000000000000000004875a1e4dddfcdf7a56214e3de512d7ae808f0190095b peer=626
2023-10-24T16:42:09Z UpdateTip: new best=00000000000000000004875a1e4dddfcdf7a56214e3de512d7ae808f0190095b height=813647 version=0x20002000 log2_work=94.496087 tx=908570690 date='2023-10-24T16:41:51Z' progress=1.000000 cache=449.4MiB(3178503txo)
2023-10-24T16:42:09Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-10-24T16:43:47Z Saw new header hash=000000000000000000048d702f2ae8ed931e10319511105d76f9fd50ab3cf2a4 height=813648
2023-10-24T16:43:47Z [net] Saw new cmpctblock header hash=000000000000000000048d702f2ae8ed931e10319511105d76f9fd50ab3cf2a4 peer=626
2023-10-24T16:43:48Z UpdateTip: new best=000000000000000000048d702f2ae8ed931e10319511105d76f9fd50ab3cf2a4 height=813648 version=0x2000a000 log2_work=94.496101 tx=908574254 date='2023-10-24T16:43:38Z' progress=1.000000 cache=449.7MiB(3180936txo)
2023-10-24T16:44:02Z Socks5() connect to qlxwo4otmsh5pnxaqsvtpookt45u5pl2y2hb6sqqr7pubtr7unvszcqd.onion:8333 failed: host unreachable
2023-10-24T16:46:27Z New outbound peer connected: version: 70016, blocks=813648, peer=1335 (block-relay-only)
2023-10-24T16:47:23Z Saw new header hash=00000000000000000002cc9d974a75ca1f5a2c0e3961ce9fd182963aadd38b2d height=813649
2023-10-24T16:47:23Z [net] Saw new cmpctblock header hash=00000000000000000002cc9d974a75ca1f5a2c0e3961ce9fd182963aadd38b2d peer=646
2023-10-24T16:47:24Z UpdateTip: new best=00000000000000000002cc9d974a75ca1f5a2c0e3961ce9fd182963aadd38b2d height=813649 version=0x223ae000 log2_work=94.496114 tx=908576942 date='2023-10-24T16:47:18Z' progress=1.000000 cache=450.0MiB(3183800txo)
2023-10-24T16:47:38Z Socks5() connect to 3xejlzg5xti5dk6dob7lnvypdbdvpl77mvv7h3gdrreqjwwjqlaegqqd.onion:8333 failed: host unreachable
2023-10-24T16:47:49Z Socks5() connect to 3xejlzg5xti5dk6dob7lnvypdbdvpl77mvv7h3gdrreqjwwjqlaegqqd.onion:8333 failed: host unreachable
2023-10-24T16:49:52Z Socks5() connect to gu5frmkxkmxc24ufkewsschp76fpdlq7v7crvylrzmrxjhzmd4fts6ad.onion:8333 failed: host unreachable
2023-10-24T16:51:56Z Saw new header hash=00000000000000000001598e50200689e5bef61653ae174a708cc1756c18b63e height=813650
2023-10-24T16:51:56Z [net] Saw new cmpctblock header hash=00000000000000000001598e50200689e5bef61653ae174a708cc1756c18b63e peer=646
2023-10-24T16:51:56Z UpdateTip: new best=00000000000000000001598e50200689e5bef61653ae174a708cc1756c18b63e height=813650 version=0x2000e000 log2_work=94.496128 tx=908578066 date='2023-10-24T16:51:38Z' progress=1.000000 cache=449.2MiB(3176849txo)
2023-10-24T16:58:59Z New outbound peer connected: version: 70016, blocks=813650, peer=1346 (block-relay-only)
2023-10-24T17:04:31Z Socks5() connect to zxddusjpw4vel6sizgaro262lx7vnvffnlfrm5sie3jufzb4ddujrpid.onion:8333 failed: host unreachable
2023-10-24T17:05:30Z Socks5() connect to zxddusjpw4vel6sizgaro262lx7vnvffnlfrm5sie3jufzb4ddujrpid.onion:8333 failed: host unreachable
2023-10-24T17:05:42Z New outbound peer connected: version: 70016, blocks=813650, peer=1353 (block-relay-only)
2023-10-24T17:10:11Z New outbound peer connected: version: 70016, blocks=813650, peer=1357 (block-relay-only)
[... cut ...]
btc_default_sync_initialblockdownload="0"
btc_default_online="1"
btc_default_peers="16"
btc_default_mempool_transactions="59960"
btc_default_error_full=""
btc_default_synced="1"
btc_default_version="v25.0.0"
btc_default_sync_percentage="100.00"
btc_default_error_short=""
btc_default_blocks_data_kb="577457780"
btc_default_running="1"
btc_default_blocks_behind="0"
btc_default_ready="1"
btc_default_blocks_verified="813656"
btc_default_port="8333"
btc_default_sync_initial_done="1"
*** POSSIBLE ERROR REPORTS ***

openoms, 2023-10-25:
The mempool space entry is causing an issue in nginx. 
try to run:
```
config.scripts/bonus.mempool.sh off
config.scripts/bonus.mempool.sh uninstall
config.scripts/bonus.mempool.sh on
```
then for litd you might need to reset the macaroons in the CONNECT menu.
and reinstall:
```
config.scripts/bonus.litd.sh off
config.scripts/bonus.litd.sh on
```

velgard1, 2023-10-27:
The first scripts works, the second scripts didn't - no such file or directory.
I'll see if it's running normal now, will consider the the issue closed in the meantime, thanks.

closed 2023-10-27 by velgard1

mr-q17, 2023-10-27:
The CPU temperature is high

system_temp_celsius="71"

I can recommended to cool down. And make sure that you have a good power source. This is often the reason for hang ups. Because of very short power spikes. These spikes are not always present.

velgard1, 2023-10-31:
Yeah, it's a bit hot - it's caused by covering the display at night, otherwise I have about 66°C.

On the other hand this was not a problem during the last years and from what I read it shouldn't be.
Obviously cooler is better, so are there general things that can be done that have an effect?

I'd assume that running services have an influence - is there any information or do I simply have to try?
Currently running: Mempool Explorer, Block Explorer, Ride the Lightning & Lightning Terminal