raspiblitz/raspiblitz #4232: Channels repeated going offline

glensidecross, 2023-10-29:
Hi.  After upgraeding to 1.10.0, every once in a while, my channel partners all seem to go offline. When I contact some of the people who I know who are running nodes connected to mine, they say their node is fine but that mine seems offline. A reboot fixes the problem, but it reoccurs periodically. 

Thanks in advance for your help!

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 22:37:24 up 17:24,  2 users,  load average: 2.78, 2.91, 3.04

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-10-29 05:20:10 GMT; 17h ago
    Process: 7719 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 7747 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 7855 (bitcoind)
      Tasks: 19 (limit: 9352)
        CPU: 2h 33min 44.082s
     CGroup: /system.slice/bitcoind.service
             └─7855 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Oct 29 05:15:06 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Oct 29 05:20:10 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Sat 2023-10-28 17:42:43 BST, ends at Sun 2023-10-29 22:37:24 GMT. --
Oct 29 05:15:06 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Oct 29 05:20:10 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-10-29T21:51:45Z [i2p] Error connecting to zysrlpii5ftrzivfcyhdrwpeyyqddbrdefnfu5q6otk5gtugmh2a.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T21:53:11Z [i2p] Error connecting to day3hgxyrtwjslt54sikevbhxxs4qzo7d6vi72ipmscqtq3qmijq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T21:54:41Z [i2p] Error connecting to lzuu6mjtu7vd55d2biphicihufipoa7vyym6xfnkmmlra3tiziia.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T21:56:07Z [i2p] Error connecting to 4fcc23wt3hyjk3csfzcdyjz5pcwg5dzhdqgma6bch2qyiakcbboa.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=4fcc23wt3hyjk3csfzcdyjz5pcwg5dzhdqgma6bch2qyiakcbboa.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=4fcc23wt3hyjk3csfzcdyjz5pcwg5dzhdqgma6bch2qyiakcbboa.b32.i2p"
2023-10-29T21:57:34Z [i2p] Error connecting to pohfcrfc7prn4bvn4xstw6nt3e7hjmb7kuj4djtsfqsskwhmhnna.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=pohfcrfc7prn4bvn4xstw6nt3e7hjmb7kuj4djtsfqsskwhmhnna.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=pohfcrfc7prn4bvn4xstw6nt3e7hjmb7kuj4djtsfqsskwhmhnna.b32.i2p"
2023-10-29T21:58:59Z [i2p] Error connecting to 6j2ezegd3e2e2x3o3pox335f5vxfthrrigkdrbgfbdjchm5h4awa.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T21:59:50Z Socks5() connect to ***.onion:8333 failed: host unreachable
2023-10-29T22:00:30Z [i2p] Error connecting to bitcoi656nll5hu6u7ddzrmzysdtwtnzcnrjd4rfdqbeey7dmn5a.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T22:01:48Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2993 seconds ago)
2023-10-29T22:01:57Z [i2p] Error connecting to day3hgxyrtwjslt54sikevbhxxs4qzo7d6vi72ipmscqtq3qmijq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T22:02:24Z New outbound peer connected: version: 70016, blocks=814425, peer=1111 (outbound-full-relay)
2023-10-29T22:03:23Z Saw new header hash=000000000000000000031fce3d12c0723c8e011c374d943c982389627bfd003a height=814426
2023-10-29T22:03:23Z [net] Saw new cmpctblock header hash=000000000000000000031fce3d12c0723c8e011c374d943c982389627bfd003a peer=12
2023-10-29T22:03:25Z UpdateTip: new best=000000000000000000031fce3d12c0723c8e011c374d943c982389627bfd003a height=814426 version=0x20000000 log2_work=94.506594 tx=911015321 date='2023-10-29T22:03:14Z' progress=1.000000 cache=211.3MiB(1448311txo)
2023-10-29T22:03:27Z [i2p] Error connecting to ejlnngarmhqvune74ko7kk55xtgbz5i5ncs4vmnvjpy3l7y63xaa.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T22:04:15Z Saw new header hash=00000000000000000003f5f04fe9843a3c26d7c269d26475cc84137527eef4f3 height=814427
2023-10-29T22:04:15Z [net] Saw new cmpctblock header hash=00000000000000000003f5f04fe9843a3c26d7c269d26475cc84137527eef4f3 peer=12
2023-10-29T22:04:15Z UpdateTip: new best=00000000000000000003f5f04fe9843a3c26d7c269d26475cc84137527eef4f3 height=814427 version=0x24a8c000 log2_work=94.506608 tx=911019096 date='2023-10-29T22:03:53Z' progress=1.000000 cache=212.0MiB(1454458txo)
2023-10-29T22:04:15Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-10-29T22:04:53Z [i2p] Error connecting to tugq6wa2ls2bv27pr2iy3da3k5ow3fzefbcvjcr22uc7w5vmevja.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T22:05:40Z New outbound peer connected: version: 70016, blocks=814427, peer=1118 (outbound-full-relay)
2023-10-29T22:06:18Z [i2p] Error connecting to vgu6llqbyjphml25umd5ztvyxrxuplz2g74fzbx75g3kkaetoyiq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-10-29T22:07:46Z [i2p] Error connecting to yc4xwin5ujenvcr6ynwkz7lnmmq3nmzxvfguele6ovqqpxgjvonq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
[... 494 lines cut ...]
ln_default_ready="1"
ln_default_recovery_mode="0"
ln_default_fees_total="219"
ln_default_activated="1"
ln_default_recovery_done="0"
btc_default_blocks_behind="0"
btc_default_peers="32"
btc_default_sync_initial_done="1"
btc_default_port="8333"
btc_default_running="1"
btc_default_synced="1"
btc_default_sync_percentage="100.00"
btc_default_activated="1"
btc_default_blocks_data_kb="580084520"
btc_default_error_full=""
btc_default_online="1"
btc_default_version="v25.0.0"
btc_default_sync_progress="0.9999996573694596"
btc_default_error_short=""
btc_default_blocks_headers="814428"
btc_default_sync_initialblockdownload="0"
btc_default_blocks_verified="814428"
btc_default_mempool_transactions="10186"
btc_default_ready="1"
*** POSSIBLE ERROR REPORTS ***

openoms, 2023-10-31:
Seems like LND is restarting for some reason.
Follow it's log to find out why - need to see what triggers the restart.

You can stop it from restarting automatically to catch the end of the logs if you set:
`Restart=always`
to 
`Restart=never`

in:
```
sudo systemctl edit --full lnd.service
```

glensidecross, 2023-10-31:
Thanks! I appreciate your time.

closed 2023-10-31 by glensidecross