raspiblitz/raspiblitz #4234: LND<->CLN PrivateKey

rootzoll, 2023-11-03:
Test if a private key (in the form of backup words) can be echanged between LND & CLN - so that people that want to change implementations at least can keep their NodeID/History even all channels need to be closed and ropened.

openoms, 2023-11-06:
This has no solution yet as LND is using the aezeed format and CLN takes BIP39 seed words so the same words resolve to a different nodeID.

rootzoll, 2024-04-23:
@openoms Core Lightning has BIP39 not natively intergrated, correct? 

So we could offer a bounty for anybody being able to (create and) convert an Aezeed-Seed to an `hsm_secret` (when user started with LND and wants to create CLN wallet) - or the other way around (hsm secret -> aezeed) with birthday of wallet being the date we introduced CLN on Raspiblitz (when user has created CLN and wants backup seeds and later create LND wallet from it).

But just something to think about: But when we still offer to run LND and CLN at the same time, we cannot run the with the same nodeid. So this converstion would just be helpfull if people want to move between CLN & LND - so this would just be an edge case.

In general ChatGPT on Azeed & BIP-39:
----------------------------------------

Converting between the aezeed format and BIP39 format for backing up private keys in Bitcoin Lightning clients is theoretically possible, but it's complex and comes with practical challenges. Let's explore these two formats and the feasibility of conversion:

### Aezeed Format
The aezeed format (AEAD SEED) is a mnemonic seed format developed specifically for use in Lightning Network wallets, particularly LND (Lightning Network Daemon). This format includes:
- A mnemonic phrase that encodes wallet seed data.
- A birthday timestamp indicating when the wallet was created, which helps the wallet synchronize only relevant blockchain data.
- Built-in error detection and correction capabilities.
- Encryption of the seed using a passphrase before being represented as a mnemonic. This feature offers additional security by requiring the passphrase to decrypt the seed.

### BIP39 Format
BIP39 is a widely used standard for generating deterministic wallets. It involves:
- Generating a mnemonic phrase from a random seed.
- Optionally using a passphrase in addition to the mnemonic for extra security (often referred to as a "25th word").
- This mnemonic is then used to generate a seed that can be used to derive private keys using BIP32 or similar hierarchical deterministic wallet algorithms.

### Conversion Feasibility
1. **Seed Extraction**: The first step in converting between these formats would be to extract the raw binary seed or entropy from the mnemonic phrase. For aezeed, this would involve decrypting the mnemonic using the passphrase, and for BIP39, deriving the entropy directly from the mnemonic (if no passphrase is used, or using it if one is included).

2. **Format Transformation**: Once you have the raw seed:
   - From **Aezeed to BIP39**: Convert the extracted seed from the aezeed format into a format suitable for BIP39. Since BIP39 expects certain entropy lengths (typically 128, 160, 192, 224, or 256 bits), the extracted seed must match these expectations or be appropriately padded or truncated.
   - From **BIP39 to Aezeed**: Here, the challenge is encoding the additional data required by aezeed (like the birthday timestamp and error-correction codes) and then encrypting it. You would need to generate appropriate metadata and apply aezeed's specific encryption and encoding methods.

3. **Practical Considerations**: While technically possible, the process is not straightforward and requires careful handling of cryptographic functions and sensitive data. It's also worth noting that this conversion isn't typically supported by wallet software, meaning that a custom implementation would be necessary. Such conversions can introduce risks, including potential exposure of private keys or incorrect wallet restoration leading to loss of funds.

4. **Security and Usability**: Each format has specific security features and usability considerations. Converting from one to another might bypass or undermine these features. For example, converting a mnemonic encrypted under aezeed (which includes additional security measures like encryption) to a BIP39 mnemonic could reduce the overall security level if not handled correctly.

In conclusion, while converting between aezeed and BIP39 formats is theoretically possible, it involves significant technical challenges and risks. It's generally advisable to use the backup format supported natively by your wallet software to maintain security and functionality. If such a conversion is absolutely necessary, it should be handled with extreme caution, ideally with the assistance of a cryptography expert.