raspiblitz/raspiblitz #4235: wrong sync progress status with alt vm installation @raspiblitz 1.10.0

HannesMC, 2023-11-04:
**Describe the bug**
Menu -> INFO shows wrong sync status
[image]

**To Reproduce**
- virtual Installation
- vHost: Proxmox 7.4.-17
- - VM OS: Debian 11.8
- - raspiblitzVer: 1.10.0

**Debug file**

`Generating debug logs. Be patient, this should take maximum 2 minutes ..
/home/admin/config.scripts/blitz.web.api.sh: Zeile 30: cd: /home/blitzapi/blitz_api: Datei oder Verzeichnis nicht gefunden
/home/admin/config.scripts/blitz.web.ui.sh: Zeile 26: cd: /home/blitzapi/blitz_web: Datei oder Verzeichnis nicht gefunden
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 01:33:48 up  3:52,  1 user,  load average: 3,28, 2,81, 2,57

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-11-03 21:42:04 CET; 3h 51min ago
    Process: 4784 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 4795 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 4803 (bitcoind)
      Tasks: 18 (limit: 9490)
     Memory: 5.8G
        CPU: 1h 3min 36.112s
     CGroup: /system.slice/bitcoind.service
             └─4803 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Nov 03 21:41:52 lab-raspiblitz-1 systemd[1]: Starting Bitcoin daemon on mainnet...
Nov 03 21:42:04 lab-raspiblitz-1 systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Thu 2023-11-02 21:59:02 CET, ends at Sat 2023-11-04 01:33:48 CET. --
Nov 03 21:41:52 lab-raspiblitz-1 systemd[1]: Starting Bitcoin daemon on mainnet...
Nov 03 21:42:04 lab-raspiblitz-1 systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-11-04T00:33:35Z UpdateTip: new best=00000000000000000b61c55f3c11617e4d328e9e719e8a86ca83dc93ae978bac height=345023 version=0x00000002 log2_work=82.307505 tx=60645924 date='2015-02-25T02:35:45Z' progress=0.068263 cache=281.9MiB(1544212txo)
2023-11-04T00:33:36Z UpdateTip: new best=000000000000000015f5d9e310ddeecea805da8f9fa4ed8193f842b30cfb77b6 height=345024 version=0x00000002 log2_work=82.307554 tx=60646432 date='2015-02-25T02:44:04Z' progress=0.068263 cache=281.9MiB(1544261txo)
2023-11-04T00:33:36Z UpdateTip: new best=000000000000000016da7380031d0144452f62ee12f847750ba4d968a5742f4a height=345025 version=0x00000002 log2_work=82.307602 tx=60647087 date='2015-02-25T02:55:20Z' progress=0.068264 cache=281.8MiB(1543593txo)
2023-11-04T00:33:36Z UpdateTip: new best=0000000000000000120de8243f496ec6ef2bc8bcc5817a17ba7774fb2fe23fce height=345026 version=0x00000002 log2_work=82.307650 tx=60648315 date='2015-02-25T03:19:55Z' progress=0.068265 cache=281.9MiB(1544256txo)
2023-11-04T00:33:36Z UpdateTip: new best=00000000000000000efe3bcea1bd190e36e839ca3e9afeefdb7f126a50d28443 height=345027 version=0x00000002 log2_work=82.307699 tx=60649395 date='2015-02-25T03:41:36Z' progress=0.068267 cache=282.0MiB(1544819txo)
2023-11-04T00:33:36Z UpdateTip: new best=00000000000000001351ba2ee0f97b39e391ea75ba3b4e29957eab20bfe1ee3f height=345028 version=0x00000002 log2_work=82.307747 tx=60650344 date='2015-02-25T03:57:25Z' progress=0.068268 cache=282.0MiB(1545353txo)
2023-11-04T00:33:36Z UpdateTip: new best=000000000000000008f6573b01a453152daf3cccaae4887b7924aab477515b0c height=345029 version=0x00000002 log2_work=82.307795 tx=60651588 date='2015-02-25T04:15:11Z' progress=0.068269 cache=282.1MiB(1545988txo)
2023-11-04T00:33:36Z UpdateTip: new best=00000000000000000cb4c22c9d9d466e8bcbd75cf498c147673519d528fe010c height=345030 version=0x00000002 log2_work=82.307844 tx=60652391 date='2015-02-25T04:30:41Z' progress=0.068270 cache=282.2MiB(1546275txo)
[... 291 lines cut ...]
ln_default_channels_total="0"
ln_default_ready="1"
ln_default_error_full=""
ln_default_tor="0"
ln_default_locked="0"
btc_default_error_short=""
btc_default_blocks_verified="345072"
btc_default_peers="10"
btc_default_running="1"
btc_default_sync_initial_done="0"
btc_default_port="8333"
btc_default_blocks_data_kb="35285920"
btc_default_blocks_headers="815184"
btc_default_synced="0"
btc_default_sync_initialblockdownload="1"
btc_default_sync_progress="0.06829316154073833"
btc_default_blocks_behind="470112"
btc_default_version="v25.0.0"
btc_default_error_full=""
btc_default_sync_percentage="683"
btc_default_mempool_transactions="0"
btc_default_ready="1"
btc_default_activated="1"
btc_default_online="1"
*** POSSIBLE ERROR REPORTS ***`

adevoss, 2023-11-04:
Same configuration (amd64 VM, v1.10.0 and 11.8)
but the info screen shows 100%

HannesMC, 2023-11-05:
The status today
[image]
>3900% .... nice...

rootzoll, 2024-05-10:
too be tested witth v1.11.1rc1 once branch vminstall is merged

rootzoll, 2024-07-08:
branch `vminstall` was merged

closed 2024-07-08 by rootzoll