raspiblitz/raspiblitz #4240: v.1.10.0 many restarts after update

romfordpele, 2023-11-08:
Hi this is a node I built a while ago (2020). Was running v1.7 for a loooong time and just updated to v1.10.0.

It worked for a day maybe but then I had issues and had to reboot. It never came back and would not unmount/mount HDD.

I am wondering if something happened to the required specs for the drive between the versions 1.7 and 1.10? Any ideas gratefully received!

Logs below:

RASPIBLITZ DEBUG LOGS
blitzversion: 1.10.0
chainnetwork: bitcoin / test
20:08:12 up 6 days, 23:44, 2 users, load average: 2.75, 2.63, 2.56

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
UNIT LOAD ACTIVE SUB DESCRIPTION
● bitcoind.service loaded failed failed Bitcoin daemon on mainnet
● lnd.service loaded failed failed Lightning Network Daemon
● logrotate.service loaded failed failed Rotate log files

LOAD = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB = The low-level unit activation state, values depend on unit type.
3 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Mon 2023-10-23 20:08:02 BST; 9s ago
Process: 3076302 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
Process: 3076336 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
CPU: 257ms

Oct 23 20:08:02 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 23 20:08:02 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Mon 2023-10-23 10:35:00 BST, ends at Mon 2023-10-23 20:08:12 BST. --
Oct 23 20:08:00 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 23 20:08:00 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.
Oct 23 20:08:00 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 3.
Oct 23 20:08:00 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet.
Oct 23 20:08:01 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Oct 23 20:08:01 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Oct 23 20:08:01 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 23 20:08:01 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.
Oct 23 20:08:01 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 4.
Oct 23 20:08:01 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet.
Oct 23 20:08:01 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Oct 23 20:08:02 raspberrypi bitcoind[3076336]: Error: Error reading configuration file: specified config file "/mnt/hdd/bitcoin/bitcoin.conf" could not be opened.
Oct 23 20:08:02 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Oct 23 20:08:02 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 23 20:08:02 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.
Oct 23 20:08:02 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Oct 23 20:08:02 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet.
Oct 23 20:08:02 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Oct 23 20:08:02 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 23 20:08:02 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
[... 204 lines cut ...]
ln_default_ready="0"
ln_default_running="0"
ln_default_error_short=""
ln_default_locked="0"
ln_default_sync_initial_done="1"
btc_default_error_short=""
btc_default_version="v25.0.0"
btc_default_running="0"
btc_default_blocks_behind="78"
btc_default_peers="2"
btc_default_activated="1"
btc_default_sync_initial_done="0"
btc_default_synced="0"
btc_default_error_full=""
btc_default_ready="0"
btc_default_sync_initialblockdownload="1"
btc_default_sync_progress="0.9980679252314205"
btc_default_mempool_transactions="0"
btc_default_sync_percentage="99.81"
btc_default_blocks_data_kb="575222624"
btc_default_port="8333"
btc_default_blocks_headers="811739"
btc_default_online="0"
btc_default_blocks_verified="811661"
*** POSSIBLE ERROR REPORTS ***

rootzoll, 2023-12-15:
An update from 1.7.0 to 1.10.0 is quite a big update jump and we normally just test the update from the previous version during development. So I would normally just recommend to jump miner versions like (1.7 > 1.8 > 1.9 > 1.10).

To fix your situation it might make sense to backup your LND data (MAINMENU > REPAIR > REPAIR-LND > BACKUP-LND) - check the rescue file that it really contains the LND data files and then setup the node fresh:

- make sure you have SEED WORDS & the Static-Channel-Backup also available (just in case)
- flash sd card with new v1.10.0 image
- on ssh setup login DONT run recover - see other options
- use FRESH SETUP and then choose to reuse blockchain
- During setup of LND - choose OLD LND and import LND rescue backup again

If this still does not work  .. do this again but dont use the old blockchain data and let SSD format fresh

Edifredi, 2023-12-27:
Hi romfordpele. Did you fix yore Node, i am asking because i have same issue (Blockchain progress: 5 restarts / lightning progress: prepare sync)

romfordpele, 2023-12-28:
Hi all, 

Sadly not @Edifredi! I'm still having trouble with this issue. I have done as @rootzoll suggested in fact I started from scratch using the existing blockchain data - I had no funds in LN anyway. It works ok for a few hours and then fails in the same way. I am trying to restore Specter Wallet balances which is taking days - I come to it to find it's dropped out again - I power the node down and restart, it works again for a while and then blockchain stops again. I wonder if it's a compatability issue with the FAT file with the older hardware? It's a RPi 4 been running since 2020.

romfordpele, 2023-12-28:
The debug logs look a bit more useful this time...

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / test
 20:18:12 up  1:03,  2 users,  load average: 1.72, 1.91, 2.32

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT             LOAD   ACTIVE SUB    DESCRIPTION
● bitcoind.service loaded failed failed Bitcoin daemon on mainnet
● lnd.service      loaded failed failed Lightning Network Daemon

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
2 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2023-12-28 19:56:53 GMT; 21min ago
    Process: 109178 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 109197 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
        CPU: 222ms

Dec 28 19:56:53 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Dec 28 19:56:53 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Wed 2023-12-27 06:55:33 GMT, ends at Thu 2023-12-28 20:18:12 GMT. --
Dec 28 19:56:52 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Dec 28 19:56:52 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Dec 28 19:56:52 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.
Dec 28 19:56:52 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Dec 28 19:56:52 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet.
Dec 28 19:56:52 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Dec 28 19:56:52 raspberrypi bitcoin.check.sh[109180]: /bin/chgrp: changing group of '/mnt/hdd/bitcoin': Read-only file system
Dec 28 19:56:52 raspberrypi bitcoin.check.sh[109182]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Dec 28 19:56:52 raspberrypi bitcoin.check.sh[109183]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Dec 28 19:56:52 raspberrypi bitcoin.check.sh[109184]: grep: /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Dec 28 19:56:52 raspberrypi bitcoin.check.sh[109185]: sed: can't read /mnt/hdd/bitcoin/bitcoin.conf: Input/output error
Dec 28 19:56:52 raspberrypi bitcoind[109197]: Error: Error reading configuration file: specified config file "/mnt/hdd/bitcoin/bitcoin.conf" could not be opened.
Dec 28 19:56:52 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Dec 28 19:56:52 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Dec 28 19:56:52 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.
Dec 28 19:56:53 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 6.
Dec 28 19:56:53 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet.
Dec 28 19:56:53 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Dec 28 19:56:53 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Dec 28 19:56:53 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-12-28T19:49:33Z Syncing txindex with block chain from height 670400
2023-12-28T19:50:31Z UpdateTip: new best=00000000000000000003aef0eb7c8c32528899344ce293eff9619ce411a9ee23 height=823236 version=0x33c86000 log2_work=94.629441 tx=943721096 date='2023-12-28T06:01:31Z' progress=0.999806 cache=63.3MiB(424031txo)
2023-12-28T19:51:39Z UpdateTip: new best=00000000000000000000d84a15f1caaad281d66e31feb7d7cab04fc67ec1539c height=823237 version=0x22762000 log2_work=94.629455 tx=943721374 date='2023-12-28T06:02:01Z' progress=0.999806 cache=65.0MiB(438532txo)
2023-12-28T19:52:11Z UpdateTip: new best=0000000000000000000282e04d00193476b2f87e5a92a758ffe73cba20bedaa4 height=823238 version=0x29238000 log2_work=94.629470 tx=943723544 date='2023-12-28T06:12:11Z' progress=0.999809 cache=66.3MiB(448465txo)
2023-12-28T19:52:11Z Syncing txindex with block chain from height 670401
[... 251 lines cut ...]
ln_default_error_full=""
ln_default_locked="0"
ln_default_online="0"
ln_default_version="0.16.4-beta"
ln_default_activated="1"
btc_default_peers="3"
btc_default_version="v25.0.0"
btc_default_activated="1"
btc_default_mempool_transactions="0"
btc_default_blocks_behind="43"
btc_default_sync_initial_done="0"
btc_default_online="0"
btc_default_running="0"
btc_default_blocks_data_kb="260"
btc_default_port="8333"
btc_default_sync_percentage="99.98"
btc_default_error_short=""
btc_default_ready="0"
btc_default_sync_initialblockdownload="1"
btc_default_blocks_headers="823291"
btc_default_blocks_verified="823248"
btc_default_synced="0"
btc_default_sync_progress="0.9998483581149491"
btc_default_error_full=""
*** POSSIBLE ERROR REPORTS ***

Edifredi, 2023-12-28:
Thanks for answer @romfordpele. My Node stuck since juni with my fund's in Lighting, and i can't find any Solution. Maybe some day.

openoms, 2024-01-22:
@romfordpele your logs are full of disk errors suggesting a hardware error

Boddei, 2024-04-17:
Hi, I'm facing the same issue.
Set up the raspi fresh with V1.10.0 and new 2TB SSD.
After few hours, during blockchain sync, it shows like 5M% HDD use.
sudo fdsik -l gives:
admin@192.xxx.xxx.xxx:~ ₿ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 29.71 GiB, 31902400512 bytes, 62309376 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6723c65e

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 62309375 61776896 29.5G 83 Linux

Disk /dev/zram0: 9.6 GiB, 10305163264 bytes, 2515909 sectors
Units: sectors of 1 * 4096 = 4096 bytes
Sector size (logical/physical): 4096 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

[image]

rootzoll, 2024-04-19:
Can you check if this error still persists with the newly released v1.11.0?

closed 2024-04-25 by rootzoll

SepticFrank, 2024-06-04:
> > Can you check if this error still persists with the newly released v1.11.0?
> 
> The issue is caused by the SSD overheating. I use a SanDisk Extreme Portable SSD 2 TB. The issue remains in v1.11.0 I solved it by putting the SSD in a plastic bag and in a cup of water for cooling. Now runs stable since several days.

Any advice to cool the SSD other than putting it in a plastic bag with water? I suspect mine is overheating as well in the enclosure.