raspiblitz/raspiblitz #4247: No Apps show up after Fatpack installation

rootzoll, 2023-11-14:
### Discussed in https://github.com/raspiblitz/raspiblitz/discussions/4244

I installed the latest Fatpack RaspiBlitz on my Raspi 4, the Bitcoin node is 100% syncrhonized for some days now.
Unfortunately, no Apps show up in the Web-UI. There is only this yellow rotating half-circle when I click on the Apps button.
Does anyone know what the problem is?</div>

cstenglein, 2023-11-14:
The blitz API should send a `installed_app_status` SSE message on startup.

Until the message arrives, there is the loading screen.

rootzoll, 2023-11-14:
TODO:
@rootzoll take a look at API how it can be made more stable
@cstenglein maybe do a fallback with an API request

fusion44, 2023-11-14:
The API actually tries to get the app status when sending the warmup data to new SSE clients:
https://github.com/fusion44/blitz_api/blob/9adcc866fb30502253070211604781fcca865c18/app/main.py#L316C13-L316C57

If it works on some machines and on some it doesn't, then it probably is a problem getting the status itself:
https://github.com/fusion44/blitz_api/blob/9adcc866fb30502253070211604781fcca865c18/app/apps/impl/raspiblitz.py#L43C8-L43C8

I think it'd be helpful to call the script status manually on failing machines:
`sh bonus.electrs.sh status`

daviogg, 2023-11-23:
Sorry, is there a workaround to solve this problem? I'm stuck too, I can't load the apps tab
 raspiblitz v 1.10 and with webUI version 1.3.0-dev

daviogg, 2023-11-27:
I did not initially install the lighting wallet components. After doing so, the Apps tab is loaded correctly

rootzoll, 2023-11-28:
TODO: Test v1.10.0 fatpack install with just Bitcoin

helishmopter, 2024-02-16:
Apologies for double posting this here again (just posted under Apps tab won't load on WebUI (#3608) #4012), but I'm really stuck with this issue..

Apps tab won't stop loading in webUI. No LN installed, just BTC. Blockchain 100 % synced. No other apps running. Tried re-installing 1.10. with same result (app tab not loading). Also attempted re-activating 'Blitz API + WebUI' in Node Settings & Options, with the result of being completely locked out of the webUI (502 Bad Gateway). Any help is much appreciated!

scarolan, 2024-03-02:
Same here - I'm not using LN, only BTC.  Blockchain is 100% synced, and no other apps running.  It just stays stuck on this spinny thing.  I'm on 1.9 but it looks like the problem also exists in 1.10.

[image]
[image]

rootzoll, 2024-03-19:
With the v1.11.0rc4 fatpack all looks good in the app section. Closing issue for release - please comment/reopen if you run into this error with v1.11.0rc4 fatpack or later.

closed 2024-03-19 by rootzoll