raspiblitz/raspiblitz #4259: [RP5] NVMe Board & Boot

rootzoll, 2023-11-17:
https://www.youtube.com/watch?v=EXWu4SUsaY8

kravens, 2023-11-19:
I've ordered the bottom (long) NVMe hat, so I'll be able to test with this thing soon 😄

openoms, 2023-11-22:
Handy WIP list of disks tested: https://pipci.jeffgeerling.com//hats/pineberry-pi-hatdrive-top.html

rootzoll, 2023-11-22:
This tutorial is also very helpfull in understanding how to be able to boot from NVMe:
https://www.jeffgeerling.com/blog/2023/nvme-ssd-boot-raspberry-pi-5

So for the future a FRESH SETUP could work like this:
- user boot up from sd card with fresh raspiblitz image
- new NVMe gets partion like: 65gb ext4 system, rest as data drive
- then sd card get cloned to nvme system partion (rpi-clone)
- sd card stays before NVMe in boot order
- sd card renders itself as un-bootable (see notes below)
- system reboots and runs/setups now from NVMe

And an UPDATE/RECOVER could then work like this:
- write a fresh image to the sd card and boot up
- because sd card is now bootable again it will start the new image
- sd card system will clone itself to NVMe system partion (rpi-clone)
- sd card renders itself as un-bootable (see notes below)
- system reboots and runs/recovers now from NVMe

--- SD CARD RENDERS ITSELF UNBOOTABLE --

It is possible to render a Raspberry Pi's SD card unbootable through a write action, which would cause the device to skip it in the boot order on the next restart. Here's how it can happen:

1. **Corrupting Boot Files**: If critical boot files on the SD card are overwritten or corrupted, the Raspberry Pi will not be able to boot from it. These files include the bootloader, kernel, or other essential files in the boot partition.

2. **Damaging the File System**: Writing actions that damage the file system structure (like the partition table or file allocation table) can make the SD card unbootable. This could happen due to improper formatting or partitioning actions.

3. **Overwriting Critical Configurations**: Altering certain configuration files (like `cmdline.txt` or `config.txt` in the boot partition) with incorrect settings can prevent the Raspberry Pi from booting.

4. **Hardware-Level Damage**: Although less common, a write action that causes excessive wear on the SD card could lead to hardware-level damage, making it unreadable.

5. **Writing Incorrect Boot Code**: Intentionally or accidentally writing incorrect or incompatible boot code to the boot sector of the SD card would result in a failure to boot.

If any of these scenarios occur, the Raspberry Pi's firmware will attempt to boot from the next available device in its boot order, such as a USB drive or network boot, if configured. It's important to handle SD card operations carefully to avoid unintentional corruption or damage. Regular backups and careful editing of critical files are recommended to mitigate these risks.

openoms, 2023-11-23:
Would be good to be able to keep the SDcard as a redundant backup target as well for things like the LND SCB, the CLN emergency.recover file and possibly the ligthning database backup.

Fun fact that my CLN node is running with the same SDcard with the database backup on it for 2 years now without issues. Still it would,be great if the failure of the backup device wouldn't mean that the node would become unbootable also.

So maybe could after flashing the OS to the NVMe partition the SDcard could be made to become unbootable by formatting to a simple FAT filesystem (readable by Windows) and use it as a backup target automatically.

rootzoll, 2023-12-08:
There will be an official M2 HAT for the RPi5 ... and a HAT+ standard
https://www.raspberrypi.com/news/double-standards/

rootzoll, 2023-12-11:
More Sites with info on NVMe Boot:

https://www.tomshardware.com/raspberry-pi/how-to-turbo-charge-your-raspberry-pi-5-with-an-nvme-boot-drive

https://docs.pineberrypi.com/

rootzoll, 2023-12-15:
In the first step for v1.11.0 we will only support the NVMe hats to provide tne NVMe as external data drive. Boot will the be targeted later.

Integrated the PineberryPi Hat and first test works good. Final Testing.

rootzoll, 2023-12-15:
BTW decided for now not to set the "dtparam=pciex1_gen=3" parameter. Sure an even faster NVMe would be nice, but by stable connection and most wide compatibility should be the default for now. 

```
Setting the dtparam=pciex1_gen=3 parameter in the Raspberry Pi 5 affects the speed at which the PCI Express (PCIe) lanes operate. By default, all PCIe lanes on the Raspberry Pi 5 operate at Gen 2.0 speeds, which is about 5 gigatransfers per second (GT/sec) per lane. This default setting is primarily due to the speed at which the board could be certified for PCI Express, as higher standards like 2.0 and 3.0 are considered high-speed interconnects and can cause problems with interference and signal issues on the board.

When you add the dtparam=pciex1_gen=3 line in the /boot/config.txt file, it upgrades the connection on the external PCIe connector to Gen 3.0 speeds, which are almost double the speed of Gen 2.0, at approximately 8 GT/sec. This change can significantly enhance the performance of devices connected to the external PCIe connector, such as NVMe SSDs or network cards. For example, a 10G ASUS Network Card can achieve speeds of 6 Gbps compared to about 3.5 Gbps under Gen 2.0, and a Kioxia XG8 NVMe SSD can achieve 900 MB/sec reads versus 450 MB/sec under Gen 2.0.

However, it's important to note that increasing the speed to Gen 3.0 can introduce certain issues, such as link errors, especially with devices that do not handle these errors gracefully. These issues can be compounded by factors like insertion loss from the flat FPC connection used in the Raspberry Pi 5. It's also worth mentioning that there's currently no way to change the default speed for the RP1 chip's 'internal' lanes in the Raspberry Pi 5, and the setting only applies to the external connector.

In summary, setting dtparam=pciex1_gen=3 can offer significant performance improvements for devices that can benefit from the increased speed and that are connected to the external PCIe connector. However, it's important to be aware of potential issues and compatibility concerns that might arise from operating at these higher speeds​
```

rootzoll, 2023-12-16:
Moved the boot section of this issue to #4324

neversleep404, 2024-01-08:
How would a migration from a USB SSD Drive to a new NVMe work?

is this correct?

1. The user initiates the RaspiBlitz Release Update/Recovery on the Raspberry Pi.
2. The user unplugs the SSD HDD, inserts the new NVMe hardware, and boots up from the SD card with a fresh RaspiBlitz image.
3. The new NVMe is partitioned as follows: 65GB ext4 for the system, rest as data drive.
4. The SD card is then cloned to the NVMe system partition using a tool like rpi-clone.
5. The SD card remains primary in the boot order.
6. The SD card is rendered unbootable.
7. Before the system reboots and runs/setups from the NVMe, the user plugs the USB SSD Drive in.
8. Open a terminal and enter `lsblk` or `sudo fdisk -l` to identify the drives and partitions. Note down the respective device identifiers (e.g., `/dev/sdb1` for the source partition on the USB SSD and `/dev/nvme0n1p2` for the target partition on the NVMe SSD).
9. Ensure no data is being written to the target partition. If it is mounted, unmount it.
10. Be aware that the `dd` command will overwrite all data on the target partition. Ensure all important data is backed up.
11. Use the `dd` command to clone the partition. The command might look something like this:
    ```bash
    sudo dd if=/dev/sdx1 of=/dev/nvmey1 bs=64K status=progress
    ```
    - `if=/dev/sdx1`: replace `/dev/sdx1` with the path to the source partition on the USB SSD.
    - `of=/dev/nvmey1`: replace `/dev/nvmey1` with the path to the target partition on the NVMe SSD.
    - `bs=64K`: stands for the block size being used. 64K is a common size, but you can adjust it as necessary.
    - `status=progress`: shows the progress of the copy operation.
12. Reboot and run the setup from the NVMe.

> So for the future a FRESH SETUP could work like this:
>
> * user boot up from sd card with fresh raspiblitz image
> * new NVMe gets partion like: 65gb ext4 system, rest as data drive
> * then sd card get [...]

rootzoll, 2024-02-25:
So far all good with the NVMe Hats. Closing for ReleaseCandidate.

closed 2024-02-25 by rootzoll