raspiblitz/raspiblitz #4267: Recover CLN onchain funds to Sparrow (HSM-to-XPRV-recovery-tool)

openoms, 2023-11-22:
BE careful this tool will expose the master private key of your CLN node in the terminal, but look  very useful for recovery purposes. 

https://github.com/dieqohc/HSM-to-XPRV-recovery-tool

Did not test, yet but panning to add to the CLN docs.

salenjak, 2023-12-03:
I am trying to test this, but without success.
What part of the hsm_secret needs to be pasted in hsm_secret file inside HSM-to-XPRV-recovery-tool folder?

What I do:
1. Decrypt hsm_secret
2. sudo -u bitcoin xxd /home/bitcoin/.lightning/bitcoin/hsm_secret
3. copy hsm_secret from raspiblitz (complete lines): 
00000000: x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  .x.1.x.1.x.1.
00000010: x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  1.x.1.x.1.1.x.
4. Open [HSM-to-XPRV-recovery-tool](https://github.com/dieqohc/HSM-to-XPRV-recovery-tool) folder 
5. paste above 2 lines inside HSM-to-XPRV-recovery-tool folder in the hsm_secret file 
6. Run python3  recovery.py

After that script will show me my XPRV and XPUB, but every time XPRV is incorrect.
I have tried many combinations of hsm secret. Few examples:
00000000: x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  .x.1.x.1.x.1.
00000010: x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  1.x.1.x.1.1.x. 

x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  .x.1.x.1.x.1.
x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  1.x.1.x.1.1.x.

x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1
x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1

00000000: x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1
00000010: x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1

x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1  x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1 x1x1

Any idea why this won't produce correct XPRV?

dieqohc, 2023-12-06:
> BE careful this tool will expose the master private key of your CLN node in the terminal, but look very useful for recovery purposes.
> 
> https://github.com/dieqohc/HSM-to-XPRV-recovery-tool
> 
> Did not test, yet but panning to add to the CLN docs.

Appreciate your support @openoms and @rootzoll, I do know printing this info to the terminal might be a risk and is not the best practice, but also wanted to keep it friendly and frictionless enough for those who are not too technical. Do you happen to know a way to make this output more secure? probably as an option for users who don't want to expose this info.

Besides the idea is that as soon as the XPRV is exposed withdraw all the funds immediately hopefully in a transaction without RBF. (Maybe I should emphasize this more on the docs.)