raspiblitz/raspiblitz #4290: RPI 5 stuck in the initial setup.

cdemkos, 2023-12-03:
After creating a new sdcard with raspiblitz-arm64-rpi-lean.img and starting the initial setup, the attached ssd drive is deleted and after restart retuns to the same initial setup window.

freerko, 2023-12-03:
I have the same issue, described it in [#4238](https://github.com/raspiblitz/raspiblitz/pull/4238)

neversleep404, 2023-12-04:
> After creating a new sdcard with raspiblitz-arm64-rpi-lean.img and starting the initial setup, the attached ssd drive is deleted and after restart retuns to the same initial setup window.

What version? I used raspiblitz-arm64-rpi-image-2023-11-23-9f23886 and raspiblitz-arm64-rpi-image-2023-11-24-7e66b82 and had no problems. But I migrate from a running PI 4 to the PI 5.

cdemkos, 2023-12-04:
I used the same image as you. I allready figured that this occurs only with a new setup. It seems to work well with a migratio from pi4 to pi5

neversleep404, 2023-12-04:
> I used raspiblitz-arm64-rpi-image-2023-11-24-7e66b82. I allready figured that this occurs only with a new setup. It seems to work well with a migration from pi4 to pi5

When you login with ssh and go to the console prompt (strg-c). 
Then df.
And if you have a mounted /mnt/hdd change in the root directory and show "ls".

[image]

cdemkos, 2023-12-04:
[image]
The ssd is connected, and when i run ubuntu on the rpi5 the partition is usable.

neversleep404, 2023-12-04:
ok. I don't see your SSD. 
What you can see it that there is no /dev/sda1
can you check with ubuntu if your SSD has a ext4 partition? You can use df -T
If not then format the SSD:

To format a storage device with the ext4 filesystem under Linux, you can use a command-line tool like `mkfs.ext4`. This process will erase all data on the device, so make sure you have backups of any important data before proceeding. Here's how to do it:

1. **Identify the Device**: First, you need to identify the storage device you want to format. You can list all connected storage devices and their partitions with the `lsblk` command. 

   ```bash
   lsblk
   ```

   Look for the device that corresponds to the one you want to format, such as `/dev/sdb` or `/dev/sdc`. Be very careful to choose the correct device, as formatting the wrong one will result in data loss.

2. **Unmount the Device**: If the device is mounted, you need to unmount it before formatting. Replace `/dev/sdx` with your actual device:

   ```bash
   sudo umount /dev/sdx
   ```

   If the device has multiple partitions (like `/dev/sdx1`, `/dev/sdx2`, etc.), unmount each of them.

3. **Format with ext4**: Use the `mkfs.ext4` command to format the device. Replace `/dev/sdx` with your device:

   ```bash
   sudo mkfs.ext4 /dev/sdx
   ```

   This command will format the entire device with the ext4 filesystem. If you want to format a specific partition, replace `/dev/sdx` with the partition identifier, like `/dev/sdx1`.

4. **Label the Filesystem** (Optional): You can also assign a label to the filesystem with the `-L` flag:

   ```bash
   sudo mkfs.ext4 -L mylabel /dev/sdx
   ```

   Replace `mylabel` with your desired filesystem label.

5. **Check the Filesystem** (Optional): After formatting, you can check the filesystem with:

   ```bash
   sudo e2fsck -f /dev/sdx
   ```

6. **Mount the Filesystem**: Finally, mount the filesystem. First, create a mount point (if it does not already exist):

   ```bash
   sudo mkdir /mnt/mydrive
   ```

   Then mount the filesystem:

   ```bash
   sudo mount /dev/sdx /mnt/mydrive
   ```

   Replace `/mnt/mydrive` with your preferred mount point.

Remember to replace `/dev/sdx` with your actual device identifier throughout these steps. Also, always double-check the device identifier before executing these commands to avoid accidental data loss.

And then try again with the RASPIBLITZ card.

cdemkos, 2023-12-05:
[image]

i can see the hd after formating to ext4 but after i run the setup it wipes the partition, and the setup starts all over

neversleep404, 2023-12-05:
Okay, I can't help further. I would now try setting up a RaspiBlitz with a PI 4 and then put the SSD into the PI 5.

cdemkos, 2023-12-05:
thank you for your time and advice, But in regard of further expansion and stabilisation of the raspiblitz project i prefer to have a working setup procedure for the rpi5.

neversleep404, 2023-12-05:
> i prefer to have a working setup procedure for the rpi5.

me too. But it was not in vain. We found that the hard drive is formatted correctly, but then it is reformatted with every new start.

cdemkos, 2023-12-09:
I tried to run the setup, with the raspiblitz-arm64-rpi-image-2023-12-08-30d2e69 unfortunatly with the same loop result
[image]
[image]
[image]

cdemkos, 2023-12-10:
With  raspiblitz-arm64-rpi-image-2023-12-10-52b517a
The problem with ECDSA key fingerprint assigning is solved and does not channge each setup loop.
Allso the wiping of the partition on the extended drive does not happen anymore.
Setup loop still persistent.

Der-Zett, 2023-12-11:
I think the problem is in the blitz.datadrive.sh script in the FORMAT EXT4 or BTRFS section.
I have noticed that sfdisk, wipefs and parted call ioctl() to inform the kernel about changes to the filesystem and this always leads to an immediate reboot and thus the loop in the setup.
sfdisk can be prevented from triggering a reboot with -no-tell-kernel and with partprobe the changes can be read without reboot. Unfortunately I have not found a way to prevent wipefs and parted from doing this.

I confirm this problem.

cdemkos, 2023-12-11:
raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.11.0rc0
Mon 11 Dec 06:56:40 GMT 2023
***********************************************
# *** /home/admin/config.scripts/blitz.ssh.sh checkrepair
# countKeyFiles(6)
# OK: SSHD RUNNING
## INIT raspiblitz.info
baseimage=raspios_arm64
cpu=aarch64
blitzapi=off
displayClass=headless
displayType=
setupPhase=setup
setupStep=0
fsexpanded=0
state=starting
btc_mainnet_sync_initial_done=0
btc_testnet_sync_initial_done=0
btc_signet_sync_initial_done=0
ln_lnd_mainnet_sync_initial_done=0
ln_lnd_testnet_sync_initial_done=0
ln_lnd_signet_sync_initial_done=0
ln_cl_mainnet_sync_initial_done=0
ln_cl_testnet_sync_initial_done=0
ln_cl_signet_sync_initial_done=0
## INIT RaspiBlitz Cache ... wait background.scan.service to finish first scan loop
- waiting for background.scan.service --> systemscan_runtime(0)
OK - logs are at 52 MB - within safety limit
Waiting for HDD/SSD ...
total 624
drwxr-xr-x   4 root root   4096 Dec 10 13:19 .
drwxr-xr-x 133 root root  12288 Dec 11 06:56 ..
-rw-r--r--   1 root root 573928 Sep 23 23:11 moduli
-rw-r--r--   1 root root   1650 Sep 23 23:11 ssh_config
drwxr-xr-x   2 root root   4096 Sep 23 23:11 ssh_config.d
-rw-r--r--   1 root root   3223 Sep 23 23:11 sshd_config
drwxr-xr-x   2 root root   4096 Dec 10 13:19 sshd_config.d
-rw-------   1 root root    505 Dec 10 13:19 ssh_host_ecdsa_key
-rw-r--r--   1 root root    173 Dec 10 13:19 ssh_host_ecdsa_key.pub
-rw-------   1 root root    399 Dec 10 13:19 ssh_host_ed25519_key
-rw-r--r--   1 root root     93 Dec 10 13:19 ssh_host_ed25519_key.pub
-rw-------   1 root root   2590 Dec 10 13:19 ssh_host_rsa_key
-rw-r--r--   1 root root    565 Dec 10 13:19 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    342 Feb 28  2020 ssh_import_id
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate:
isMounted: 0
hddCandidate: sda
HDD/SSD connected: sda
No HDMI switch found.
No FS EXPAND needed. needsExpansion(0) fsexpanded(0)
No WIFI RESTORE because no /var/cache/raspiblitz/hdd-inspect/wpa_supplicant.conf
No SSH SERVER CERTS RESTORE because no /var/cache/raspiblitz/hdd-inspect
No SSHRESET switch found.
No DISPLAY RESTORE because no /var/cache/raspiblitz/hdd-inspect/raspiblitz.conf
No UASP FIX needed
isMounted: 0
HDD already part of system: 0
Temp mounting (1) data drive
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df 
Filesystem      1K-blocks    Used  Available Use% Mounted on
udev              4080704       0    4080704   0% /dev
tmpfs              824288    5264     819024   1% /run
/dev/mmcblk0p2   30099228 5158484   23391864  19% /
tmpfs             4121360      32    4121328   1% /dev/shm
tmpfs                5120       0       5120   0% /run/lock
/dev/mmcblk0p1     522232  164932     357300  32% /boot/firmware
tmpfs               32768       0      32768   0% /var/cache/raspiblitz
tmpfs              824272     144     824128   1% /run/user/1000
/dev/sda1      1921693896      28 1824003452   1% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda
└─sda1      ext4
mmcblk0
├─mmcblk0p1 vfat   bootfs
└─mmcblk0p2 ext4   rootfs
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
hddRaspiVersion=''
isSMART=0
isSSD=1
datadisk='sda'
datapartition='sda1'
hddCandidate='sda'
hddPartitionCandidate='sda1'
hddBlocksBitcoin=0
hddBytes=2000364240896
hddGigaBytes=1862
hddUsedInfo='28K (1%)'
hddDataFreeBytes=1867779534848
hddDataFreeKB=1824003452
hddDataFreeGB=1739
hddAdapterUSB=''
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

Displaying Info Message: Please start Setup
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION

rootzoll, 2023-12-13:
@Der-Zett @cdemkos can you try if the following RaspberryPi image (Download under Artifacts) fixes this:
https://github.com/raspiblitz/raspiblitz/actions/runs/7201966440

cdemkos, 2023-12-14:
[image]

Der-Zett, 2023-12-14:
I can confirm that the handling of hard disks works properly. Can you explain roughly what you have changed in the new image, or what exactly the error was?

rootzoll, 2023-12-14:
I was fixing the HDD/SSD detection in general and then the reboot disappeared in the process - so it seems that an empty hdd var somewhere (empty string) triggered the reboot.

freerko, 2023-12-15:
I tried the newest  arm64-rpi-lean-image-build (https://github.com/raspiblitz/raspiblitz/actions/runs/7216383851) and now the initial setup works on my RPi5 as well, thanks!

cdemkos, 2023-12-20:
i can confirm this problem isnt present any more with Version 1.11.0rc1
BTC Fullnode. Lightning + Tor completle synced.

closed 2024-02-25 by rootzoll