raspiblitz/raspiblitz #4291: Not possible to run two USB SSD simultanously

cdemkos, 2023-12-03:
When two external USB SSD (one working drive one data storage drive) drives are connected to the RPI5 the initial setup wipes both USB SSD drives

rootzoll, 2023-12-04:
The setup at the moment does not support two SSDs connected. This was because the every RPi below 5 was not able to deliver enough stable power for two USB SSDs and setups with external Power Delivery were just an edge case use. 

Maybe the RPi5 now is able to support more than one USB SSD - but if that is 100% stable with all SSD adapters & models is still to be tested.

Instead of having two SSDs sepereating OS & Data ... I think still one drive with two partitions would then make more sense ... a second SSD would then make more sense in an optional mirrow-RAID setup from my perspective. 

So this is more of a reasearch topic. If somebody wants to test & report how situation changed with RPi5, please collect that all here in the comments.

cdemkos, 2023-12-04:
The RPI5 has a new chip to be able to handle the load, unlike the RPI4  (RP1 I/O controller)
[image]

rootzoll, 2025-05-12:
RaspiBlitz v1.12.0 will support a more flexible drive layout from the software side - allowing boot from NVMe and seperation of storage & data drive - but mostly trageted for Proxmox - this is the new default for raspberryPi:

[image]

Closing this issue.

closed 2025-05-12 by rootzoll