raspiblitz/raspiblitz #4298: KeyError: 'amount_msat' Got an invoice but could not parse the data: 'amount_msat'

rootzoll, 2023-12-05:
### Discussed in https://github.com/raspiblitz/raspiblitz/discussions/4286

<div type='discussions-op-text'>

<sup>Originally posted by **neversleep404** December  1, 2023</sup>
This error occurs many times per second. Can I fix this, or is it a bug in the code?"

sudo journalctl -u blitzapi --since "2023-12-01 09:08:00"

(...)

Dec 01 09:08:04 HerzAusGold python[1151]: 
Dec 01 09:08:04 HerzAusGold python[1151]: 2023-12-01 09:08:04 | ❌ | cln_jrpc.py:678 | Got an invoice but could not parse the data: 'amount_msat'
Dec 01 09:08:04 HerzAusGold python[1151]: 2023-12-01 09:08:04 | ℹ️ | cln_jrpc.py:967 | Refreshing pay_index
Dec 01 09:08:04 HerzAusGold python[1151]: 2023-12-01 09:08:04 | ❌ | cln_jrpc_blitz.py:104 | An error has been caught in function 'list_invoices', process 'MainProc>
Dec 01 09:08:04 HerzAusGold python[1151]: Traceback (most recent call last):
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/impl/cln_jrpc.py", line 673, in listen_invoices
Dec 01 09:08:04 HerzAusGold python[1151]:     i = Invoice.from_cln_json(data)
Dec 01 09:08:04 HerzAusGold python[1151]:         │       │             └ {'label': 'keysend-1699837071.985357627', 'bolt11': 'lnbc1pj4zay0sp5404tpuuxxqrfsx848g632t>
Dec 01 09:08:04 HerzAusGold python[1151]:         │       └ <classmethod(<function Invoice.from_cln_json at 0x7fff04b0b100>)>
Dec 01 09:08:04 HerzAusGold python[1151]:         └ <class 'app.lightning.models.Invoice'>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/models.py", line 706, in from_cln_json
Dec 01 09:08:04 HerzAusGold python[1151]:     amt = parse_cln_msat(i["amount_msat"])
Dec 01 09:08:04 HerzAusGold python[1151]:           │              └ {'label': 'keysend-1699837071.985357627', 'bolt11': 'lnbc1pj4zay0sp5404tpuuxxqrfsx848g632tw9s26>
Dec 01 09:08:04 HerzAusGold python[1151]:           └ <function parse_cln_msat at 0x7fff04b08540>
Dec 01 09:08:04 HerzAusGold python[1151]: KeyError: 'amount_msat'
Dec 01 09:08:04 HerzAusGold python[1151]: During handling of the above exception, another exception occurred:
Dec 01 09:08:04 HerzAusGold python[1151]: Traceback (most recent call last):
Dec 01 09:08:04 HerzAusGold python[1151]:   File "<frozen runpy>", line 198, in _run_module_as_main
Dec 01 09:08:04 HerzAusGold python[1151]:   File "<frozen runpy>", line 88, in _run_code
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/__main__.py", line 4, in <module>
Dec 01 09:08:04 HerzAusGold python[1151]:     uvicorn.main()
Dec 01 09:08:04 HerzAusGold python[1151]:     │       └ <Command main>
Dec 01 09:08:04 HerzAusGold python[1151]:     └ <module 'uvicorn' from '/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/__init__.py'>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
Dec 01 09:08:04 HerzAusGold python[1151]:     return self.main(*args, **kwargs)
Dec 01 09:08:04 HerzAusGold python[1151]:            │    │     │       └ {}
Dec 01 09:08:04 HerzAusGold python[1151]:            │    │     └ ()
Dec 01 09:08:04 HerzAusGold python[1151]:            │    └ <function BaseCommand.main at 0x7fff06690c20>
Dec 01 09:08:04 HerzAusGold python[1151]:            └ <Command main>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 1055, in main
Dec 01 09:08:04 HerzAusGold python[1151]:     rv = self.invoke(ctx)
Dec 01 09:08:04 HerzAusGold python[1151]:          │    │      └ <click.core.Context object at 0x7fff073fb150>
Dec 01 09:08:04 HerzAusGold python[1151]:          │    └ <function Command.invoke at 0x7fff06690040>
Dec 01 09:08:04 HerzAusGold python[1151]:          └ <Command main>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
Dec 01 09:08:04 HerzAusGold python[1151]:     return ctx.invoke(self.callback, **ctx.params)
Dec 01 09:08:04 HerzAusGold python[1151]:            │   │      │    │           │   └ {'port': 11111, 'host': '0.0.0.0', 'root_path': '/api', 'app': 'app.main:app'>
Dec 01 09:08:04 HerzAusGold python[1151]:            │   │      │    │           └ <click.core.Context object at 0x7fff073fb150>
Dec 01 09:08:04 HerzAusGold python[1151]:            │   │      │    └ <function main at 0x7fff065b54e0>
Dec 01 09:08:04 HerzAusGold python[1151]:            │   │      └ <Command main>
Dec 01 09:08:04 HerzAusGold python[1151]:            │   └ <function Context.invoke at 0x7fff066916c0>
Dec 01 09:08:04 HerzAusGold python[1151]:            └ <click.core.Context object at 0x7fff073fb150>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 760, in invoke
Dec 01 09:08:04 HerzAusGold python[1151]:     return __callback(*args, **kwargs)
Dec 01 09:08:04 HerzAusGold python[1151]:                        │       └ {'port': 11111, 'host': '0.0.0.0', 'root_path': '/api', 'app': 'app.main:app', 'uds': Non>
Dec 01 09:08:04 HerzAusGold python[1151]:                        └ ()
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/main.py", line 404, in main
Dec 01 09:08:04 HerzAusGold python[1151]:     run(
Dec 01 09:08:04 HerzAusGold python[1151]:     └ <function run at 0x7fff06727100>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/main.py", line 569, in run
Dec 01 09:08:04 HerzAusGold python[1151]:     server.run()
Dec 01 09:08:04 HerzAusGold python[1151]:     │      └ <function Server.run at 0x7fff06726ac0>
Dec 01 09:08:04 HerzAusGold python[1151]:     └ <uvicorn.server.Server object at 0x7fff06594810>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/server.py", line 60, in run
Dec 01 09:08:04 HerzAusGold python[1151]:     return asyncio.run(self.serve(sockets=sockets))
Dec 01 09:08:04 HerzAusGold python[1151]:            │       │   │    │             └ None
Dec 01 09:08:04 HerzAusGold python[1151]:            │       │   │    └ <function Server.serve at 0x7fff06726b60>
Dec 01 09:08:04 HerzAusGold python[1151]:            │       │   └ <uvicorn.server.Server object at 0x7fff06594810>
Dec 01 09:08:04 HerzAusGold python[1151]:            │       └ <function run at 0x7fff06b5cb80>
[... 36 lines cut ...]
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/impl/specializations/cln_jrpc_blitz.py", line 176, in listen_invoices
Dec 01 09:08:04 HerzAusGold python[1151]:     async for i in super().listen_invoices():
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/impl/cln_jrpc.py", line 681, in listen_invoices
Dec 01 09:08:04 HerzAusGold python[1151]:     await self._refresh_invoice_sub(True)
Dec 01 09:08:04 HerzAusGold python[1151]:           │    └ <function LnNodeCLNjRPC._refresh_invoice_sub at 0x7fff040899e0>
Dec 01 09:08:04 HerzAusGold python[1151]:           └ <app.lightning.impl.specializations.cln_jrpc_blitz.LnNodeCLNjRPCBlitz object at 0x7fff0499a510>
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/impl/cln_jrpc.py", line 969, in _refresh_invoice_sub
Dec 01 09:08:04 HerzAusGold python[1151]:     invoices = await self.list_invoices(
Dec 01 09:08:04 HerzAusGold python[1151]:                      │    └ <function LnNodeCLNjRPCBlitz.list_invoices at 0x7fff0408a200>
Dec 01 09:08:04 HerzAusGold python[1151]:                      └ <app.lightning.impl.specializations.cln_jrpc_blitz.LnNodeCLNjRPCBlitz object at 0x7fff0499a510>
Dec 01 09:08:04 HerzAusGold python[1151]: > File "/home/blitzapi/blitz_api/app/lightning/impl/specializations/cln_jrpc_blitz.py", line 104, in list_invoices
Dec 01 09:08:04 HerzAusGold python[1151]:     return await super().list_invoices(
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/impl/cln_jrpc.py", line 307, in list_invoices
Dec 01 09:08:04 HerzAusGold python[1151]:     invoices.append(Invoice.from_cln_json(i))
Dec 01 09:08:04 HerzAusGold python[1151]:     │        │      │       │             └ {'label': 'keysend-1699837071.985357627', 'bolt11': 'lnbc1pj4zay0sp5404tpuuxxq>
[... cut ...]
Dec 01 09:08:04 HerzAusGold python[1151]:   File "/home/blitzapi/blitz_api/app/lightning/models.py", line 706, in from_cln_json
Dec 01 09:08:04 HerzAusGold python[1151]:     amt = parse_cln_msat(i["amount_msat"])
Dec 01 09:08:04 HerzAusGold python[1151]:           │              └ {'label': 'keysend-1699837071.985357627', 'bolt11': 'lnbc1pj4zay0sp5404tpuuxxqrfsx848g632tw9s26>
Dec 01 09:08:04 HerzAusGold python[1151]:           └ <function parse_cln_msat at 0x7fff04b08540>
Dec 01 09:08:04 HerzAusGold python[1151]: KeyError: 'amount_msat'
</div>

rootzoll, 2023-12-05:
@fusion44 can you take a look at this

neversleep404, 2023-12-05:
> I'd consider this a bug in the code at the moment.
> 
> @neversleep404 When you manually decode an invoice via command line what output do you get?
> 
> Command: `lightning-cli decode lnbc154...`
> 

The output of my log file is likely being truncated due to the configuration of the logging system.
I got only:
Dec 05 22:18:00 HerzAusGold python[1152]:         │       │             └ {'label': 'keysend-1699837071.985357627', 'bolt11': 'lnbc1pj4zay0sp5404tpuuxxqrfsx848g632tw9s26tvs5e69je409uva7pfw82duqspp5a7...
Dec 05 22:18:00 HerzAusGold python[1152]:         │       └ <classmethod(<function Invoice.from_cln_json at 0x7fff56cbb100>)>

and  then (with or without ...)   lightning-cli decode lnbc1pj4zay0sp5404tpuuxxqrfsx848g632tw9s26tvs5e69je409uva7pfw82duqspp5a7...
{
   "code": -32602,
   "message": "string: Bad bech32 string: invalid token '\"lnbc1pj4zay0sp5404tpuuxxqrfsx848g632tw9s26tvs5e69je409uva7pfw82duqspp5a7...\"'"
}

I tried increase the Width of the Terminal Window etc. but it did't help.
Any suggestions?

neversleep404, 2023-12-06:
Dec 06 20:05:02 HerzAusGold python[1152]: 2023-12-06 20:05:01 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Dec 06 20:05:07 HerzAusGold python[1152]: 2023-12-06 20:05:07 | ℹ️ | cln_jrpc.py:112 | CLN ConnectionRefusedError. Retrying in 10 seconds.
Dec 06 20:05:11 HerzAusGold python[1152]: 2023-12-06 20:05:11 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Dec 06 20:05:17 HerzAusGold python[1152]: 2023-12-06 20:05:17 | ℹ️ | cln_jrpc.py:112 | CLN ConnectionRefusedError. Retrying in 10 seconds.
Dec 06 20:05:22 HerzAusGold python[1152]: 2023-12-06 20:05:22 | ✔️ | service.py:39 | Bitcoin repository initialized
Dec 06 20:05:27 HerzAusGold python[1152]: 2023-12-06 20:05:27 | ℹ️ | cln_jrpc.py:112 | CLN ConnectionRefusedError. Retrying in 10 seconds.
Dec 06 20:05:37 HerzAusGold python[1152]: 2023-12-06 20:05:37 | ℹ️ | cln_jrpc.py:117 | Setting up invoice waitanyinvoice subscription.
Dec 06 20:05:37 HerzAusGold python[1152]: 2023-12-06 20:05:37 | ℹ️ | cln_jrpc.py:967 | Refreshing pay_index
Dec 06 20:05:37 HerzAusGold python[1152]: 2023-12-06 20:05:37 | ❌ | cln_jrpc_blitz.py:104 | An error has been caught in function 'list_invoices', process 'MainProcess' (1152), thread 'Mai>
Dec 06 20:05:37 HerzAusGold python[1152]: Traceback (most recent call last):
Dec 06 20:05:37 HerzAusGold python[1152]:   File "<frozen runpy>", line 198, in _run_module_as_main
Dec 06 20:05:37 HerzAusGold python[1152]:   File "<frozen runpy>", line 88, in _run_code
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/__main__.py", line 4, in <module>
Dec 06 20:05:37 HerzAusGold python[1152]:     uvicorn.main()
Dec 06 20:05:37 HerzAusGold python[1152]:     │       └ <Command main>
Dec 06 20:05:37 HerzAusGold python[1152]:     └ <module 'uvicorn' from '/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/__init__.py'>
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
Dec 06 20:05:37 HerzAusGold python[1152]:     return self.main(*args, **kwargs)
Dec 06 20:05:37 HerzAusGold python[1152]:            │    │     │       └ {}
Dec 06 20:05:37 HerzAusGold python[1152]:            │    │     └ ()
Dec 06 20:05:37 HerzAusGold python[1152]:            │    └ <function BaseCommand.main at 0x7fff08490c20>
Dec 06 20:05:37 HerzAusGold python[1152]:            └ <Command main>
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 1055, in main
Dec 06 20:05:37 HerzAusGold python[1152]:     rv = self.invoke(ctx)
Dec 06 20:05:37 HerzAusGold python[1152]:          │    │      └ <click.core.Context object at 0x7fff0920b150>
Dec 06 20:05:37 HerzAusGold python[1152]:          │    └ <function Command.invoke at 0x7fff08490040>
Dec 06 20:05:37 HerzAusGold python[1152]:          └ <Command main>
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
Dec 06 20:05:37 HerzAusGold python[1152]:     return ctx.invoke(self.callback, **ctx.params)
Dec 06 20:05:37 HerzAusGold python[1152]:            │   │      │    │           │   └ {'port': 11111, 'host': '0.0.0.0', 'root_path': '/api', 'app': 'app.main:app', 'uds': None, 'fd': Non>
Dec 06 20:05:37 HerzAusGold python[1152]:            │   │      │    │           └ <click.core.Context object at 0x7fff0920b150>
Dec 06 20:05:37 HerzAusGold python[1152]:            │   │      │    └ <function main at 0x7fff083b54e0>
Dec 06 20:05:37 HerzAusGold python[1152]:            │   │      └ <Command main>
Dec 06 20:05:37 HerzAusGold python[1152]:            │   └ <function Context.invoke at 0x7fff084916c0>
Dec 06 20:05:37 HerzAusGold python[1152]:            └ <click.core.Context object at 0x7fff0920b150>
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/click/core.py", line 760, in invoke
Dec 06 20:05:37 HerzAusGold python[1152]:     return __callback(*args, **kwargs)
Dec 06 20:05:37 HerzAusGold python[1152]:                        │       └ {'port': 11111, 'host': '0.0.0.0', 'root_path': '/api', 'app': 'app.main:app', 'uds': None, 'fd': None, 'reload':>
Dec 06 20:05:37 HerzAusGold python[1152]:                        └ ()
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/main.py", line 404, in main
Dec 06 20:05:37 HerzAusGold python[1152]:     run(
Dec 06 20:05:37 HerzAusGold python[1152]:     └ <function run at 0x7fff08527100>
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/main.py", line 569, in run
Dec 06 20:05:37 HerzAusGold python[1152]:     server.run()
Dec 06 20:05:37 HerzAusGold python[1152]:     │      └ <function Server.run at 0x7fff08526ac0>
Dec 06 20:05:37 HerzAusGold python[1152]:     └ <uvicorn.server.Server object at 0x7fff08394810>
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/venv/lib/python3.11/site-packages/uvicorn/server.py", line 60, in run
Dec 06 20:05:37 HerzAusGold python[1152]:     return asyncio.run(self.serve(sockets=sockets))
Dec 06 20:05:37 HerzAusGold python[1152]:            │       │   │    │             └ None
[... cut ...]
Dec 06 20:05:37 HerzAusGold python[1152]:   File "/home/blitzapi/blitz_api/app/lightning/models.py", line 706, in from_cln_json
Dec 06 20:05:37 HerzAusGold python[1152]:     amt = parse_cln_msat(i["amount_msat"])
Dec 06 20:05:37 HerzAusGold python[1152]:           │              └ {'label': 'keysend-1699837071.985357627', 'bolt11': 'lnbc1pj4zay0sp5404tpuuxxqrfsx848g632tw9s26tvs5e69je409uva7pfw82duq>
Dec 06 20:05:37 HerzAusGold python[1152]:           └ <function parse_cln_msat at 0x7fff06908540>
Dec 06 20:05:37 HerzAusGold python[1152]: KeyError: 'amount_msat'

fusion44, 2023-12-07:
OK, seems like I CLN doesn't provide the `amount_msat` at all when decoding a keysend payment with a value of 0. We need to fix that in the backend. It's clear for me what todo, thank you. Will be fixed in the next version.

neversleep404, 2023-12-07:
>  Will be fixed in the next version.
It was my pleasure to help. If there's anything I can assist you with, please let me know