raspiblitz/raspiblitz #4301: Update to BitcoinCore 26.0

rootzoll, 2023-12-06:
see https://bitcoincore.org/bin/bitcoin-core-26.0/

marssystems, 2023-12-07:
I upgraded my Rpi4 production and dev nodes as well as my Rpi5 test node (on Rasbian 12), to Bitcoin Core 26.0 - no issues (so far).

redmac68, 2023-12-07:
Apologies if this the wrong place for this comment, but I have tried to update Bitcoin Core to version 26.0 on two rpi4 raspiblitz nodes ... one running raspiblitz 1.8.0 and the other running raspiblitz 1.10.0. Both update attempts eventually failed after this message:

error: Data directory "/mnt/hdd/bitcoin" contains a "bitcoin.conf" file which is ignored, because a different configuration file "/home/admin/.bitcoin/bitcoin.conf" from data directory "/home/admin/.bitcoin" is being used instead. Possible ways to address this would be to:
- Delete or rename the "bitcoin.conf" file in data directory "/mnt/hdd/bitcoin".
- Change datadir= or conf= options to specify one configuration file, not two, and use includeconf= to include any other configuration files.
- Set allowignoredconf=1 option to treat this condition as a warning, not an error.

# BUILD FAILED --> Was not able to install bitcoind version(26.0)

openoms, 2023-12-07:
Can confirm, update to v26.0 works, but displays the error above.

helishmopter, 2024-02-14:
Same issue as above, update to v26.0 displays error, but it seems to work after reboot. Raspiblitz v 1.10, fresh install, 100 % synced, BTC only, no LN. I'm new to this, is this reason for concern?

helishmopter, 2024-02-16:
Getting the same error when updating from v25 to v26. It's displayed correctly as v26 after reboot. Can this be ignored or is that of any concern?

openoms, 2024-02-16:
> Getting the same error when updating from v25 to v26. Version is displayed correctly as v26 after reboot. Can this be ignored or is that of any concern?

safe to ignore

rootzoll, 2024-02-25:
So far all good. Closing for ReleaseCandidate.

closed 2024-02-25 by rootzoll