raspiblitz/raspiblitz #4304: [Research] Can RPi5 Power-Button be used to trigger shutdown?

rootzoll, 2023-12-07:
See if the PRi5 power button works like the optional GPIO pins buttons - see: https://chat.openai.com/share/60f9ce66-67e7-444b-8d8c-7fad893e05cc or if there is some other way to utiluze it.

openoms, 2023-12-09:
Yes, it works on my RPi5. With systemd service run:
```
Before=halt.target shutdown.target reboot.target
```
the safe shutdown script can be run also when the poweroff is triggered visa the button

https://github.com/raspiblitz/raspiblitz/issues/3114#issuecomment-1837191811

openoms, 2023-12-09:
From:
https://forums.raspberrypi.com/viewtopic.php?t=359649
from the power button and even from the touchscreen the events can be picked up with:
```
sudo su -
# install evtest
apt install -y evtest

# watch the power button
evtest --grab /dev/input/event0

# touchscreen
evtest --grab /dev/input/event1
```

This works:

```
# place in /etc/systemd/logind.conf
HandlePowerKey=ignore

# restart logind
sudo systemctl restart systemd-logind

# in the sudoers file (sudo visudo)
# add permission for nobody to use sudo with the shutdown script
nobody ALL=(ALL) NOPASSWD: sudo /home/admin/config.scripts/blitz.shutdown.sh

# create /etc/triggerhappy/triggers.d/powerbutton.conf
echo "KEY_POWER    1    "/home/admin/config.scripts/blitz.shutdown.sh trigger" | 
  sudo tee -a sudo nano /etc/triggerhappy/triggers.d/powerbutton.conf

# restart triggerhappy
sudo systemctl  restart triggerhappy

# watch triggerhappy
sudo journalctl -fu triggerhappy

# press do power button to initiate the safe shutdown
```

openoms, 2023-12-09:
A quick script to test, aimed to be part of the `build_sdcard.sh` later.
Works on the RPi5 for me. Would like to test on amd64 as well (and maybe someone has an Rpi4 with a power button set):

```
# *** SAFE SHUTDOWN ***
# Modify /etc/systemd/logind.conf
if ! grep "HandlePowerKey=ignore" /etc/systemd/logind.conf; then
  echo "HandlePowerKey=ignore" | sudo tee /etc/systemd/logind.conf
fi
sudo systemctl restart systemd-logind

SUDOERS_LINE="nobody ALL=(ALL) NOPASSWD: /home/admin/config.scripts/blitz.shutdown.sh"
# Check if the line already exists in the sudoers file
if ! grep -Fxq "$SUDOERS_LINE" /etc/sudoers; then
  echo "Adding line to sudoers file..."
  echo "$SUDOERS_LINE" | sudo tee -a /etc/sudoers > /dev/null
fi

# Create Triggerhappy trigger file
echo 'KEY_POWER    1    "/home/admin/config.scripts/blitz.shutdown.sh trigger"' |
  sudo tee /etc/triggerhappy/triggers.d/powerbutton.conf
sudo systemctl restart triggerhappy

echo "# watch: sudo journalctl -fu triggerhappy"
echo "# press the power button to test"
```

rootzoll, 2023-12-13:
PR merged up for final testing in next release candidate

openoms, 2024-01-01:
The restart command causes poweroff:
```
₿ restart
Command to restart your RaspiBlitz
Execute the blitz command 'restart'? (y/n): y

Raspberry Pi 5 Model B Rev 1.0
When shutdown is complete the activity light turns red and the LCD turns white on the Raspberry Pi 5 Model B Rev 1.0.
It will then reboot again automatically.
-----------------------------------------------
stopping lightningd - please wait ..
stopping slightningd - please wait ..
stop bitcoind (1) - please wait ..
Bitcoin Core stopping
stop bitcoind (2) - please wait ..
# Killing the processes using /mnt/hdd
# Stop tor with: 'kill -SIGTERM 1265'
# Attempt to unmount /mnt/hdd
umount: /mnt/hdd: target is busy.
starting shutdown ...

Broadcast message from root@raspberrypi on pts/7 (Mon 2024-01-01 18:32:10 GMT):      "raspberrypi" 18:32 01-Jan-24

The system will reboot now!

Broadcast message from root@raspberrypi on pts/7 (Mon 2024-01-01 18:32:10 GMT):

The system will reboot now!

WARN: Missing $DBUS_SESSION_BUS_ADDRESS .. 
RUNNING FALLBACK REBOOT .. 
```
will need t check, might be because of the FALLBACK REBOOT

openoms, 2024-01-02:
ok, restart works on the RPI5 using the latest build

rootzoll, 2024-03-19:
Tested multiple times. This Button is a lifesaver. Closing issue.

closed 2024-03-19 by rootzoll