raspiblitz/raspiblitz #4305: LCD is not working on manual build

rootzoll, 2023-12-07:
When I build the image from dev on RaspiOS Bookwork I dont the the LCD to work - it stays white. Doesnt matter if i buidl the sd card with "lcd" or as "headless" and try to switch LCD on later. Happens on RPi4 & RPi5.

Strange thing is, when I install the GutHub actions build (arm64 lean that is compiled "headless") I can switch on the LCD later and it works.

@openoms any idea where the difference might be?

openoms, 2023-12-07:
You can find the build log in the github action - can also be downloaded (faster to search that way).

Need to compare it with the output of the manual build. Are there any errors (especially around the LCD driver install)?

openoms, 2024-01-02:
Could be related: https://github.com/raspiblitz/raspiblitz/pull/4347

rootzoll, 2024-01-15:
I dont think it was #4347 - because it was just happeniong when I manual ran then buildscript .. but its not that relevant anymore because when I am using packer locally nor for the release build. So closing this issue.

closed 2024-01-15 by rootzoll