raspiblitz/raspiblitz #4311: LND wont Start after Update

Tanki90, 2023-12-09:
**Describe the bug**
Hello,
I had Raspiblitz version 1.8.0 installed, wanted to update to 1.10.0, then the device froze... after 24 hours I hard rebooted the device and started the device with a new SD card image.

Since then I get the message "Unbe able to determine number river" when starting up.

The new installation goes through, but then lnd doesn't start.
you can also see it in the logs.

Then I installed version 1.8.0 again but the error is still there.

Then I updated to the latest lnd version from the menu.

Nothing helped
Can someone help me, I don't know what the problem is.

I think this was the Problem!?
But why and how can i solve this?

78: Dec 06 22:01:26 raspberrypi lnd[45579]: failed to load config: /home/bitcoin/.lnd/lnd.conf:33: unknown option: routerrpc.apriorihopprob

**Debug file**
debug -l
"
Printing debug logs. Be patient, this should take maximum 2 minutes ..
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory
Redacting ..

redacting file: /var/cache/raspiblitz/debug.log
https://l.termbin.com/sm2k

openoms, 2023-12-09:
it is time ti update to  a more recent release.

Also the solution is given in the log file. Remove the flagged options from the lnd.conf:
```
/home/bitcoin/.lnd/lnd.conf:33: unknown option: routerrpc.apriorihopprob
```

Tanki90, 2023-12-10:
thx... need to delete the whole routerrpc lines. thx for your help

closed 2023-12-10 by Tanki90