raspiblitz/raspiblitz #4323: SSH Main Menu Fails to Launch After 4 Hours of RaspiBlitz Uptime

neversleep404, 2023-12-16:
**Describe the bug**
The SSH main menu fails to launch after the RaspiBlitz has been running for an extended period. It becomes inaccessible and only a restart resolves the issue.

PI 5 running on raspiblitz-arm64-rpi-image-2023-12-14-fac733a
The issue has been persisting for several weeks
This issue may potentially be related to Issue #4298

**To Reproduce**
After 4 Hours of Running RaspiBlitz, the SSH Main Menu Fails to Start During SSH Login.
At least for the first hour, SSH Main Menu works perfectly.

**Debug file**

[image]

rootzoll, 2023-12-18:
So the line "# starting ssh menu loop ..." is the last line you see and then it hangs forever, correct?

https://github.com/raspiblitz/raspiblitz/blob/622135c5704d3482a7113bf0744be30b01f53f6d/home.admin/00raspiblitz.sh#L79

neversleep404, 2023-12-18:
> So the line "# starting ssh menu loop ..." is the last line you see and then it hangs forever, correct?
Correct.

neversleep404, 2023-12-23:
After a New Installation and SEED+SCB Words Seed & channel.backup file recovery all working fine.

closed 2023-12-23 by neversleep404